arendst/Tasmota #13903: After Upgrade to Version 10.0.0 the BMP Shields are not recogized on D1 Mini and WEMOS anymore

Nesie, 2021-12-03:
### PROBLEM DESCRIPTION
After Upgrade to Version 10.0.0 the BMP Shields are not recogized on D1 Mini and WEMOS anymore
Upgrade was done over the air with:   http://ota.tasmota.com/tasmota/release/tasmota.bin.gz

After Downsizing to Version 9.5.0 everything is working again. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): __D1 Mini / Wemos___
- [ ] Tasmota binary firmware version number used: __10.0.0___
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-12-03:
Needed infos missing, this makes us impossible to help. Please provide all infos. Thx.

sfromis, 2021-12-03:
Assuming that "BMP Shield" includes a sensor like BMP280/BMP180 or BME280, you need tasmota-sensors.bin.gz to make it work.

Nesie, 2021-12-03:
Provide the output of command:

14:16:18.807 CMD: Backlog Template
14:16:18.943 MQT: stat/tasmota_65D0B1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}

14:13:23.444 CMD: Module
14:13:23.450 MQT: stat/tasmota_65D0B1/RESULT = {"Module":{"0":"Generic"}}

14:15:03.283 CMD: GPIO 255
14:15:03.293 MQT: stat/tasmota_65D0B1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"1312":"DS18x20"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

Output of Log:
14:30:51.046 SRC: WebCommand from 192.168.XXX.XXX
14:30:51.048 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
14:30:51.055 MQT: stat/tasmota_65D0B1/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_65D0B1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:30:51.086 MQT: stat/tasmota_65D0B1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin-DE.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:02","StartupUTC":"2021-12-03T13:28:49","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2021-10-15T11:19:29","SaveCount":22,"SaveAddress":"FB000"}}
14:30:51.122 MQT: stat/tasmota_65D0B1/STATUS2 = {"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021.10.19 08:23:57","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"383/699"}}
14:30:51.142 MQT: stat/tasmota_65D0B1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["B24E",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","2805C8000000060000005A0A000000000000","00008080","00006000","00004000"]}}
14:30:51.173 MQT: stat/tasmota_65D0B1/STATUS4 = {"StatusMEM":{"ProgramSize":613,"Free":388,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
14:30:51.207 MQT: stat/tasmota_65D0B1/STATUS5 = {"StatusNET":{"Hostname":"tasmota-65D0B1-4273","IPAddress":"192.168.XXX.XXX","Gateway":"192.168.XXX.XXX","Subnetmask":"255.255.255.0","DNSServer1":"192.168.XXX.XXX","DNSServer2":"0.0.0.0","Mac":"48:3F:DA:65:D0:B1","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:30:51.234 MQT: stat/tasmota_65D0B1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.XXX.XXX","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_65D0B1","MqttUser":"XXXXXX","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:30:51.258 MQT: stat/tasmota_65D0B1/STATUS7 = {"StatusTIM":{"UTC":"2021-12-03T13:30:51","Local":"2021-12-03T14:30:51","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:25","Sunset":"16:54"}}
14:30:51.279 MQT: stat/tasmota_65D0B1/STATUS10 = {"StatusSNS":{"Time":"2021-12-03T14:30:51","DS18B20":{"Id":"03069779B6B0","Temperature":20.7},"TempUnit":"C"}}
14:30:51.292 MQT: stat/tasmota_65D0B1/STATUS11 = {"StatusSTS":{"Time":"2021-12-03T14:30:51","Uptime":"0T00:02:02","UptimeSec":122,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"XXXX","BSSId":"E0:28:6D:82:74:20","Channel":9,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:03"}}}
14:30:54.548 WIF: Prüfe Verbindung...

Generall Config:

DeviceType:  Generic 0
1 Sensor DS18B20 on D4 GPIO2 ,   is found on 9.5.0   and 10.0.0
1 BMP180  on D2 GPIO4/ D1 GPIO5  as I2C SDA / SCL ,  is found on 9.5.0 but not on 10.0.0

Tried both 10.0.0 firmwares:  tasmota.bin-DE.gz and tasmota-sensors.bin-DE.gz
no difference allways DS18 is found, and BMP180 not.

After switching back to 9.5.0  tasmota.bin-DE  both sensors worked again.

closed 2021-12-03 by Nesie

Jason2866, 2021-12-03:
There is no precompiled `tasmota-sensors.bin-DE.gz`

Nesie, 2021-12-03:
Ok, yes that was my fault. I tried it now with tasmota-sensors.bin.gz and it works.
I tried it before with tasmota-sensors.bin-DE.gz and I didn't got any error that there is no such file.
Maybe some problems with the browser cache.
Many thanks for your support !!!