arendst/Tasmota #13905: Problem with detecting SHT30 Wemos Shield

EmilSodergren, 2021-12-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I try to get readings from my Wemos SHT30 Shield V2.0.0 but it is not detected by my Wemos D1 mini board. I am using the sensors-variant of the binary and I have tried with versions 10.0.0.4, 10.0.0, 9.3.1 and 8.5.1. I have used 2 different Wemos D1 boards and 2 different SHT30 shields. I do not get any indication that the board is even connected. I have mounted the units on a breadboard to be able to verify that the shield gets 3.3V.   

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x] Tasmota binary firmware version number used: 10.0.0.4, 10.0.0, 9.3.1 and 8.5.1 (all using the -sensors variant)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:10:59.976 MQT: stat/home/kitchen/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:11:00.201 MQT: stat/home/kitchen/RESULT = {"Module":{"18":"Generic"}}
14:11:00.408 MQT: stat/home/kitchen/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"4864":"ADC Range"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:12:05.983 MQT: stat/home/kitchen/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:12:06.217 MQT: stat/home/kitchen/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:12:06.464 MQT: stat/home/kitchen/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:12:55.852 MQT: stat/home/kitchen/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["kitchen"],"Topic":"kitchen","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:12:55.859 MQT: stat/home/kitchen/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release-9.3.1/tasmota-sensors.bin.gz","RestartReason":"External System","Uptime":"0T00:14:15","StartupUTC":"2021-12-03T12:58:40","Sleep":50,"CfgHolder":4617,"BootCount":106,"BCResetTime":"2021-01-08T20:13:59","SaveCount":177,"SaveAddress":"F9000"}}
14:12:55.867 MQT: stat/home/kitchen/STATUS2 = {"StatusFWR":{"Version":"10.0.0.4(sensors)","BuildDateTime":"2021-12-02T22:09:19","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"421/699"}}
14:12:55.875 MQT: stat/home/kitchen/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MrE@100-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
14:12:55.891 MQT: stat/home/kitchen/STATUS4 = {"StatusMEM":{"ProgramSize":668,"Free":332,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
14:12:55.899 MQT: stat/home/kitchen/STATUS5 = {"StatusNET":{"Hostname":"kitchen-6257","IPAddress":"192.168.60.8","Gateway":"192.168.60.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.60.1","DNSServer2":"0.0.0.0","Mac":"48:3F:DA:00:38:71","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:12:55.905 MQT: stat/home/kitchen/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.14","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_003871","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:12:55.913 MQT: stat/home/kitchen/STATUS7 = {"StatusTIM":{"UTC":"2021-12-03T13:12:55","Local":"2021-12-03T14:12:55","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:25","Sunset":"16:54"}}
14:12:55.927 MQT: stat/home/kitchen/STATUS10 = {"StatusSNS":{"Time":"2021-12-03T14:12:55","ANALOG":{"Range":1}}}
14:12:55.934 MQT: stat/home/kitchen/STATUS11 = {"StatusSTS":{"Time":"2021-12-03T14:12:55","Uptime":"0T00:14:15","UptimeSec":855,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MrE@100-IoT","BSSId":"C2:FB:E4:4A:2A:60","Channel":11,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.002 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F8, Count 178
00:00:00.055 FRC: Some settings have been reset (4)
00:00:00.057 CFG: CR 421/699, Busy 0
00:00:00.062 CFG: CR 421/699, Busy 0
00:00:00.064 ROT: Mode 1
00:00:00.066 SRC: Restart
00:00:00.068 Project tasmota - Tasmota Version 10.0.0.4(sensors)-2_7_4_9(2021-12-02T22:09:19)
00:00:00.081 SHT: Sensor did not ACK command
00:00:00.200 WIF: Checking connection...
00:00:00.201 WIF: Attempting connection...
00:00:00.537 WIF: Connecting to AP1 MrE@100-IoT Channel 11 BSSId C2:FB:E4:4A:2A:60 in mode 11n as kitchen-6257...
00:00:01.756 WIF: Checking connection...
00:00:01.757 WIF: Connected
00:00:02.009 HTP: Web server active on kitchen-6257 with IP address 192.168.60.8
00:00:02.406 WIF: Sending Gratuitous ARP
00:00:02.408 NTP: Synch time...
00:00:02.451 RTC: UTC 2021-12-03T13:15:11, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
14:15:11.000 NTP: Synched
[... 16 lines cut ...]
14:15:16.395 MQT: tele/home/kitchen/STATE = {"Time":"2021-12-03T14:15:16","Uptime":"0T00:00:09","UptimeSec":9,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MrE@100-IoT","BSSId":"C2:FB:E4:4A:2A:60","Channel":11,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}
14:15:16.427 MQT: tele/home/kitchen/SENSOR = {"Time":"2021-12-03T14:15:16","ANALOG":{"Range":1}}
14:15:17.390 APP: Boot Count 107
14:15:17.592 CFG: Saved to flash at F7, Count 179, Bytes 4096
14:15:21.420 MQT: tasmota/discovery/483FDA003871/config = {"ip":"192.168.60.8","dn":"Tasmota","fn":["kitchen",null,null,null,null,null,null,null],"hn":"kitchen-6257","mac":"483FDA003871","md":"Generic","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"10.0.0.4","t":"kitchen","ft":"%prefix%/home/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
14:15:21.481 MQT: tasmota/discovery/483FDA003871/sensors = {"sn":{"Time":"2021-12-03T14:15:21","ANALOG":{"Range":1}},"ver":1} (retained)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect the device as shown in https://tasmota.github.io/docs/SHT30/ . No soldering has been done. Power the Wemos

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The output of the first page should read as the example, showing temp, humidity and dewpoint. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Have tried with two boards, two shields and 4 different prebuilt firmwares. 

**(Please, remember to close the issue when the problem has been addressed)**

EmilSodergren, 2021-12-03:
Ok... the "sensor did not ACK command" surely sounds bad. Any thoughts of why this happens?

Jason2866, 2021-12-03:
What do you get with `I2Cscan`?
Disable all not needed sensor drivers with `I2CDriver`

EmilSodergren, 2021-12-03:
```lua
15:29:35.471 CMD: I2CScan
15:29:35.473 SRC: WebConsole from 192.168.1.195
15:29:35.475 CMD: Grp 0, Cmnd 'I2CSCAN', Idx 1, Len 0, Data ''
15:29:35.496 MQT: stat/home/kitchen/RESULT = {"I2CScan":"No devices found"}
```
```lua
15:30:24.647 SRC: WebConsole from 192.168.1.195
15:30:24.649 CMD: Grp 0, Cmnd 'I2CDRIVER', Idx 1, Len 0, Data ''
15:30:24.656 MQT: stat/home/kitchen/RESULT = {"I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48"}
```

EmilSodergren, 2021-12-03:
Ok, I will try that. Isn't the SHT3X on driver 15 by the way? I will try to play around with that and get back to you

sfromis, 2021-12-03:
You can see driver numbers in the link I posted. And yes, SHT3X is number 15.

EmilSodergren, 2021-12-03:
:warning: :warning: :warning: WARNING! WARNING! WARNING! :warning: :warning: :warning:
SOFTWARE GUY TRYING TO DO HARDWARE!!!
:warning: :warning: :warning: WARNING! WARNING! WARNING! :warning: :warning: :warning:

With some "improved" electrical connectivity between Wemos and SHT shield everything works beautifully... :man_facepalming: . Thank you all for your help! Fantastic piece of software by the way...

closed 2021-12-03 by EmilSodergren