arendst/Tasmota #13916: POWER1 OFF issed after setting PulseTime to another Relay

mrslvd, 2021-12-04:
### PROBLEM DESCRIPTION
I have 8 relays on my board. When I set PulseTime (for example: pulsetime8 10) a power1 off command is issued by tasmota device.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32S custom board
- [x] Tasmota binary firmware version number used: 10.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP-Flasher-Windows-x64
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior: 
write a command: pulsetime8 10

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect that only relay8 output will perform any action. In this case power8 off after 1 second.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
The following is copied from console:
12:38:47.859 CMD: pulsetime8 10
12:38:47.865 RSL: RESULT = {"PulseTime8":{"Set":10,"Remaining":10}}
12:38:48.923 RSL: RESULT = {"POWER8":"OFF"}
12:38:48.926 RSL: POWER8 = OFF
12:38:48.931 RSL: RESULT = {"POWER1":"OFF"}
12:38:48.934 RSL: POWER1 = OFF

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-12-04:
Failed to reproduce.

If you did not ignore the template for documenting your situation, you could have supplied relevant info for seeing what happens, especially the extra details from increasing the logging level.

mrslvd, 2021-12-04:
Sorry, I've missed this. I hope the updated info will help.
Best regards,
Miro

barbudor, 2021-12-04:
I can reproduce.
If you set pulsetime on every power 1 to 8, the automatic power off of any is also issuing an off on power1

This happens as soon as there is more than 8 power devices (8 relays + 1 PWM or 9 relays).
It should not behave like that
The expected behavior is
`Defined PulseTime for relays <1-8> will also be active for correspondent Relay <9-16>.`
So in your case only POWER9 OFF should also trigger POWER1 OFF, not the other

I'm looking into it

barbudor, 2021-12-04:
Fixed in https://github.com/arendst/Tasmota/pull/13920
Please wait for PR to be merged and new development binaries made available at http://ota.tasmota.com/tasmota32/

mrslvd, 2021-12-04:
Thanks!
Best regards,
Miro

closed 2021-12-04 by mrslvd