arendst/Tasmota #13919: After update Tasmota32 to ver10.0.0 GPIO16 and GPIO17 toggles ON and OFF after Power cycle or  Restart

mrslvd, 2021-12-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After updating to ver10.0.0 GPIO16 and GPIO17 are toggling for a while. I don't have this issue with ver9.5.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32S on custom board
- [x] Tasmota binary firmware version number used: tasmota32 ver10.0.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP-Flasher-Windows-x64
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:41:18.816 CMD: Backlog Template; Module; GPIO 255
14:41:18.886 RSL: RESULT = {"NAME":"ESP32-FULL-3","GPIO":[1,1,1,1,229,418,1,1,1,230,416,417,228,227,225,224,0,1,1,226,0,0,1,231,0,0,0,0,4704,4705,194,195,192,0,0,193],"FLAG":0,"BASE":1}
14:41:19.040 CMD: 
14:41:19.101 RSL: RESULT = {"Module":{"0":"ESP32-FULL-3"}}
14:41:19.369 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"229":"Relay6"},"GPIO5":{"418":"PWM3"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"230":"Relay7"},"GPIO14":{"416":"PWM1"},"GPIO15":{"417":"PWM2"},"GPIO16":{"228":"Relay5"},"GPIO17":{"227":"Relay4"},"GPIO18":{"225":"Relay2"},"GPIO19":{"224":"Relay1"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"226":"Relay3"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"231":"Relay8"},"GPIO32":{"4704":"ADC Input1"},"GPIO33":{"4705":"ADC Input2"},"GPIO34":{"194":"Switch_n3"},"GPIO35":{"195":"Switch_n4"},"GPIO36":{"192":"Switch_n1"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"193":"Switch_n2"}}
14:41:19.709 CMD: 
14:41:19.719 CMD: 

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:42:11.731 CMD: status 0
14:42:11.739 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"ESP_FULL_3","FriendlyName":["Tasmota","","","","","","",""],"Topic":"tasmota_488D1C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:42:11.773 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:02:46","StartupUTC":"2021-12-04T13:39:25","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2021-12-04T09:12:11","SaveCount":55}}
14:42:11.804 RSL: STATUS2 = {"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021-10-19T08:26:22","Core":"1_0_7_4","SDK":"v3.3.5-245-g95097b38d","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"401/699"}}
14:42:11.822 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NightProwler2G4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00004000"]}}
14:42:11.860 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1269,"Free":1856,"Heap":132,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52,59","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,127"}}
14:42:11.909 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-488D1C-3356","IPAddress":"192.168.0.181","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"192.168.0.1","Mac":"7C:9E:BD:48:8D:1C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:42:11.936 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_488D1C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:42:11.956 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-12-04T13:42:11","Local":"2021-12-04T14:42:11","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:26","Sunset":"16:54"}}
14:42:12.109 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-12-04T14:42:11","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON","ANALOG":{"A1":101,"A2":0},"ESP32":{"Temperature":53.3},"TempUnit":"C"}}
14:42:12.133 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-12-04T14:42:12","Uptime":"0T00:02:47","UptimeSec":167,"Heap":131,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":94,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","Dimmer":7,"Color":"121212","HSBColor":"0,0,7","Channel":[7,7,7],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"NightProwler2G4","BSSId":"04:D4:C4:3D:76:80","Channel":1,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:06"}}}
14:42:12.186 RSL: STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"StoreProhibited","EPC":"4008e606","EXCVADDR":"00000000","CallChain":["4008e603","4008ea35","40172d37","40089ca1","401bbcaf","40173f53","40091301"]}}
14:42:12.506 CMD: 

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
THIS IS AFTER RESTART
00:00:00.002 HDW: ESP32-D0WDQ6 
00:00:00.002 HDW: FoundPSRAM=0 CanUsePSRAM=0
00:00:00.032 UFS: FlashFS mounted with 304 kB free
00:00:00.086 CFG: Loaded from File, Count 56
00:00:00.098 QPC: Count 1
00:00:00.101 CFG: CR 401/699, Busy 0
00:00:00.108 CFG: CR 401/699, Busy 0
00:00:00.109 CFG: CR 401/699, Busy 0
00:00:00.111 ROT: Mode 1
00:00:00.113 BRY: GC from 742 to 742 bytes (in 0 ms)
00:00:00.117 BRY: GC from 1525 to 1525 bytes (in 0 ms)
00:00:00.163 BRY: GC from 3087 to 2509 bytes (in 1 ms)
00:00:00.182 BRY: GC from 3701 to 3156 bytes (in 1 ms)
00:00:00.182 BRY: Berry initialized, RAM used=3156
[... 41 lines cut ...]
14:43:14.888 RSL: POWER9 = OFF
14:43:14.896 TFS: File 'autoexec.bat' not found
14:43:15.082 QPC: Reset
14:43:15.531 HTP: Main Menu
14:43:17.002 APP: Boot Count 12
14:43:17.551 CFG: Saved, Count 57, Bytes 4096
14:43:19.499 RSL: STATE = {"Time":"2021-12-04T14:43:19","Uptime":"0T00:00:12","UptimeSec":12,"Heap":136,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":329,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","Dimmer":7,"Color":"121212","HSBColor":"0,0,7","Channel":[7,7,7],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"NightProwler2G4","BSSId":"04:D4:C4:3D:76:80","Channel":1,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:06"}}
14:43:19.679 RSL: SENSOR = {"Time":"2021-12-04T14:43:19","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON","ANALOG":{"A1":113,"A2":15},"ESP32":{"Temperature":53.3},"TempUnit":"C"}
14:43:35.403 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Before updating to ver10.0.0 I don't have this problem

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mrslvd, 2021-12-04:
Just to confirm. I upload again ver9.5 to the device and GPIO16 and GPIO7 are not toggling after Power cycling or Restart.
Best regards,
Miro

sfromis, 2021-12-04:
This is as expected. Gpio16 and 17 are polled for presence of PSRAM at boot time. Better not use them for other purposes, unless this is not an issue to you.

mrslvd, 2021-12-04:
I read about this but with version 9.5.0 I don't observe the same behaviour.

sfromis, 2021-12-04:
Yes, having PSRAM detection in all binaries is a recent improvement, not present in version 9.5.

mrslvd, 2021-12-04:
Is there an option if using self compiled bin to disable the PSRAM detection?

sfromis, 2021-12-04:
Not as an option. If you go very deep, you could change the process to tear out PSRAM support, not a simple thing to do. Much more work than just moving the relays to suitable pins instead.

mrslvd, 2021-12-04:
On some modules as I know GPIO16 and GPIO17 are used for PSRAM but I have considered this and because of this I am using modules that are not using these GPIOs. My board was designed with ver9.5 and now I am a little bit surprised... Never mind... This does not depend on me...

sfromis, 2021-12-04:
The reason for not having flexibility is that PSRAM detection is done outside of Tasmota, and it was not deemed worth it to duplicate everything for having with and without the detection. With most use cases for the pins, the polling is not a problem, but relays can indeed trigger.

mrslvd, 2021-12-04:
The problem is that this change in version 10 comes after the boards were designed and manufactured... Bad surprise... 
Thanks for your help.
Best regards,
Miro

closed 2021-12-04 by mrslvd