arendst/Tasmota #13923: MiEl HVAC module - read temperature resolution & incorrect temp in tele/tasmota/HVACSETTINGS

logon84, 2021-12-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am Using Miel HVAC and even when I set TempRes to 1, device room temperature and set point temperature are not read with 0.5 prescissión, still integers with ".0" attached. Wouldn't it be logical to read temps from the Mithsubishi heatpump with the resolution it has built-in (0.5) when TempRes is set to 1?

Also, after setting the work temperature to 25.5 in Mitsubishi control panel, temperature reported by tele/tasmota/HVACSETTINGS is wrong (I get 9.0)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): MiEl HVAC
- [X] Tasmota binary firmware version number used: 10.0.0.4
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: USB to Serial TTL & platformio run -e tasmota-ES --target upload --upload-port /dev/ttyUSB0
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:36:51.142 MQT: stat/heatpump/RESULT = {"NAME":"heatpump","GPIO":[1,5376,1,5408,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
00:36:51.382 MQT: stat/heatpump/RESULT = {"Module":{"0":"heatpump"}}
00:36:51.657 MQT: stat/heatpump/RESULT = {"GPIO0":{"0":"Ninguno"},"GPIO1":{"5376":"MiEl HVAC Tx"},"GPIO2":{"0":"Ninguno"},"GPIO3":{"5408":"MiEl HVAC Rx"},"GPIO4":{"0":"Ninguno"},"GPIO5":{"0":"Ninguno"},"GPIO9":{"0":"Ninguno"},"GPIO10":{"0":"Ninguno"},"GPIO12":{"0":"Ninguno"},"GPIO13":{"0":"Ninguno"},"GPIO14":{"0":"Ninguno"},"GPIO15":{"0":"Ninguno"},"GPIO16":{"0":"Ninguno"},"GPIO17":{"0":"Ninguno"}}
00:36:58.484 MQT: tele/heatpump/SENSOR = {"Time":"2021-12-05T00:36:58","MiElHVAC":{"Temperature":22.0,"Operation":"ON","Compressor":"OFF","roomtemp":"0300000C009CAC000000000000000000","status":"060000000102727FE000000000000000","stage":"09000004020000000000000000000000"},"TempUnit":"C"} (Grabado)

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:37:53.671 MQT: stat/heatpump/STATUS = {"Status":{"Module":0,"DeviceName":"MiEl-HVAC","FriendlyName":["Tasmota"],"Topic":"heatpump","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0,"InfoRetain":1,"StateRetain":1}}
00:37:53.677 MQT: stat/heatpump/STATUS1 = {"StatusPRM":{"Baudrate":2400,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T13:23:43","StartupUTC":"2021-12-04T10:14:10","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2021-12-02T17:47:40","SaveCount":86,"SaveAddress":"F9000"}}
00:37:53.681 MQT: stat/heatpump/STATUS2 = {"StatusFWR":{"Version":"10.0.0.4(tasmota)","BuildDateTime":"2021-12-04T11:12:29","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"368/699"}}
00:37:53.685 MQT: stat/heatpump/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXXt",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008209","2805C80001000600003C5A0A000000000000","00000080","00006000","00004180"]}}
00:37:53.697 MQT: stat/heatpump/STATUS4 = {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["0000040A","8FDAC787","04368001","000000CF","010013C0","C000F981","04004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,44,45","Sensors":"1,2,3,4,5,6"}}
00:37:53.705 MQT: stat/heatpump/STATUS5 = {"StatusNET":{"Hostname":"tasm-heatpump","IPAddress":"192.168.1.151","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"80:7D:3A:69:76:5D","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:37:53.710 MQT: stat/heatpump/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1883,"MqttClientMask":"XXXXXXX","MqttClient":"XXXXXXX","MqttUser":"XXXXXXX","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:37:53.717 MQT: stat/heatpump/STATUS7 = {"StatusTIM":{"UTC":"2021-12-04T23:37:53","Local":"2021-12-05T00:37:53","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:27","Sunset":"16:53"}}
00:37:53.725 MQT: stat/heatpump/STATUS10 = {"StatusSNS":{"Time":"2021-12-05T00:37:53","MiElHVAC":{"Temperature":22.0,"Operation":"ON","Compressor":"OFF","roomtemp":"0300000C009CAC000000000000000000","status":"0600000001020F7FE000000000000000","stage":"09000004030000000000000000000000"},"TempUnit":"C"}}
00:37:53.731 MQT: stat/heatpump/STATUS11 = {"StatusSTS":{"Time":"2021-12-05T00:37:53","Uptime":"0T13:23:43","UptimeSec":48223,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"XXXXXXX","BSSId":"14:91:82:73:67:85","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
sending FC42013010 06000000000000000000000000000000 77
00:39:14.359 MiElHVAC: response data 020000010116030500000CB300000000
00:39:14.511 MiElHVAC: sending FC42013010 02000000000000000000000000000000 7b
00:39:14.721 MiElHVAC: response data 020000010116030500000CB300000000

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Subscribe to /cmnd/tasmota/SENSOR and watch "Temperature" when the HVAC has 21.5 roomtemp on control panel:
{"Time":"2021-12-05T00:42:55","MiElHVAC":{"Temperature":22.0,"Operation":"ON","Compressor":"OFF","roomtemp":"0300000C009AAC000000000000000000","status":"060000000104977FE100000000000000","stage":"09000000030000000000000000000000"},"TempUnit":"C"}

Subscribe to /tele/tasmota/HVACSETTINGS and watch wrong Temp value when the HVAC has been set to 25.5 in control panel:
{"Power":"ON","Mode":"heat","HAMode":"heat","Temp":9.0,"FanSpeed":"2","SwingV":"5","SwingH":"swing","Bytes":"020000010116030500000CB300000000"}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
/cmnd/tasmota/SENSOR when roomtemp is 21.5 should show:
{"Time":"2021-12-05T00:42:55","MiElHVAC":{"Temperature":21.5,"Operation":"ON","Compressor":"OFF","roomtemp":"0300000C009AAC000000000000000000","status":"060000000104977FE100000000000000","stage":"09000000030000000000000000000000"},"TempUnit":"C"}

/tele/tasmota/HVACSETTINGS when work roomtemp is set  to 25.5 should show:
{"Power":"ON","Mode":"heat","HAMode":"heat","Temp":25.5,"FanSpeed":"2","SwingV":"5","SwingH":"swing","Bytes":"020000010116030500000CB300000000"}
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-12-07:
Thanks for reporting and providing the solution with your PR #13936

Closing due to the fix is provided in your PR.

Thanks

closed 2021-12-07 by ascillato2

logon84, 2022-06-24:
Regarding this module, sometimes when I activate or deactivate the module via MQTT (I always use mqtt), a SRC SWITCH appears and sets the to the original state:
14:17:56.203 MiElHVAC: response data 060000000000199DD400000000000000
14:17:56.258 MiElHVAC: sending FC42013010 02000000000000000000000000000000 7b
14:17:56.361 SRC: MQTT
14:17:56.364 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 2, Pld 1, Data 'ON'
14:17:56.370 MQT: stat/heatpump/RESULT = {"POWER":"ON"}
14:17:56.375 MQT: stat/heatpump/POWER = ON
14:17:56.462 MiElHVAC: sending FC41013010 01010001000000000000000000000000 7b
14:17:56.564 MiElHVAC: response data 020000000307030500000CB000000000
14:17:56.566 SRC: Switch
14:17:56.574 MQT: stat/heatpump/RESULT = {"POWER":"OFF"}
14:17:56.578 MQT: stat/heatpump/POWER = OFF

Have I some miss configuration? I was thinking, will this be quickly fixable  using a rule?

logon84, 2022-06-24:
I think I solved it by setting Setoption114

logon84, 2022-06-25:
UPDATE: not solved with Setoption114

PR #13936 (2021-12-08): Support for enhanced temp registers in newer Mitsubishi units
## Description:
This PR enables using more accurate temperature resolution registers (0.5 res) in newer Mitsubishi devices. Ir provides support for reading and setting half degree temperatures (21.5, 22.5, 23.5, etc....). Using old registers in newer Mitsubishis models could lead to wrong temperature readings (Ex: setting 25.5C in Mitsubishi wall panel triggers working temp of 9C in HVACESETTINGS.
**Related issue (if applicable):** fixes #13923

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [... cut]