arendst/Tasmota #13924: Smart Meter delta calculations can produce spikes on restart

pputerla, 2021-12-05:
### PROBLEM DESCRIPTION
Uninitialized value of dvalues[x] in /tasmota/xsns_53_sml.ino:1555 causes spikes (aka random high or low value) on restart.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): *NodeMCU*
- [x] Tasmota binary firmware version number used: *10.0.0(TasmoCompiler-esp82664M)*
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
01:14:29.350 MQT: stat/tasmota_C478CF/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
01:14:29.559 MQT: stat/tasmota_C478CF/RESULT = {"Module":{"0":"Generic"}}
01:14:29.815 MQT: stat/tasmota_C478CF/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
01:25:59.846 CMD: status 0
01:25:59.854 MQT: stat/tasmota_C478CF/STATUS = {"Status":{"Module":0,"DeviceName":"Power Meter Lancut","FriendlyName":["Power Meter Lancut"],"Topic":"tasmota_C478CF","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":60,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
01:25:59.887 MQT: stat/tasmota_C478CF/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:02","StartupUTC":"2021-12-05T00:21:57","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2021-11-11T16:47:00","SaveCount":114,"SaveAddress":"3FB000"}}
01:25:59.923 MQT: stat/tasmota_C478CF/STATUS2 = {"StatusFWR":{"Version":"10.0.0(TasmoCompiler-esp82664M)","BuildDateTime":"2021-11-11T15:56:46","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"405/699"}}
01:25:59.945 MQT: stat/tasmota_C478CF/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Pimpek","evora1"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00206000","00004000"]}}
01:25:59.975 MQT: stat/tasmota_C478CF/STATUS4 = {"StatusMEM":{"ProgramSize":644,"Free":1400,"Heap":12,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87C0458F","042E0001","1400000F","410013C0","C800EDC9","08244084","00201000","00000020"],"Drivers":"1,2,3,4,6,8,9,10,12,18,19,22,24,26,27,29,30,35,37,38,45,50","Sensors":"1,2,3,53"}}
01:26:00.002 MQT: stat/tasmota_C478CF/STATUS5 = {"StatusNET":{"Hostname":"tasmota-C478CF-6351","IPAddress":"192.168.0.117","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"89.228.4.126","DNSServer2":"31.11.173.2","Mac":"E8:DB:84:C4:78:CF","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
01:26:00.025 MQT: stat/tasmota_C478CF/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.112","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C478CF","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
01:26:00.051 MQT: stat/tasmota_C478CF/STATUS7 = {"StatusTIM":{"UTC":"2021-12-05T00:26:00","Local":"2021-12-05T01:26:00","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:27","Sunset":"16:53"}}
01:26:00.069 MQT: stat/tasmota_C478CF/STATUS10 = {"StatusSNS":{"Time":"2021-12-05T01:26:00","METER":{"Total":57.512,"Power":68}}}
01:26:00.081 MQT: stat/tasmota_C478CF/STATUS11 = {"StatusSTS":{"Time":"2021-12-05T01:26:00","Uptime":"0T00:04:03","UptimeSec":243,"Heap":12,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Pimpek","BSSId":"BC:F6:85:42:28:94","Channel":1,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

given the following script is used & SML is enabled:
```lua
>D
>B
=>sensor53 r
>M 1
+1,2,c,0,-10,METER
1,1-0:1.8.0*255(@6400,Total,kWh,Total,3
1,=d 1 9 @1,Power,W,Power,0  
#
```

step 1 - restart
step 2 - read power value during first 9 seconds - will be 0 (btw, should not be, power is not defined yet)
step 3 - read power value between 9th and 18th second - will be in random range due to uninitialized value of variable (PR pending)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Power reads should  not be 0 nor random value. 
Power show be available only after first *valid* calculation.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
(the setup is using 9 seconds for the refresh delay (`1,=d 1 9 @1,Power,W,Power,0`)
restart:
[image]
after ~9 seconds:
[image]
after ~2*9 seconds and more:
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
PR showing what could be fixed is on it's way.

**(Please, remember to close the issue when the problem has been addressed)**

pputerla, 2021-12-05:
I can imagine in certain compilations and memory alignments, each restart will incidentally initialize dvalues with zeros thus the bug is hard to observe. But once it's observed it's reproduceable.

Anyway the PR also aims to expose the sensor only after it's calculation is valid. In this case the dvalues (given is initialized) is not 0.

This apparently means the power calculations will be visible after 2 * measuring interval.

pputerla, 2021-12-05:
I've verified (as per comment in PR referring to #11053) and smlj  && "upsecs" works well.

Still I emphasize the necessity of initializing dvalues. Just for the sake of fixing runtime bug.

BTW. I had to figure out to include USE_SML_SCRIPT_CMD. It wasn't obvious.

PS. For refrence, one of my SML scripts look like this:
```lua
>D
>B
smlj=0
=>sensor53 r
>R
smlj=0
=>sensor53 r
>S
if upsecs>11
then
smlj=1
endif
>M 1
+1,2,c,1,-5,L3F.MID
1,1-0:1.8.0*255(@1000,Total,kWh,Total,3
1,=m 1+#9032 @1,Total STOEN,kWh,Total STOEN,2  
1,=d 1 5 @1,Power,W,Power,0  
#
```

ascillato2, 2021-12-07:
Thanks for sharing your solution !

closed 2021-12-07 by ascillato2