arendst/Tasmota #13959: OTA Update to 10.1.0 ESP32 WEBCAM not possible

helmar74, 2021-12-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I tried to update my ESP32 WEBCAM from version 10.0.0 to 10.1.0 via OTA.
It fails with "not enough space error" although there is more than the 1037KB available:
[image]

[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32CAM
- [ ] Tasmota binary firmware version number used: 10.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
6:38:56.755 MQT: stat/webcam1/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
16:38:56.955 MQT: stat/webcam1/RESULT = {"Module":{"2":"ESP32-Cam"}}
16:38:57.180 MQT: stat/webcam1/RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"704":"SPI MOSI1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"},"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"0":"None"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:39:59.604 MQT: stat/webcam1/STATUS = {"Status":{"Module":2,"DeviceName":"Webcam1","FriendlyName":["Webcam1"],"Topic":"webcam1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
16:39:59.634 MQT: stat/webcam1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-webcam.bin","RestartReason":"Software reset digital core","Uptime":"0T00:03:22","StartupUTC":"2021-12-08T15:36:37","Sleep":50,"CfgHolder":4617,"BootCount":71,"BCResetTime":"2020-07-07T11:46:44","SaveCount":204}}
16:39:59.667 MQT: stat/webcam1/STATUS2 = {"StatusFWR":{"Version":"10.0.0(webcam)","BuildDateTime":"2021-10-19T08:26:40","Core":"1_0_7_4","SDK":"v3.3.5-245-g95097b38d","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6","CR":"461/699"}}
16:39:59.687 MQT: stat/webcam1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID1","SSID2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00004000"]}}
16:39:59.721 MQT: stat/webcam1/STATUS4 = {"StatusMEM":{"ProgramSize":1023,"Free":1984,"Heap":91,"PsrMax":4095,"PsrFree":3345,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F90058A","00040001","00000082","00000000","40000981","80000080","00200000","00000000"],"Drivers":"1,2,4,7,8,9,10,12,24,27,29,38,50,81","Sensors":"2,6,127"}}
16:39:59.751 MQT: stat/webcam1/STATUS5 = {"StatusNET":{"Hostname":"webcam1","IPAddress":"192.168.1.71","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"24:6F:28:13:D6:24","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:39:59.775 MQT: stat/webcam1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"webcam1","MqttClient":"webcam1","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:39:59.796 MQT: stat/webcam1/STATUS7 = {"StatusTIM":{"UTC":"2021-12-08T15:39:59","Local":"2021-12-08T16:39:59","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:30","Sunset":"16:53"}}
16:39:59.819 MQT: stat/webcam1/STATUS10 = {"StatusSNS":{"Time":"2021-12-08T16:39:59","ESP32":{"Temperature":46.7},"TempUnit":"C"}}
16:39:59.829 MQT: stat/webcam1/STATUS11 = {"StatusSTS":{"Time":"2021-12-08T16:39:59","Uptime":"0T00:03:22","UptimeSec":202,"Heap":91,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":786,"MqttCount":1,"POWER":"ON","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SSID1","BSSId":"34:31:C4:76:5C:90","Channel":11,"Mode":"11n","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:05"}}}
16:39:59.870 MQT: stat/webcam1/STATUS12 = {"StatusSTK":{"Exception":7,"Reason":"PCValue","EPC":"4016a6c8","EXCVADDR":"00000000","CallChain":["400855fe","40082ced","400881ad","400621b6","40093d23","40093d5e","40094068","400857c9","40178fdd","4010c241","40101de0","40101fe1","400eb372","4018fc69","4010028a","40100482","4010066c","40100ea6","40101a30","401003cf","400dc163","400ebca3","400e5f33","400f97f6","400f98a7","40111d41"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Update ESP32CAM from 10.0.0 to 10.1.0 with OTA

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-12-08:
Hi, can you try again but this time having the Tasmota console in a separated window so as to catch the logs? Please, set weblog to 4 before that. Thanks.

helmar74, 2021-12-08:
> Hi, can you try again but this time having the Tasmota console in a separated browser tab so as to catch the logs? Please, set weblog to 4 before that. Thanks.

Hi, sure. Here we go:
`18:55:42.663 WIF: Checking connection...
18:55:48.562 CFG: Saved, Count 208, Bytes 4096
18:55:48.562 UPL: File tasmota32-webcam_v1010.bin
18:55:48.578 UPL: File type 0
18:55:52.130 HTP: Upload done
18:55:53.712 HTP: Main Menu
18:55:53.766 CAM: Stream init
18:55:53.769 CAM: User template
18:55:53.769 CAM: PSRAM found
18:55:58.073 CAM: Init failed to get the frame on time
18:56:00.747 CAM: Handle camserver
18:56:00.747 CAM: Create client
18:56:00.748 CAM: Start stream
18:56:04.752 CAM: Frame fail
18:56:04.753 CAM: Stream exit
18:56:06.156 CAM: Handle camserver
18:56:06.157 CAM: Create client
18:56:06.158 CAM: Start stream
18:56:10.160 CAM: Frame fail
18:56:10.161 CAM: Stream exit
18:56:10.960 CAM: Handle camserver
18:56:10.960 CAM: Create client
18:56:10.961 CAM: Start stream
18:56:14.963 CAM: Frame fail
18:56:14.964 CAM: Stream exit
18:56:16.185 HTP: Firmware Upgrade
18:56:18.892 CAM: Handle camserver
18:56:18.893 CAM: Create client
18:56:18.894 CAM: Start stream
18:56:22.896 CAM: Frame fail
18:56:22.897 CAM: Stream exit
00:00:00.002 HDW: ESP32-D0WDQ6 (PSRAM)
00:00:00.003 HDW: FoundPSRAM=1 CanUsePSRAM=1
00:00:00.006 UFS: FlashFS mounted with 48 kB free
00:00:00.013 CFG: Loaded from File, Count 209
00:00:00.027 QPC: Count 1
00:00:00.029 SPI: Hardware using GPIO14(CLK), GPIO15(MOSI) and GPIO02(MISO)
00:00:00.031 ROT: Mode 1
00:00:00.643 SRC: Restart
00:00:00.645 Project tasmota - Webcam1 Version 10.0.0(webcam)-1_0_7_4(2021-10-19T08:26:40)
00:00:00.896 WIF: Checking connection...
00:00:00.896 WIF: Attempting connection...
00:00:01.380 WIF: Connecting to AP1 hs.net_aussen_24GHz Channel 11 BSSId 34:31:C4:76:5C:90 in mode 11n as webcam1...
00:00:02.399 WIF: Checking connection...
00:00:02.399 WIF: Attempting connection...
00:00:03.399 WIF: Checking connection...
00:00:03.399 WIF: Attempting connection...
00:00:04.279 WIF: Checking connection...
00:00:04.279 WIF: Connected
00:00:04.534 HTP: Web server active on webcam1 with IP address 192.168.1.71
00:00:04.821 NTP: Synch time...
00:00:05.009 RTC: UTC 2021-12-08T17:56:37, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
18:56:37.000 NTP: Synched
18:56:37.633 MQT: Attempting connection...
18:56:37.680 MQT: Connected
18:56:37.687 MQT: tele/webcam1/LWT = Online (retained)
18:56:37.696 MQT: cmnd/webcam1/POWER = 
18:56:37.697 MQT: Subscribe to cmnd/webcam1/#
18:56:37.700 MQT: Subscribe to cmnd/tasmotas/#
18:56:37.702 MQT: Subscribe to cmnd/webcam1_fb/#
18:56:37.707 MQT: tele/webcam1/INFO1 = {"Info1":{"Module":"ESP32-Cam","Version":"10.0.0(webcam)","FallbackTopic":"cmnd/webcam1_fb/","GroupTopic":"cmnd/tasmotas/"}}
18:56:37.714 MQT: tele/webcam1/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"webcam1","IPAddress":"192.168.1.71"}}
18:56:37.733 MQT: tele/webcam1/INFO3 = {"Info3":{"RestartReason":{"Exception":7,"Reason":"PCValue","EPC":"4016a6c8","EXCVADDR":"00000000","CallChain":["40085607","40082ced","400881ad","400621bc","40093d23","40093d5e","40094068","400857c9","40178fdd","4010c241","40101de0","40101fe1","400eb372","4018fc69","4010028a","40100482","4010066c","40100ea6","40101a30","401003cf","400dc163","400ebca3","400e5f33","400f97f6","400f98a7","40111d41"]}}}
18:56:37.765 MQT: stat/webcam1/RESULT = {"POWER":"ON"}
18:56:37.772 MQT: stat/webcam1/POWER = ON
18:56:37.776 TFS: File 'autoexec.bat' not found
18:56:38.944 QPC: Reset
18:56:40.941 APP: Boot Count 73
18:56:41.291 CFG: Saved, Count 210, Bytes 4096
18:56:41.955 MQT: tele/webcam1/STATE = {"Time":"2021-12-08T18:56:41","Uptime":"0T00:00:10","UptimeSec":10,"Heap":139,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":87,"MqttCount":1,"POWER":"ON","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"hs.net_aussen_24GHz","BSSId":"34:31:C4:76:5C:90","Channel":11,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:05"}}
18:56:41.987 MQT: tele/webcam1/SENSOR = {"Time":"2021-12-08T18:56:41","ESP32":{"Temperature":48.3},"TempUnit":"C"}
18:56:46.956 MQT: tasmota/discovery/246F2813D624/config = {"ip":"192.168.1.71","dn":"Webcam1","fn":["Webcam1",null,null,null,null,null,null,null],"hn":"webcam1","mac":"246F2813D624","md":"ESP32-Cam","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"10.0.0","t":"webcam1","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":1,"sho":[0,0,0,0],"ver":1} (retained)
18:56:46.961 MQT: tasmota/discovery/246F2813D624/sensors = {"sn":{"Time":"2021-12-08T18:56:46","ESP32":{"Temperature":47.8},"TempUnit":"C"},"ver":1} (retained)
18:56:56.643 WIF: Checking connection...`

sfromis, 2021-12-08:
Better wrap text blocks in triple backtick, before and after the whole block. What you posted there is hardly readable.

ascillato2, 2021-12-08:
done :)

Jason2866, 2021-12-08:
Not a OTA `tasmota32-webcam_v1010.bin` upload via file.

sfromis, 2021-12-08:
Something bad is happening.....
`{"Info3":{"RestartReason":{"Exception":7,"Reason":"PCValue","EPC":"4016a6c8","EXCVADDR":"00000000","CallChain":["40085607","40082ced",.........`

helmar74, 2021-12-08:
Strange behaviour ;-)
I tried the OTA update nearly 10 times, after saving settings and trying again it worked.
Thank you for your help

closed 2021-12-08 by helmar74