arendst/Tasmota #13963: PowerDelta detecting negative values (alternating) and therefore spamming MQTT

niklasfink, 2021-12-08:
### PROBLEM DESCRIPTION
When using PowerDelta, an excessive amount of MQTT messages are being posted, as PowerDelta seems to cycle between positive and negative values. (Which can be recognized by the MARGINS message.)

When setting `PowerDelta 50`, even though the Power value only cycles around 100-101, MQTT messages are being sent. (See example in Screenshot section)

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): [Nous Smart WIFI Socket A1](https://nous.technology/product/nous-smart-wi-fi-socket-a1.html) (same as Gosund SP111)
- [X] Tasmota binary firmware version number used: tasmota 10.0.0 (just updated via OTA)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: tuya-convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
18:28:26.971 MQT: stat/tasmota_173B59/RESULT = {"NAME":"Gosund SP111 v1.1","GPIO":[320,0,576,0,2656,2720,0,0,2624,32,0,224,0,0],"FLAG":0,"BASE":45}
18:28:27.197 MQT: stat/tasmota_173B59/RESULT = {"Module":{"0":"Gosund SP111 v1.1"}}
18:28:27.453 MQT: stat/tasmota_173B59/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
no rules in place

```
- [X] Provide the output of this command: `Status 0`:
```lua
18:25:41.523 CMD: Status 0
18:25:41.530 MQT: stat/tasmota_173B59/STATUS = {"Status":{"Module":0,"DeviceName":"nous01","FriendlyName":["nous01"],"Topic":"tasmota_173B59","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:25:41.561 MQT: stat/tasmota_173B59/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T04:36:13","StartupUTC":"2021-12-08T12:49:28","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2021-12-07T22:17:55","SaveCount":65,"SaveAddress":"F7000"}}
18:25:41.599 MQT: stat/tasmota_173B59/STATUS2 = {"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021-10-19T08:23:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"423/699"}}
18:25:41.618 MQT: stat/tasmota_173B59/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wifi_24ghz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00004000"]}}
18:25:41.651 MQT: stat/tasmota_173B59/STATUS4 = {"StatusMEM":{"ProgramSize":612,"Free":388,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:25:41.685 MQT: stat/tasmota_173B59/STATUS5 = {"StatusNET":{"Hostname":"nous01","IPAddress":"redacted","Gateway":"redacted","Subnetmask":"255.255.255.0","DNSServer1":"redacted","DNSServer2":"192.168.0.1","Mac":"redacted","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:25:41.708 MQT: stat/tasmota_173B59/STATUS6 = {"StatusMQT":{"MqttHost":"redacted","MqttPort":1883,"MqttClientMask":"nous01","MqttClient":"nous01","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:25:41.734 MQT: stat/tasmota_173B59/STATUS7 = {"StatusTIM":{"UTC":"2021-12-08T17:25:41","Local":"2021-12-08T18:25:41","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:30","Sunset":"16:53"}}
18:25:41.753 MQT: stat/tasmota_173B59/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:25:41.772 MQT: stat/tasmota_173B59/STATUS10 = {"StatusSNS":{"Time":"2021-12-08T18:25:41","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":0.988,"Yesterday":0.000,"Today":0.988,"Power":116,"ApparentPower":116,"ReactivePower": 0,"Factor":1.00,"Voltage":231,"Current":0.503}}}
18:25:41.796 MQT: stat/tasmota_173B59/STATUS11 = {"StatusSTS":{"Time":"2021-12-08T18:25:41","Uptime":"0T04:36:13","UptimeSec":16573,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"redacted","BSSId":"24:5A:4C:5D:49:D0","Channel":6,"Mode":"11n","RSSI":24,"Signal":-88,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua

```

### TO REPRODUCE
Happened to me "out of the box" - see Screenshots area.

### EXPECTED BEHAVIOUR

Expected behavior would be, that the negative values wouldn't be happening and when I set PowerDelta 50, it only emits an MQTT message when a change of power of 50% is detected. Example in watts 100 -> 150 = emit message. 100 -> 120 = do not emit message.

### SCREENSHOTS

```lua
18:17:40.307 CMD: PowerDelta 50
18:17:40.314 MQT: stat/tasmota_173B59/RESULT = {"PowerDelta1":50}
18:17:46.389 MQT: tele/tasmota_173B59/MARGINS = {"MARGINS":{"PowerDelta":-119}}
18:17:46.398 MQT: tele/tasmota_173B59/SENSOR = {"Time":"2021-12-08T18:17:46","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":0.957,"Yesterday":0.000,"Today":0.957,"Power":121,"ApparentPower":121,"ReactivePower": 0,"Factor":1.00,"Voltage":230,"Current":0.525}}
18:17:47.388 MQT: tele/tasmota_173B59/MARGINS = {"MARGINS":{"PowerDelta":119}}
18:17:47.397 MQT: tele/tasmota_173B59/SENSOR = {"Time":"2021-12-08T18:17:47","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":0.957,"Yesterday":0.000,"Today":0.957,"Power":120,"ApparentPower":120,"ReactivePower": 0,"Factor":1.00,"Voltage":230,"Current":0.523}}
18:17:58.443 MQT: tele/tasmota_173B59/MARGINS = {"MARGINS":{"PowerDelta":-118}}
18:17:58.452 MQT: tele/tasmota_173B59/SENSOR = {"Time":"2021-12-08T18:17:58","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":0.958,"Yesterday":0.000,"Today":0.958,"Power":122,"ApparentPower":122,"ReactivePower": 0,"Factor":1.00,"Voltage":231,"Current":0.527}}
18:17:59.426 MQT: tele/tasmota_173B59/MARGINS = {"MARGINS":{"PowerDelta":120}}
18:17:59.435 MQT: tele/tasmota_173B59/SENSOR = {"Time":"2021-12-08T18:17:59","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":0.958,"Yesterday":0.000,"Today":0.958,"Power":122,"ApparentPower":122,"ReactivePower": 0,"Factor":1.00,"Voltage":231,"Current":0.527}}
18:18:03.386 MQT: tele/tasmota_173B59/MARGINS = {"MARGINS":{"PowerDelta":-119}}
18:18:03.393 MQT: tele/tasmota_173B59/SENSOR = {"Time":"2021-12-08T18:18:03","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":0.958,"Yesterday":0.000,"Today":0.958,"Power":120,"ApparentPower":120,"ReactivePower": 0,"Factor":1.00,"Voltage":230,"Current":0.522}}
```
... and this goes on

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-12-09:
Alas, I cannot reproduce. Using a Sonoff Pow with a 25W bulb load switched On and Off it behaves as expected:

```
13:52:09.511 MQT: tele/pow2/SENSOR = {"Time":"2021-12-09T13:52:09","ENERGY":{"TotalStartTime":"2021-09-29T00:00:00","Total":49.104,"TotalTariff":[14.006,111.156],"Yesterday":0.137,"Today":0.078,"Period":0.09,"Power":5.40,"ApparentPower":13.91,"ReactivePower":12.80,"Factor":0.39,"Voltage":228.0,"Current":0.061}}
13:52:13.432 CMD: powerdelta 50
13:52:13.439 MQT: stat/pow2/RESULT = {"PowerDelta1":50}
13:52:27.500 MQT: tele/pow2/MARGINS = {"MARGINS":{"PowerDelta":26}}
13:52:27.510 MQT: tele/pow2/SENSOR = {"Time":"2021-12-09T13:52:27","ENERGY":{"TotalStartTime":"2021-09-29T00:00:00","Total":49.104,"TotalTariff":[14.006,111.156],"Yesterday":0.137,"Today":0.078,"Power":21.96,"ApparentPower":21.96,"ReactivePower":0.00,"Factor":1.00,"Voltage":228.8,"Current":0.096}}
13:52:43.488 MQT: tele/pow2/MARGINS = {"MARGINS":{"PowerDelta":-26}}
13:52:43.497 MQT: tele/pow2/SENSOR = {"Time":"2021-12-09T13:52:43","ENERGY":{"TotalStartTime":"2021-09-29T00:00:00","Total":49.104,"TotalTariff":[14.006,111.156],"Yesterday":0.137,"Today":0.078,"Power":5.30,"ApparentPower":11.92,"ReactivePower":10.70,"Factor":0.44,"Voltage":229.2,"Current":0.052}}
```

arendst, 2021-12-09:
What kind of load do you use?

niklasfink, 2021-12-09:
I'm using a multi-socket which has a monitor, light and laptop attached. Just for comparison, this is the outcome when I set PowerDelta 110:

```lua
14:07:18.121 MQT: tele/tasmota_173B59/MARGINS = {"MARGINS":{"PowerDelta":12}}
14:07:18.130 MQT: tele/tasmota_173B59/SENSOR = {"Time":"2021-12-09T14:07:18","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":3.195,"Yesterday":1.740,"Today":1.455,"Power":103,"ApparentPower":103,"ReactivePower": 0,"Factor":1.00,"Voltage":231,"Current":0.445}}
14:07:20.123 MQT: tele/tasmota_173B59/MARGINS = {"MARGINS":{"PowerDelta":-11}}
14:07:20.133 MQT: tele/tasmota_173B59/SENSOR = {"Time":"2021-12-09T14:07:20","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":3.196,"Yesterday":1.740,"Today":1.456,"Power":103,"ApparentPower":103,"ReactivePower": 0,"Factor":1.00,"Voltage":231,"Current":0.448}}
14:07:29.087 MQT: tele/tasmota_173B59/MARGINS = {"MARGINS":{"PowerDelta":11}}
14:07:29.096 MQT: tele/tasmota_173B59/SENSOR = {"Time":"2021-12-09T14:07:29","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":3.196,"Yesterday":1.740,"Today":1.456,"Power":107,"ApparentPower":107,"ReactivePower": 0,"Factor":1.00,"Voltage":231,"Current":0.463}}
14:08:42.128 MQT: tele/tasmota_173B59/MARGINS = {"MARGINS":{"PowerDelta":16}}
14:08:42.135 MQT: tele/tasmota_173B59/SENSOR = {"Time":"2021-12-09T14:08:42","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":3.200,"Yesterday":1.740,"Today":1.460,"Power":109,"ApparentPower":109,"ReactivePower": 0,"Factor":1.00,"Voltage":231,"Current":0.471}}
14:08:43.129 MQT: tele/tasmota_173B59/MARGINS = {"MARGINS":{"PowerDelta":-13}}
14:08:43.138 MQT: tele/tasmota_173B59/SENSOR = {"Time":"2021-12-09T14:08:43","ENERGY":{"TotalStartTime":"2021-12-07T22:17:55","Total":3.200,"Yesterday":1.740,"Today":1.460,"Power":109,"ApparentPower":109,"ReactivePower": 0,"Factor":1.00,"Voltage":231,"Current":0.471}}
```

Again, even though I would expect that an MQTT message is only emitted when the delta of the power variable has changed by 10 or more, it also emits a message with much smaller power changes (or even no visible changes). So I guess my understanding of PowerDelta is wrong and it does not use the absolute values of the "Power" variable? 

Is there a way to make it reliably work as in the following example?

PowerDelta 110
Power: 50 -> 55 (no message)
Power: 50 -> 61 (emit message)

I've read something that the last 3s are being looked at for the delta comparison. If that's the case, why do I see a very steady power display in the web view which is only sometimes jumping by very few watts?

arendst, 2021-12-09:
Powerdelta works as you expect and can be seen above at my first respond.

What is strange from your results is that with your reported load I cannot believe you have a powerfactor of 1. Your load is not 100% resistive so expect at least a different powerfactor. Considering this it might well be that at every second powerdelta poll time there might be a difference in power usage detected.

Retry with the resistive bulb (no led light) alone and see what happens.

niklasfink, 2021-12-09:
Seems likely, that it is wrongly calibrated then. I did the calibration with a 9W and 7W LED bulb. Unfortunately, I don't own an incandescent bulb anymore. All other resistive loads, such as a toaster or a hairdryer only have power ranges.

Is there another way to verify the calibration? I do have a multimeter or an out-of-the-box calibrated Xiaomi Zigbee power socket.

This is me toggling the hairdryer between high and low heat:
```lua
16:07:00.581 CMD: PowerDelta 300
16:07:00.587 MQT: stat/tasmota_5CCBDB/RESULT = {"PowerDelta1":300}
16:07:11.780 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":332}}
16:07:11.788 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T16:07:11","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.039,"Yesterday":0.000,"Today":0.039,"Power":117,"ApparentPower":117,"ReactivePower": 0,"Factor":1.00,"Voltage":228,"Current":0.514}}
16:07:12.780 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":341}}
16:07:12.788 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T16:07:12","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.040,"Yesterday":0.000,"Today":0.040,"Power":357,"ApparentPower":357,"ReactivePower": 0,"Factor":1.00,"Voltage":228,"Current":1.562}}
16:07:28.762 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":-675}}
16:07:28.768 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T16:07:28","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.044,"Yesterday":0.000,"Today":0.044,"Power":111,"ApparentPower":111,"ReactivePower": 0,"Factor":1.00,"Voltage":228,"Current":0.486}}
16:07:41.781 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":681}}
16:07:41.789 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T16:07:41","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.045,"Yesterday":0.000,"Today":0.045,"Power":118,"ApparentPower":118,"ReactivePower": 0,"Factor":1.00,"Voltage":229,"Current":0.514}}
16:07:58.752 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":-675}}
16:07:58.762 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T16:07:58","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.049,"Yesterday":0.000,"Today":0.049,"Power":290,"ApparentPower":360,"ReactivePower":214,"Factor":0.80,"Voltage":229,"Current":1.573}}
16:08:12.778 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":273}}
16:08:12.786 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T16:08:12","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.051,"Yesterday":0.000,"Today":0.051,"Power":117,"ApparentPower":117,"ReactivePower": 0,"Factor":1.00,"Voltage":230,"Current":0.511}}
16:08:13.776 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":402}}
16:08:13.785 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T16:08:13","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.051,"Yesterday":0.000,"Today":0.051,"Power":361,"ApparentPower":361,"ReactivePower": 0,"Factor":1.00,"Voltage":230,"Current":1.574}}
16:08:24.764 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":-676}}
16:08:24.772 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T16:08:24","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.054,"Yesterday":0.000,"Today":0.054,"Power":288,"ApparentPower":297,"ReactivePower":71,"Factor":0.97,"Voltage":228,"Current":1.304}}
```

sfromis, 2021-12-09:
You **must** use a purely resistive load for calibration, which is not a LED bulb. If you have a variable resistive load available, you can verify using a "good" energy meter. I'm doing that, as I'd not trust the watts printed on any device to be exact. One reason for variations is that it may simply depend on voltage variations. While a hairdryer is not the worst, it is not a purely resistive load due to the fan, and some models may also have a thermostat fundction.

Jason2866, 2021-12-09:
A toaster is pure resistive load. A high load (like a toaster) is in my experience better than a 60W classic bulb. What do you mean with "the toaster has power ranges"?

niklasfink, 2021-12-09:
I'm not sure how all the people with a Tasmota flashed smart socket calibrate their devices properly, but I finally made it via a high risk experimental assembly, which I do not recommend to do.

I took out my best multimeter from Aliexpress, even with true rms (which I learned is good) and started measuring the ampere. This is how it looked. (I did not touch the cables during power flow 🙂)
[image]

And I ended up with perfect PowerDelta reportings:
```lua
19:39:13.360 MQT: stat/tasmota_5CCBDB/RESULT = {"PowerDelta1":103}
19:39:23.323 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta": 5}}
19:39:23.331 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T19:39:23","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.057,"Yesterday":0.000,"Today":0.057,"Power":74,"ApparentPower":86,"ReactivePower":44,"Factor":0.86,"Voltage":233,"Current":0.369}}
19:39:42.314 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":-6}}
19:39:42.325 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T19:39:42","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.057,"Yesterday":0.000,"Today":0.057,"Power":69,"ApparentPower":86,"ReactivePower":52,"Factor":0.80,"Voltage":233,"Current":0.369}}
19:40:07.328 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":-35}}
19:40:07.336 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T19:40:07","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.058,"Yesterday":0.000,"Today":0.058,"Power":34,"ApparentPower":85,"ReactivePower":78,"Factor":0.39,"Voltage":230,"Current":0.370}}
19:40:08.329 MQT: tele/tasmota_5CCBDB/MARGINS = {"MARGINS":{"PowerDelta":-27}}
19:40:08.337 MQT: tele/tasmota_5CCBDB/SENSOR = {"Time":"2021-12-09T19:40:08","ENERGY":{"TotalStartTime":"2021-12-07T22:48:07","Total":0.058,"Yesterday":0.000,"Today":0.058,"Power": 7,"ApparentPower":85,"ReactivePower":85,"Factor":0.08,"Voltage":230,"Current":0.370}}
```

Thanks for the help!

closed 2021-12-09 by niklasfink