arendst/Tasmota #13972: Sonoff rf bridge learning but not sending

atv2016, 2021-12-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I see all 433mhz messages and codes. When I put it in learning mode data it says learning complete and it beeps but data is always 000000. When I save the rfcode manually to rfkey1 it is saved.

when sending rfkey1 nothing happens. I include rfsync rflow and rfhigh in the backlog command.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____sonoff rf bridge 
- [ ] Tasmota binary firmware version number used: _____latest
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

atv2016, 2021-12-09:
how do i disable weblog again?

atv2016, 2021-12-09:
It seems even when i tell the values to tasmota data is still 0.

First i use backlog command, then i use rfkey1 with weblog set to 4 but data field not reflecting that..

:57:30.610 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 62, Data 'rfsync 12440; rflow 420; rfhigh 1210; rfcode #B9D671; rfkey1 4'
15:57:30.633 SRC: Backlog
15:57:30.635 CMD: Grp 0, Cmnd 'RFSYNC', Idx 1, Len 5, Data '12440'
15:57:30.642 MQT: stat/tasmota_5AF452/RESULT = {"RfSync":12440}
15:57:30.849 SRC: Backlog
15:57:30.852 CMD: Grp 0, Cmnd 'RFLOW', Idx 1, Len 3, Data '420'
15:57:30.857 MQT: stat/tasmota_5AF452/RESULT = {"RfLow":420}
15:57:31.099 SRC: Backlog
15:57:31.102 CMD: Grp 0, Cmnd 'RFHIGH', Idx 1, Len 4, Data '1210'
15:57:31.107 MQT: stat/tasmota_5AF452/RESULT = {"RfHigh":1210}
15:57:31.349 SRC: Backlog
15:57:31.352 CMD: Grp 0, Cmnd 'RFCODE', Idx 1, Len 7, Data '#B9D671'
15:57:31.364 MQT: stat/tasmota_5AF452/RESULT = {"RfCode":"#B9D671"}
15:57:31.602 SRC: Backlog
15:57:31.605 CMD: Grp 0, Cmnd 'RFKEY', Idx 1, Len 1, Data '4'
15:57:31.610 MQT: stat/tasmota_5AF452/RESULT = {"RfKey1":"Saved"}
15:57:31.835 CFG: Saved to flash at F8, Count 62, Bytes 4096
15:57:31.841 DMP: A0 55
15:57:35.211 DMP: A4 34 26 01 A4 04 D8 E5 24 2C 55
15:57:35.216 MQT: tele/tasmota_5AF452/RESULT = {"Time":"2021-12-09T15:57:35","RfReceived":{"Sync":13350,"Low":420,"High":1240,"Data":"E5242C","RfKey":"None"}}
15:57:37.011 WIF: Checking connection...
15:57:40.649 DMP: A4 2D DC 01 AE 04 CE 72 92 16 55
15:57:40.654 MQT: tele/tasmota_5AF452/RESULT = {"Time":"2021-12-09T15:57:40","RfReceived":{"Sync":11740,"Low":430,"High":1230,"Data":"729216","RfKey":"None"}}
15:57:42.356 CMD: RFKey1
15:57:42.358 SRC: WebConsole from 192.168.50.211
15:57:42.360 CMD: Grp 0, Cmnd 'RFKEY', Idx 1, Len 0, Data ''
15:57:42.373 MQT: stat/tasmota_5AF452/RESULT = {"RfKey1":"Learned sent"}

atv2016, 2021-12-21:
i resolved this by saving the rfkey1 and sending it *after* i added a remote to the alarm panel. can be closed.

closed 2021-12-21 by atv2016

DavideCenc, 2022-04-22:
> i resolved this by saving the rfkey1 and sending it _after_ i added a remote to the alarm panel. can be closed.

can you explain better, i have the same problem. thank you