arendst/Tasmota #13979: Sonoff Zigbee Bridge MAC_INDIRECT_TIMEOUT issue and restarting

Taffy9325, 2021-12-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Sonoff Zigbee Bridge keeps restarting, and giving zbroute error mac indirect timeout fault.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [ x] Tasmota binary firmware version number used: 10.1.0 and  ncp-uart-sw_6.7.8_115200.ota for zigbee chip
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

00:04:34.803 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:04:35.011 RSL: RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
00:04:35.265 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:00:05:34.475 RSL: STATUS0 = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_9D8281","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:02:59","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"1970-01-01T00:00:00","SaveCount":26,"SaveAddress":"1FB000"},"StatusFWR":{"Version":"10.1.0(zbbridge)","BuildDateTime":"2021-12-08T14:47:37","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"365/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ASUSHughes",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00000000000000","00000080","00006002","00004000"]},"StatusMEM":{"ProgramSize":671,"Free":1120,"Heap":26,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","00000000"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50","Sensors":"1"},"StatusNET":{"Hostname":"tasmota-9D8281-0641","IPAddress":"192.168.1.132","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"1.1.1.1","Mac":"E8:DB:84:9D:82:81","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9D8281","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"1970-01-01T00:05:34","Local":"1970-01-01T00:05:34","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"},"StatusSNS":{"Time":"1970-01-01T00:05:34"},"StatusSTS":{"Time":"1970-01-01T00:05:34","Uptime":"0T00:02:59","UptimeSec":179,"Vcc":3.480,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":2543,"MqttCount":0,"Wifi":{"AP":1,"SSId":"ASUSHughes","BSSId":"E0:63:DA:15:4E:DF","Channel":13,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here: 00:00:29.436 ZIG: Resetting EZSP device
00:00:29.959 RSL: INFO1 = {"Info1":{"Module":"Sonoff ZbBridge","Version":"10.1.0(zbbridge)","FallbackTopic":"cmnd/DVES_9D8281_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:00:29.961 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-9D8281-0641","IPAddress":"192.168.1.132"}}
00:00:29.962 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
00:00:30.676 RSL: RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2}}
00:00:30.731 RSL: RESULT = {"ZbState":{"Status":55,"Version":"6.7.8.0","Protocol":8,"Stack":2}}
00:00:30.732 RSL: RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
00:00:32.174 RSL: RESULT = {"ZbState":{"Status":56,"IEEEAddr":"0x04CD15FFFEE46455","ShortAddr":"0x0000","DeviceType":1}}
00:00:32.280 ZIG: Subscribe to group 0 'ZbListen0 0'
00:00:32.334 RSL: RESULT = {"ZbState":{"Status":0,"Message":"Started"}}
00:00:32.335 ZIG: Zigbee started
00:00:32.365 ZIG: Zigbee device information found in EEPROM (1 devices - 19 bytes)
00:00:32.377 ZIG: No Zigbee device data
00:00:32.520 QPC: Reset
00:00:34.489 RSL: STATE = {"Time":"1970-01-01T00:00:34","Uptime":"0T00:00:10","UptimeSec":10,"Vcc":3.480,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"ASUSHughes","BSSId":"E0:63:DA:15:4E:DF","Channel":13,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}
00:00:35.718 RSL: SENSOR = {"ZbReceived":{"0xA996":{"Device":"0xA996","Reachable":false}}}
00:00:41.118 ZIG: {"ZbEZSPReceived":"80004296A9"}
00:00:41.120 RSL: RESULT = {"ZbRouteError":{"ShortAddr":"0xA996","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
00:00:41.225 ZIG: {"ZbEZSPReceived":"80004296A9"}
00:00:41.227 RSL: RESULT = {"ZbRouteError":{"ShortAddr":"0xA996","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
00:00:41.333 ZIG: {"ZbEZSPReceived":"80004296A9"}
00:00:41.335 RSL: RESULT = {"ZbRouteError":{"ShortAddr":"0xA996","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
00:00:46.947 RSL: RESULT = {"ZbParent":{"Device":"0x0000","Children":1,"ChildInfo":["0x00124B00226C75F8"]}}
00:00:50.362 ZIG: {"ZbEZSPReceived":"80004296A9"}
00:00:50.364 RSL: RESULT = {"ZbRouteError":{"ShortAddr":"0xA996","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
00:00:50.467 ZIG: {"ZbEZSPReceived":"80004296A9"}
00:00:50.469 RSL: RESULT = {"ZbRouteError":{"ShortAddr":"0xA996","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
00:00:50.577 ZIG: {"ZbEZSPReceived":"80004296A9"}
00:00:50.580 RSL: RESULT = {"ZbRouteError":{"ShortAddr":"0xA996","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
00:00:51.276 RSL: RESULT = {"SerialLog":{"0":{"Active":"0"}}}
00:00:51.280 RSL: RESULT = {"WebLog":4}
00:00:51.282 SRC: Backlog
[... 268 lines cut ...]
00:01:34.612 ZIG: {"ZbEZSPReceived":"80004296A9"}
00:01:34.613 RSL: RESULT = {"ZbRouteError":{"ShortAddr":"0xA996","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
00:01:34.616 ZIG: {"ZbEZSPReceived2":"7A21900180004296A9"}
00:01:34.616 ZIG: ZbEZSPSentRaw 80
00:01:34.617 ZIG: {"ZbEZSPReceived":"80004296A9"}
00:01:34.619 RSL: RESULT = {"ZbRouteError":{"ShortAddr":"0xA996","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
00:01:34.622 ZIG: {"ZbEZSPReceived2":"0A2190013F000096A904010600010140010000CC016600"}
00:01:34.623 ZIG: ZbEZSPSentRaw 81
00:01:34.624 ZIG: {"ZbEZSPReceived":"3F000096A904010600010140010000CC016600"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_ happening all the time

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._ show data from connected zigbee devices without rebooting every 60 seconds

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-12-09:
Hi, MAC_INDIRECT_TIMEOUT means that your zigbee devices are not connected to your coordinator and you need to pair them again.

Your status 0 is incomplete. Please, do it again. the command is `STATUS 0`, with the space in between.

Jason2866, 2021-12-10:
As long the devicedoes not get Time via NTP it will not work correctly.

ascillato2, 2021-12-12:
Hi, Any news on this?

Taffy9325, 2021-12-12:
> Hi, Any news on this?

Good evening, so I’ve been working on this all weekend, turns out the Zigbee chip was the issue. Re flashed the chip with previous version and works perfectly now.

closed 2021-12-12 by ascillato2