arendst/Tasmota #13987: MQTT Server Problem With Topic

mochoandre, 2021-12-10:
### PROBLEM DESCRIPTION
I Have updated one of my nodemcu with latest 10.1.0 and start to see tasmota publishing mqtt messages to default %topic% instead of the expected my topic. Reinserted my topic but tasmota always revert to default topic.

Made a configuration reset and as soon as I insert ip of the mqtt server tasmota reverts to default topic. If I do not insert mqtt server tasmota saves the topic and it survives reboots.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: tasmota 10.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: tasmota firmware upgrade
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
4:46:37.354 MQT: stat/tasmota_EA7647/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
04:46:37.581 MQT: stat/tasmota_EA7647/RESULT = {"Module":{"18":"Generic"}}
04:46:37.834 MQT: stat/tasmota_EA7647/RESULT = {"GPIO0":{"226":"Relay3"},"GPIO1":{"0":"None"},"GPIO2":{"227":"Relay4"},"GPIO3":{"0":"None"},"GPIO4":{"225":"Relay2"},"GPIO5":{"224":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"32":"Button1"},"GPIO13":{"160":"Switch1"},"GPIO14":{"35":"Button4"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
04:50:02.995 MQT: stat/tasmota_EA7647/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":135,"Free":376,"Rules":"on Switch1#state=1 do Backlog Publish stat/%topic%/PIR1 ON; RuleTimer1 30 endon on Rules#Timer=1 do Publish stat/%topic%/PIR1 OFF endon"}}

04:50:03.219 MQT: stat/tasmota_EA7647/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
4:52:14.278 CMD: Status 0
04:52:14.286 MQT: stat/tasmota_EA7647/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"tasmota_EA7647","ButtonTopic":"0","Power":12,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
04:52:14.302 MQT: stat/tasmota_EA7647/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:17:35","StartupUTC":"2021-12-10T03:34:39","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2021-12-10T04:04:04","SaveCount":48,"SaveAddress":"F4000"}}
04:52:14.319 MQT: stat/tasmota_EA7647/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"367/699"}}
04:52:14.333 MQT: stat/tasmota_EA7647/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["homedeviot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
04:52:14.354 MQT: stat/tasmota_EA7647/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
04:52:14.369 MQT: stat/tasmota_EA7647/STATUS5 = {"StatusNET":{"Hostname":"PCSala","IPAddress":"10.10.20.151","Gateway":"10.10.20.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.20.1","DNSServer2":"0.0.0.0","Mac":"10:52:1C:EA:76:47","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
04:52:14.383 MQT: stat/tasmota_EA7647/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.20.2","MqttPort":1883,"MqttClientMask":"PCSala","MqttClient":"PCSala","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
04:52:14.396 MQT: stat/tasmota_EA7647/STATUS7 = {"StatusTIM":{"UTC":"2021-12-10T03:52:14","Local":"2021-12-10T04:52:14","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:32","Sunset":"16:52"}}
04:52:14.414 MQT: stat/tasmota_EA7647/STATUS10 = {"StatusSNS":{"Time":"2021-12-10T04:52:14","Switch1":"ON"}}
04:52:14.428 MQT: stat/tasmota_EA7647/STATUS11 = {"StatusSTS":{"Time":"2021-12-10T04:52:14","Uptime":"0T00:17:35","UptimeSec":1055,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"ON","Wifi":{"AP":1,"SSId":"homedeviot","BSSId":"94:83:C4:02:4D:25","Channel":9,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:03"}}}
04:52:15.359 MQT: stat/tasmota_EA7647/RESULT = {"POWER1":"OFF"}
04:52:15.363 MQT: stat/tasmota_EA7647/POWER1 = OFF

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
04:53:24.090 RUL: SWITCH1#STATE=1 performs "Backlog Publish stat/tasmota_EA7647/PIR1 ON; RuleTimer1 30"
04:53:24.101 MQT: stat/tasmota_EA7647/PIR1 = ON
04:53:24.334 MQT: stat/tasmota_EA7647/RESULT = {"T1":29,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
04:53:28.146 MQT: stat/tasmota_EA7647/RESULT = {"POWER1":"OFF"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Fill in mqtt data

### EXPECTED BEHAVIOUR
respect mqtt topic

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-12-10:
Hi,

You can't set mqtt client equals to mqtt topic. That is why is being reverted. Those values must be different.

ascillato2, 2021-12-10:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-12-10 by ascillato2

mochoandre, 2021-12-10:
No! Tryed to change and topic also reverts! I'v'e edit my post! 
****EDIT ******
It looks like if i fill in the topic configuration box with Upercase letters tasmota reverts back to default topic. If I put only lowercase letters it saves. Don't know why but in previous versions I have always used Upercase letters without problems.
*************************

In Previous Versions I've always done equal names in client and topic  without problems. Is there any technical reason why they can not be equal and keep things neat?

ascillato, 2021-12-10:
From your status 0, `"MqttClient":"PCSala",`, so Topic can't be `PCSala` too.
They must be different as explained in the docs and in older issues. Sorry.

Tasmota don't allow those to be equal since the initial versions. You may manage to introduce them by issuing the commands through MQTT in older versions.

ascillato2, 2021-12-12:
#14019

This restriction is now disabled. You can use the same string for MqttClient and topic.