arendst/Tasmota #13989: Energytotal does not set total energy to correct value

stefanbode, 2021-12-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If setting energytotal to 5000 and the today energy is already 1500 then the resulting total energy in teleinfo is 6500. Expectytion is that teleinfo reporting is 5000 on total energy.
Version 10.0.0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-12-10:
What I see in the code of `xdrv_03_energy.ino` is that the routine `ResponseCmndEnergyTotalYesterdayToday` is called by the commands to reset energy counters, and the published *output* including data for today does not mean that the internally saved value includes today too. Implicitly, EnergyTotal is at start of "today", and I do not see it as unreasonable (albeit a bit confusing) that the consumption for today is included when reporting energy data.

Jason2866, 2021-12-10:
Adding a explanation to the docs how it is designed would help.

stefanbode, 2021-12-11:
MY proposal would be to substract the today value from the entered one and set it. In this case the result is as expected

sfromis, 2021-12-11:
Subtracting would not work well, as that would make the stored value negative (not a good fit for unsigned integers), and if the next action then was to zero the today value, things would get weird.

arendst, 2021-12-11:
Agree with @sfromis . Another issue here is that total energy is stored in flash at midnight and all energy today is just added to it until again midnight.

In short, it's as designed. Just look at it at another angle and it might make sense.

sfromis, 2021-12-11:
Added a short explanation to the EnergyTotal command. Feel free to improve :-)

Jason2866, 2021-12-11:
Answered from project owner.

closed 2021-12-11 by Jason2866