arendst/Tasmota #13990: ESP32 email gets task watchdog triggered

Linkenelis, 2021-12-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
when sending email,1 in 5 the esp32 watchdog gets triggered

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 TTGO T7 v1.4
- [x] Tasmota binary firmware version number used: 0x0A000000
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Platformio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```
I used email log and compiled. this is serial output of fail, about 10 secs performing the SSL/TLS handshake:
14:07:20.537 CMD: sendmail[smtp.gmail.com:465:linkenelis@gmail.com:passw:linkenelis@gmail.com:to@x.com:Test]my body of the message
14:07:20.545 heap: 181636
14:07:20.548 mailsize: 103
14:07:20.550 smtp.gmail.com - 465 - linkenelis@gmail.com - passw
14:07:20.554 linkenelis@gmail.com - to@x.com - Test - my body of the message
14:07:20.563 ESP32mail: > C: ESP Mail Client v1.2.0
14:07:20.567 ESP32mail: > C: connect to SMTP server
14:07:20.571 ESP32mail: > C: host > smtp.gmail.com
14:07:20.577 ESP32mail: > C: port > 465
14:07:20.581 ESP32mail: > C: starting socket
14:07:20.585 ESP32mail: > C: connecting to Server
14:07:20.619 ESP32mail: > C: seeding the random number generator
14:07:20.625 ESP32mail: > C: setting up the SSL/TLS structure
14:07:20.629 ESP32mail: ! W: Skipping SSL Verification. INSECURE!
14:07:20.633 ESP32mail: > C: setting hostname for TLS session
14:07:20.639 ESP32mail: > C: performing the SSL/TLS handshake
E (98925) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (98925) task_wdt:  - IDLE1 (CPU 1)
E (98925) task_wdt: Tasks currently running:
E (98925) task_wdt: CPU 0: IDLE0
E (98925) task_wdt: CPU 1: loopTask
E (98925) task_wdt: Aborting.
abort() was called at PC 0x4017720f on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008e27b:0x3ffbf650 0x4008e6ad:0x3ffbf670 0x4017720f:0x3ffbf690 0x400898e1:0x3ffbf6b0 0x401d9fbf:0x3ffbc820 0x4017842b:0x3ffbc840 0x40090f79:0x3ffbc860

Rebooting...

Here it continued correctly (it seems to me that the wait after "performing the SSL/TLS handshake" was notably shorter:
14:09:58.412 ESP32mail: > C: performing the SSL/TLS handshake
14:10:05.421 ESP32mail: > C: verifying peer X.509 certificate
14:10:05.424 ESP32mail: > C: smtp server connected
14:10:05.463 ESP32mail: < S: 220 smtp.gmail.com ESMTP hq37sm1431708ejc.116 - gsmtp
14:10:05.465 ESP32mail: > C: send smtp command, HELO
14:10:05.504 ESP32mail: < S: 250-smtp.gmail.com at your service, [80.115.170.75]
14:10:05.511 ESP32mail: < S: 250-SIZE 35882577
14:10:05.515 ESP32mail: < S: 250-8BITMIME
14:10:05.521 ESP32mail: < S: 250-AUTH LOGIN PLAIN XOAUTH2 PLAIN-CLIENTTOKEN OAUTHBEARER XOAUTH
14:10:05.527 ESP32mail: < S: 250-ENHANCEDSTATUSCODES
14:10:05.532 ESP32mail: < S: 250-PIPELINING
14:10:05.537 ESP32mail: < S: 250-CHUNKING
14:10:05.542 ESP32mail: < S: 250 SMTPUTF8
14:10:05.544 ESP32mail: > C: send smtp command, AUTH PLAIN
14:10:05.546 ESP32mail: > C: linkenelis@gmail.com
14:10:05.550 ESP32mail: > C: linkenelis@gmail.com ****************
14:10:05.727 ESP32mail: < S: 235 2.7.0 Accepted
14:10:05.730 ESP32mail: > C: send Email
14:10:05.731 ESP32mail: > C: send message header
14:10:05.769 ESP32mail: < S: 250 2.1.0 OK hq37sm1431708ejc.116 - gsmtp
14:10:05.804 ESP32mail: < S: 250 2.1.5 OK hq37sm1431708ejc.116 - gsmtp
14:10:05.807 ESP32mail: > C: send message body
14:10:05.852 ESP32mail: > C: Finish the message sending
14:10:06.137 ESP32mail: < S: 250 2.0.0 OK hq37sm1431708ejc.116 - gsmtp
14:10:06.143 ESP32mail: < S: 250 2.0.0 OK hq37sm1431708ejc.116 - gsmtp
14:10:06.148 ESP32mail: < S: 250 2.0.0 OK hq37sm1431708ejc.116 - gsmtp
14:10:06.152 ESP32mail: > C: terminate the SMTP session
14:10:06.547 ESP32mail: < S: 250 2.0.0 OK hq37sm1431708ejc.116 - gsmtp
14:10:06.549 ESP32mail: > C: Message sent successfully
14:10:06.556 ESP32mail: > C: cleaning SSL connection
14:10:06.559 ESP32mail: > C: cleaning SSL connection
14:10:06.561 ESP32mail: > C: cleaning SSL connection
14:10:06.566 RSL: RESULT = {"Sendmail":"Done"}
### TO REPRODUCE
_Steps to reproduce the behavior:_
from console give the sendmail command 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-12-10:
Can you please provide the version of your build? You didn't fill the template

ascillato, 2021-12-10:
> Tasmota binary firmware version number used: 0x0A000000

? that doesn't seem Tasmota. Also your logs are not from Tasmota. Are you sure you have flashed Tasmota on your device?

ascillato2, 2021-12-10:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

Linkenelis, 2021-12-10:
const uint32_t VERSION = 0x0A000000; ?its hex, where do I find binary?
or version 10.0.0
taken from github 27-10-2021
It is Tasmota ;-)
I enabled: #define DEBUG_EMAIL_PORT in sendemail_ESP32.ino for this build. Just to get this in serial output. Normally there was nothing, just watchdog triggered...

s-hadinger, 2021-12-10:
Please use 10.1.0. We have removed the watchdog

Linkenelis, 2021-12-10:
AH, OK thanks

closed 2021-12-10 by Linkenelis

Linkenelis, 2021-12-10:
Works wonderfull
Thanks again