arendst/Tasmota #13998: EZOO2

plant-studio, 2021-12-10:
### PROBLEM DESCRIPTION
When using EZO_O2 sensor. The O2 reading in Tasmota is perfect. 

EZO Oxygen | 20.95 %

However the MQTT message reads 

tele/oxygen/SENSOR = {"Time":"2021-12-10T23:46:17","EZO":{"Oxygen":1073741456}}

When I enter:
sensor78 R

It does give me the correct output

stat/oxygen/RESULT = {"Sensor":"20.95"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): D1 Mini
- [ ] Tasmota binary firmware version number used: Tasmota 10.1.0.1
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

user_config_override.h
added the following before the last #endif
#define USE_EZOO2

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-12-10:
Driver formats the value into a string, but then formats the JSON with a placeholder for a number.
https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_78_ezoo2.ino#L36-L48

plant-studio, 2021-12-11:
Yes, that's the file I've been looking at. C is above my head. The only thing that changes every minute that it posts is the Time. Always has 1073741456 for a number.

**tele/oxygen/SENSOR = {"Time":"2021-12-10T23:46:17","EZO":{"Oxygen":1073741456}}**

sfromis, 2021-12-11:
That number looks more like the address of the string, thus it is very reasonable that it does not change. It is a simple coding error in Tasmota, fix should be easy. If you want to do it yourself, try `%s` instead of `%d` in the `ResponseAppend_P`

barbudor, 2021-12-11:
Looking into it now

barbudor, 2021-12-11:
Or wait for PR to be perged : https://github.com/arendst/Tasmota/pull/14000

ascillato2, 2021-12-11:
Closing as the bug has been fixed. Thanks for reporting. And thanks @barbudor for fixing it

closed 2021-12-11 by ascillato2