arendst/Tasmota #13999: Tasmota 10.1.0 does not Accept Upercase in topic!!!!!

mochoandre, 2021-12-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Latest version of tasmota 10.1.0 Does not accept Upercase in topic and reverts back to default topic!!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Several ones but in this example Sonoff Bridge
- [ x] Tasmota binary firmware version number used: 10.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: webinterface
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
04:05:18.390 CMD: Backlog Template; Module; GPIO 255`:
04:05:18.413 MQT: tasmota_DB5FB3/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,4704],"FLAG":0,"BASE":18}
04:05:18.640 MQT: tasmota_DB5FB3/stat/RESULT = {"Module":{"25":"Sonoff Bridge"}}
04:05:18.895 MQT: tasmota_DB5FB3/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No Rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
04:06:53.421 MQT: tasmota_DB5FB3/stat/STATUS = {"Status":{"Module":25,"DeviceName":"151SonoffBrdg","FriendlyName":["151SonoffBrdg"],"Topic":"tasmota_DB5FB3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
04:06:53.429 MQT: tasmota_DB5FB3/stat/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:14:06","StartupUTC":"2021-12-11T02:52:47","Sleep":50,"CfgHolder":4617,"BootCount":137,"BCResetTime":"2020-05-07T17:00:53","SaveCount":344,"SaveAddress":"F5000"}}
04:06:53.439 MQT: tasmota_DB5FB3/stat/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"401/699"}}
04:06:53.446 MQT: tasmota_DB5FB3/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["homedeviot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C80001000600003C5AFF000000000000","00000280","00006000","00004000"]}}
04:06:53.459 MQT: tasmota_DB5FB3/stat/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
04:06:53.464 MQT: tasmota_DB5FB3/stat/STATUS5 = {"StatusNET":{"Hostname":"151SonoffBrdg","IPAddress":"10.10.20.230","Gateway":"10.10.20.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.20.1","DNSServer2":"0.0.0.0","Mac":"DC:4F:22:DB:5F:B3","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
04:06:53.470 MQT: tasmota_DB5FB3/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.20.2","MqttPort":1883,"MqttClientMask":"XXX151SonoffBrdgC","MqttClient":"XXX151SonoffBrdgC","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
04:06:53.479 MQT: tasmota_DB5FB3/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-12-11T03:06:53","Local":"2021-12-11T04:06:53","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:33","Sunset":"16:52"}}
04:06:53.486 MQT: tasmota_DB5FB3/stat/STATUS10 = {"StatusSNS":{"Time":"2021-12-11T04:06:53"}}
04:06:53.494 MQT: tasmota_DB5FB3/stat/STATUS11 = {"StatusSTS":{"Time":"2021-12-11T04:06:53","Uptime":"0T00:14:06","UptimeSec":846,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"homedeviot","BSSId":"94:83:C4:02:4D:25","Channel":9,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
The issue is invisible on console because when you save it automatically reboots and reverts to default

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just save something with upercase in topic and it automaticaly reverts to default topic

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
It should accept Upercase as mqtt is case sensitive

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-12-11:
[image]

No issue, please close and open a discussion, or join Discord server for getting help in using Tasmota

sfromis, 2021-12-11:
Typical reason for topic reverting to default is **nothing** about uppercase or not, but due to trying to make Topic and MqttClient the same. This is against a documented stipulation.

barbudor, 2021-12-11:
We alredy told him that in 2 other issues openned this week 
But he prefers to keep openning unqualified issues instead of asking for help 
 ¯\_(ツ)_/¯

ascillato2, 2021-12-11:
Hi, 

Sorry but opening 3 times the same issue is not going to change it. Tasmota don't allow MqttClient to be the same as MQTT Topic. It is as designed. Opening the same issue over and over again won't change that. Sorry.

Closing this issue as it is **duplicated** from https://github.com/arendst/Tasmota/issues/13987
Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-12-11 by ascillato2

ascillato2, 2021-12-12:
#14019

This restriction is now disabled. You can use the same string for MqttClient and topic.