arendst/Tasmota #14012: YES!! THere is a bug in web interface and how it saves

mochoandre, 2021-12-12:
Despite moderator cutting Discussion there is a bug in tasmota web interface and how it saves mqtt data!!! Yes you can not have same name of client and topic or it reverts to default topic!!! but you need to first change client name, reboot and after you can change topic!! you can not change the the two at the same time even though they are diferent!!!!!!

AND GUESS WHAT!!!! EVEN THOUGH TILL VERSION 9.5 YOU COULD HAVE THE SAME CLIENT NAME AND TOPIC !!! AND AGAINST ME I WRITE!!! IF YOU SAVE FIRST THE CLIENT NAME AND AFTER YOU CHANGE THE TOPIC, THEY CAN STILL BE BOTH THE SAME!!! (I UNDERSTAND IT AGAINST MQTT RULES WICH I DID NOT KNOW BEFORE)

```lua
  STATUS 0 output here:

02:27:24.278 CMD: Status 0
02:27:24.286 MQT: stat/SonoffT1EU/STATUS = {"Status":{"Module":29,"DeviceName":"SonoffT1EU","FriendlyName":["Luz Escadas","Luz Hall"],"Topic":"SonoffT1EU","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
02:27:24.291 MQT: stat/SonoffT1EU/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:31","StartupUTC":"2021-12-12T01:07:53","Sleep":50,"CfgHolder":4617,"BootCount":222,"BCResetTime":"2021-02-10T23:39:17","SaveCount":5498,"SaveAddress":"F9000"}}
02:27:24.300 MQT: stat/SonoffT1EU/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"417/699"}}
02:27:24.309 MQT: stat/SonoffT1EU/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["homedeviot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
02:27:24.322 MQT: stat/SonoffT1EU/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
02:27:24.330 MQT: stat/SonoffT1EU/STATUS5 = {"StatusNET":{"Hostname":"SonoffT1EU","IPAddress":"10.10.20.31","Gateway":"10.10.20.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.20.1","DNSServer2":"192.168.9.46","Mac":"DC:4F:22:DE:9F:BD","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
02:27:24.337 MQT: stat/SonoffT1EU/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.20.2","MqttPort":1883,"MqttClientMask":"SonoffT1EU","MqttClient":"SonoffT1EU","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
02:27:24.345 MQT: stat/SonoffT1EU/STATUS7 = {"StatusTIM":{"UTC":"2021-12-12T01:27:24","Local":"2021-12-12T02:27:24","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:34","Sunset":"16:52"}}
02:27:24.353 MQT: stat/SonoffT1EU/STATUS10 = {"StatusSNS":{"Time":"2021-12-12T02:27:24"}}

```

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mochoandre, 2021-12-12:
AND GUESS WHAT MORE!!! YES YOU CAN UPDATE TILL 10.1.0 WITH SAME CLIENT NAME AND TOPIC AND TASMOTA ACCEPTS IT!!! AS SOON AS YOU CHANGE IP ADRESS OF THE SERVER, I GUESS BECAUSE TASMOTA CAN NOT COMUNICATE WITH SERVER, IT CHANGES THE TOPIC AFTER!!!

STRANGE??? YES!!! BUT IF THIS IS "BY DESIGN" ?!?!?!?!?!??

ascillato2, 2021-12-12:
Hi, 

Thanks for reporting.

Tasmota don't allow MQTTCLIENT and TOPIC to be the same as the MQTT standard recommends.

What is your proposal? What do you need Tasmota to do respect this?

------

Just for reference:
Issue duplicated from #13999, #13987 and olders. Also this is explained in the docs.

closed 2021-12-12 by Jason2866

ascillato2, 2021-12-12:
https://github.com/arendst/Tasmota/pull/14019

The restriction has been disabled for you. Please, update to latest Tasmota from development branch.