arendst/Tasmota #14013: Factory reset of a device similar to LSC MC Lights

mihsu81, 2021-12-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When the name of the template is "Deltaco SH-LRGB10M" only the green and blue lights are on, the button doesn't have any effect as well the Dimmer and HSBColor commands.
When I change the name of the template from "Deltaco SH-LRGB10M" to "LSC MC Lights" the lights blink for a couple of seconds and switch off followed by a factory reset of the device.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Deltaco SH-LRGB10M
- [x] Tasmota binary firmware version number used: 10.1.0.0 and 10.1.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

09:40:25.838 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
09:40:25.944 SRC: Backlog
09:40:25.946 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
09:40:25.953 MQT: stat/tasmota_0C2BA6/RESULT = {"NAME":"Deltaco SH-LRGB10M","GPIO":[0,0,0,0,544,32,0,0,3840,0,3872,0,0,0],"FLAG":0,"BASE":18}
09:40:26.161 SRC: Backlog
09:40:26.164 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
09:40:26.169 MQT: stat/tasmota_0C2BA6/RESULT = {"Module":{"0":"Deltaco SH-LRGB10M"}}
09:40:26.410 SRC: Backlog
09:40:26.413 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
09:40:26.423 MQT: stat/tasmota_0C2BA6/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"544":"LedLink"},"GPIO5":{"32":"Button1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"3840":"Output Hi"},"GPIO13":{"0":"None"},"GPIO14":{"3872":"Output Lo"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:41:43.395 MQT: stat/tasmota_0C2BA6/STATUS = {"Status":{"Module":0,"DeviceName":"CHRISTMAS-LIGHTS","FriendlyName":["CHRISTMAS-LIGHTS"],"Topic":"tasmota_0C2BA6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:41:43.430 MQT: stat/tasmota_0C2BA6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:01","StartupUTC":"2021-12-12T08:35:42","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2021-11-07T16:09:49","SaveCount":10,"SaveAddress":"F8000"}}
09:41:43.469 MQT: stat/tasmota_0C2BA6/STATUS2 = {"StatusFWR":{"Version":"10.1.0.1(tasmota)","BuildDateTime":"2021-12-12T09:47:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"400/699"}}
09:41:43.486 MQT: stat/tasmota_0C2BA6/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["SunnyDay",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","2805C80001000600003C5A0A000000000000","000002E8","00806000","00004000"]}}
09:41:43.520 MQT: stat/tasmota_0C2BA6/STATUS4 = {"StatusMEM":{"ProgramSize":617,"Free":384,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00020020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
09:41:43.554 MQT: stat/tasmota_0C2BA6/STATUS5 = {"StatusNET":{"Hostname":"tasmota-0C2BA6-2982","IPAddress":"192.168.45.11","Gateway":"192.168.45.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.45.1","DNSServer2":"0.0.0.0","Mac":"F4:CF:A2:0C:2B:A6","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:41:43.581 MQT: stat/tasmota_0C2BA6/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.45.100","MqttPort":1883,"MqttClientMask":"CHRISTMAS-LIGHTS","MqttClient":"CHRISTMAS-LIGHTS","MqttUser":"mihsu81","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:41:43.603 MQT: stat/tasmota_0C2BA6/STATUS7 = {"StatusTIM":{"UTC":"2021-12-12T08:41:43","Local":"2021-12-12T09:41:43","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:34","Sunset":"16:52"}}
09:41:43.626 MQT: stat/tasmota_0C2BA6/STATUS10 = {"StatusSNS":{"Time":"2021-12-12T09:41:43"}}
09:41:43.635 MQT: stat/tasmota_0C2BA6/STATUS11 = {"StatusSTS":{"Time":"2021-12-12T09:41:43","Uptime":"0T00:06:01","UptimeSec":361,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"SunnyDay","BSSId":"0C:9D:92:40:7F:40","Channel":1,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Change template from: {"NAME":"Deltaco SH-LRGB10M","GPIO":[0,0,0,0,544,32,0,0,3840,0,3872,0,0,0],"FLAG":0,"BASE":18}
to {"NAME":"LSC MC Lights","GPIO":[0,0,0,0,544,32,0,0,3840,0,3872,0,0,0],"FLAG":0,"BASE":18}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The device should work irrespective of the template name and shouldn't factory reset.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The switching power supply is: https://www.gp-electronic.com/product/smart-wifi-power/ouguilishi8-4w-wifi.html (GP-SWXXXDCXXXX-IP44(EU)-1CT,IP44 SERIES) which is slightly different than the one from the template (https://www.gp-electronic.com/product/integrated-switching-power-supply-with-built-in-controller/eugs/european-vertical-6w8-function.html).
Also the board looks slightly different GP-SW084-039 REV:00 compared to the one from the template: GP-SW084-052 REV:01
[image]
[image]
[image]
[image]

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-12-12:
Hi
The [template](https://templates.blakadder.com/lsc_smart_connect_3006824.html) for LSC MC lights say that you need to self compile your own tasmota with
```
#define USE_LSC_MCSL                             // Add support for GPE Multi color smart light as sold by Action in the Netherlands (+1k1 code)
```
Have you done that?

arendst, 2021-12-12:
This hardware will not work with template LSC MSCL. It lacks the additional microcontroller for the two wire led string. No need to try.

arendst, 2021-12-12:
From your pictures it looks like the hardware uses an H-bridge to control a two-wire led string. As far as I know Tasmota has no driver for this H-bridge (yet).

Perhaps the original "Deltaco SH-LRGB10M" template supplier can enlighten us.

mihsu81, 2021-12-12:
@barbudor  Yep. Did that.
@arendst You're right. I had a closer look and the microcontroller is missing on my version of the board. I'm more than willing to help with the driver for the H-bridge by testing on my device.
Thanks a lot. 😊

mihsu81, 2021-12-12:
If it helps, "3840 Output Hi" on GPIO12 turns on the Green and Blue LEDs and putting "3840 Output Hi" on GPIO14 turns on the Red and Yellow LEDs. Can't have both GPIOs set to Output Hi, as that will factory reset the TYWE2S after the LEDs start blinking alternatively for a couple of seconds.

barbudor, 2021-12-12:
I'm not sure what you expect from Tasmota but the below should make the leds blink
GPIO12 : Relay 1
GPIO13 : Relay 2
GPIO5 : Button 1
Commands:
```
Interlock 1,2
Interlock 1
Rule1 on power1#state=1 do backlog var1 2; ruletimer1 3 endon on power2#state=1 do backlog var1 1; ruletimer1 3 endon on rules#timer=1 do power%var1% 1 endon
Rule1 1
```

mihsu81, 2021-12-12:
Thanks @barbudor. I was looking at getting more or less the same light modes as with the tuya binary.
By using relays i cannot have all 4 LED types on at the same time and dim them.

barbudor, 2021-12-12:
Tasmota light schemes do not support such kind of leds as of today
And with regards to dimming, it would also requires writing a driver for that kind of led driver.
It would need someone having such kind of device and wanting to spend time in writing a driver for something which is quite limited vs addressable leds

ascillato2, 2021-12-12:
Moving to Ideas Discussions as this is a Feature Request.