arendst/Tasmota #14021: Shouldn't PowerOnState default to 3 ?

jerome83136, 2021-12-12:
Hello all,

I'm here to get some advices on how to resolve the problem I'm facing with my Athom SW113EU switches.

I'm using these switches to control my roller shutters.
The motor of these shutters requires a single brief impulsion on its black wire to make them going up, down and stopping.

I got it working by using this configuration:
- PowerOnState: 0  --> force all relays to "OFF" after power on
- SwitchMode1: 13  --> pushon mode (1 = on, switch off using PulseTime)
- PulseTime1: 3   --> Time to keep the relay1 "ON" after pressing it. The relays will go "OFF" after delay expiration.

My problem is: in some rare case (ie: power anomaly) Tasmota reset the switch and I loose the whole configuration.
In this case, Tasmota restarts the switch on a "default" template and the relays are activated during the power on cycle.
Because of this, my shutters are moving in a unwanted way. (opening while they must remain closed, indefinitely opening/closing, etc..)
I think it is mainly due to the "SwitchMode1: 13" setting being reset.

So I wonder how I could:
- Prevent Tasmota from reseting my configuration ?
- Force OFF the relays of my switches after a restart ? (the most important thing I require)

I've tried this:
```
- PowerOnState: 0
- SetOption65: 1
- Compiling tasmota with "#define APP_POWERON_STATE      0"
```

Unfortunately, nothing worked. After a reset, my switches template falls back to "generic" and my relays 

I wonder if I should force the "template" in my custom Tasmota firmware ? (in order to get the PowerOnState:0 working)

Example:
```
//#define MODULE                 SONOFF_BASIC      // [Module] Select default module from tasmota_template.h
#ifdef ESP8266
#define FALLBACK_MODULE        SONOFF_BASIC      // [Module2] Select default module on fast reboot where USER_MODULE is user template
//#define USER_TEMPLATE "{"NAME":"Athom SW113EU","GPIO":[576,290,0,33,225,34,0,0,32,224,289,226,288,0],"FLAG":0,"BASE":18}"  // [Template] Set JSON template
#endif  // ESP8266
```

Thank you for your help
Best regards

closed 2021-12-12 by close-issue-app[bot]

barbudor, 2021-12-12:
If your Tasmota device reverts to default settings this is may be due to one of the [recovery](https://tasmota.github.io/docs/Device-Recovery/) mechanisms.
Most probably, as you are refering to power problems, the 7 quick power cycle. 
If you power doesn't come back cleanly, Tasmota may believe you are applying this recovery procedure which can be very usefull for devices that are no more accessible after installation.
You can disable this procedure with [`SetOption65 1`](https://tasmota.github.io/docs/Commands/#setoption65)

jerome83136, 2021-12-12:
Thank you for your answer.

In fact, my device only triggers recovery mechanism when I need to remove/replace the glass plate on top of the "touch buttons" (As this manipulation generates a little physical "choc"; I think this is why the recovery mechanism triggers. I'm not really sure on this...)
So this is quite rare.

But I would like to know what would happen if I'm not at home and the device has to restart by itself.
I want to avoid any damage to my shutters motors (because they are supposed to be controlled by "push buttons" and when Tasmota restarts; I want to ensure it will preserve the "push button" config rather than falling back to the default "toggle" config)

I finally ended up with the above config:

- user_config_override.h (compiled the latest DEV version)

```
[...]
#define APP_POWERON_STATE      POWER_ALL_OFF
#ifdef ESP8266
#define USER_TEMPLATE "{\"NAME\":\"Athom SW113EU\",\"GPIO\":[158,54,0,18,22,19,0,0,17,21,53,23,52],\"FLAG\":0,\"BASE\":18}"
#define FALLBACK_MODULE        SONOFF_BASIC
#endif
#define USER_BACKLOG "module 0"
```

- tasmota config:
```
SwitchMode1: 13
PulseTime1: 3
```

Not sure its really elegant, but it seems to work for now.
I tried simulating several power interruptions and it worked as expected.
I will keep an eye on it and post improvments if relevant.

Thank you again for your help.

Regards

barbudor, 2021-12-12:
A power cut should create weird behavior. PowerOnState is here to insure that
Except that during a few hundreds of millisecondes, until tasmota software gets control of the chip, the state of the pins may nt be defined.