arendst/Tasmota #14032: Can't configure WiFi due to scattered screen after (erased) flashed to 9.5.0

Hans-Sh, 2021-12-13:
### PROBLEM DESCRIPTION
After Tasmotizing (erase & Flash) a Sonoff Basic R2 with 9.5.0 as adviced in topic 7770 the device won’t connect to Hlinc (TP Link Deco M5 mesh network) but does connect after the wifi 3 command  to HLinc and  Hlinc2 (Asus Mesh, 5Ghz disabled) using Wifi Manager. This new issue is setting the Wifi configuration to add a second SSID then the Wifi config screen is meshed up. This device had Tasmota 6.5.0 before and did connected with Hlinc or Hlinc2 without problems.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [v] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ v] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ v] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ v] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ -] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ v] Device used (e.g., Sonoff Basic): R2
- [ v] Tasmota binary firmware version number used: 9.5.0 as adviced in topic 7770
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ v] Flashing tools used: Tasmotizer 1.2
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:19:29.565 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:19:29.774 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
18:19:29.989 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: none used

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:20:15.566 CMD: status 0
18:20:15.572 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_AF718E","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:20:15.604 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:01:48","StartupUTC":"2021-12-13T17:18:27","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2021-12-13T17:04:21","SaveCount":19,"SaveAddress":"F9000"}}
18:20:15.635 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"363/699"}}
18:20:15.655 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HLinc","HLinc2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000480","00006000","00000000"]}}
18:20:15.685 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:20:15.714 RSL: STATUS5 = {"StatusNET":{"Hostname":"HLso7","IPAddress":"192.168.2.92","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"84:F3:EB:AF:71:8E","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:20:15.735 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AF718E","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:20:15.756 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-12-13T17:20:15","Local":"2021-12-13T18:20:15","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:35","Sunset":"16:52"}}
18:20:15.775 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-12-13T18:20:15"}}
18:20:15.782 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-12-13T18:20:15","Uptime":"0T00:01:48","UptimeSec":108,"Heap":27,"SleepMode":"Normal","Sleep":50,"LoadAvg":22,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"HLinc","BSSId":"28:EE:52:C1:A4:52","Channel":10,"Mode":"11g","RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:14"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:01:27.148 WIF: Connect failed with AP timeout
wifi 3
00:01:46.078 CMD: wifi 3
00:01:46.197 RSL: RESULT = {"Wifi":"ON","Mode":"11g"}
[00]r[18][02][00]lœØrß[00]Œ#[02]ânÀ[04]ˆà[0C][18]Œ[0C][1C]‚ì[1C]pŒ|Ž‚Ÿ[00]ìx[02]’ŸÇ’Üäl[1C]p[0C][18][0C]ònnä[02]Ä;ònÄ’Üä[0C][1B]Ž[0C]bŽ#l`[1B]$`[1B][04]nnàÃÜ€[0C][1C]à‚ÇÀl[1C]€ŒœÀ[0C][1C]€[0C]bÀ[04]nâãnÀäŽ’[0C][1F][0C]bÀÄ¾~ònî[03]ÄÁŒŽ[00]l`[1B]ü[12]Ü#‚nÀ[04]Žr[18][02][0E][02]nrŽ’ß;[02]ÄÀ[0C]?[0C][02]Ç;[08][02][0E]rßÛ’nÀ[04][0E]àbÜÀl`[1B]ÄžÇl[1F]œl[1F]à[0C][1F]âlœ;[18][02]ü‚n„bà
00:00:00.001 HDW: ESP8266EX
00:00:00.034 CFG: Loaded from flash at F8, Count 4
00:00:00.039 QPC: Count 1
00:00:00.047 Project tasmota Tasmota Version 9.5.0(tasmota)-2_7_4_9(2021-06-17T08:26:35)
00:00:00.199 WIF: WifiManager active for 3 minutes
00:00:00.707 HTP: Web server active on tasmota_AF718E-4494 with IP address 192.168.4.1
00:00:06.001 QPC: Reset
00:01:11.918 WIF: Connecting to AP HLinc as tasmota_AF718E-4494 ...
00:01:21.287 WIF: SSId1 HLinc: Connected - IP Address 192.168.2.92
00:01:38.685 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
[... 139 lines cut ...]
-	Seriallog 4
-	So56 0
-	So57 1
-	So56 1

In Chrome 
-	Configure
o	Wifi
	Scan:
•	First attempt Nothing visible
	Second attempt:
•	The scattered web page visual
[image]

### EXPECTED BEHAVIOUR
To see the normal Wifi Config page to add a second SSID

### SCREENSHOTS
Second attempt, slightly different screen
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-12-13:
Hi,

Seems that your issue is just not enough Wi-Fi signal. From your logs `"RSSI":44`. Below 50, you will experience disconnections.

Please, improve the signal to your device.

Hans-Sh, 2021-12-13:
Hi, thansk for your response. This configuration is up and running for years now. Why do I have this mixed up screen as of from 9.5.0? It happens with 10.0.0 too

ascillato, 2021-12-13:
Also, you can force a Wi-Fi Radio recalibration in Tasmota by issuing the command (reset 3). Your settings will remain, it will just recalibrate the wi-fi radio. This command requires that after the device restart, you must power cycle it (meaning that you have to let the device without energy for some seconds and energize it again)

ascillato, 2021-12-13:
> Hi, thanks for your response. This configuration is up and running for years now. Why do I have this mixed up screen as of from 9.5.0? It happens with 10.0.0 too

Mixed screen is that the wi-fi connection is unstable. The RSSI is too low. Please, try the Wi-Fi Radio calibration.

Hans-Sh, 2021-12-13:
This device is now 1,5 meter away from the HLinc (TP Link Deco M5). Same effect.
Status 0:
19:34:51.108 CMD: status 0
19:34:51.114 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_AF718E","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:34:51.145 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:02:30","StartupUTC":"2021-12-13T18:32:21","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2021-12-13T17:04:21","SaveCount":60,"SaveAddress":"F8000"}}
19:34:51.177 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"363/699"}}
19:34:51.196 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HLinc","HLinc2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","000004C0","00006000","00000000"]}}
19:34:51.227 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
19:34:51.256 RSL: STATUS5 = {"StatusNET":{"Hostname":"HLso7","IPAddress":"192.168.2.92","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"84:F3:EB:AF:71:8E","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:34:51.276 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AF718E","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:34:51.298 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-12-13T18:34:51","Local":"2021-12-13T19:34:51","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:35","Sunset":"16:52"}}
19:34:51.316 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-12-13T19:34:51"}}
19:34:51.324 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-12-13T19:34:51","Uptime":"0T00:02:30","UptimeSec":150,"Heap":27,"SleepMode":"Normal","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"HLinc","BSSId":"28:EE:52:C1:A4:52","Channel":10,"Mode":"11g","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:18"}}}
[image]

ascillato, 2021-12-13:
Were you able to perform reset 3 and power cycle?

Hans-Sh, 2021-12-13:
Yes indeed.

ascillato, 2021-12-13:
If so, your firmware is corrupted and you will need to erase all flash and flash again your device with latest precompiled Tasmota ( http://ota.tasmota.com/tasmota/tasmota.bin.gz )

Hans-Sh, 2021-12-13:
Did Tasmotize (and erase) with 10.2.0.1. Used Wifi Manager to connect. Seems to be more critical, had to move device next to router to connect to the TP Link mesh network. Could configure Wifi to set second SSID (ASUS Mesh, no 5Ghz) and positioned it next to this router. Did Set option 56 and 57 to ON. Reset 3 (implies reboot)  and it connected fast in N mode to the Asus mesh network.  Moved it near the TP-Link Deco M5 router and it connected to that network.  Next phase is to position it somewhere between the ASUS and the TP-Link Deco M5 mesh networks to see if Setoption57 will optimize based on best network signal.  Thanks, will keep you posted.

ascillato2, 2021-12-13:
Great to know that re flashing have solved your issue. Thanks.

closed 2021-12-13 by ascillato2