arendst/Tasmota #14035: Tasmota32 zigbee brigde does not allow any wifi connection after boot

sim0njo, 2021-12-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I flashed tasmota32-zigbeebridge FW to a DevkitC WROVER-E module with the command below. 

After rebooting the module it starts and indicates that Webserver is listening on 192.168.4.1. When I try to connect to the module's AP I see it in the list of networks, but connecting does NOT succeed, normally you see a connection being logged in the terminal window but this doesnt happen.

I tried the same using the WebInstaller but it gave the same result.

When I flash another ESP32 project from ESP-IDF it works as expected, so the DevkitC is in working order.

>>>
esptool.py --chip esp32 --port COM13 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32-zigbeebridge.bin
esptool.py v3.2
Serial port COM13
Connecting......
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c:87:ce:e3:5f:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00141fff...
Flash params set to 0x0330
Compressed 15872 bytes to 10320...
Wrote 15872 bytes (10320 compressed) at 0x00001000 in 0.4 seconds (effective 302.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 304.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 409.7 kbit/s)...
Hash of data verified.
Compressed 1250816 bytes to 845057...
Wrote 1250816 bytes (845057 compressed) at 0x00010000 in 12.2 seconds (effective 823.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
<<<
Booting the module...
>>>
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.003 HDW: ESP32-D0WD-V3 (PSRAM)
00:00:00.014 UFS: FlashFS mounted with 308 kB free
00:00:00.037 CFG: Loaded from File, Count 69
00:00:00.045 QPC: Count 1
00:00:00.103 CFG: no '*.autoconf' file found
00:00:00.109 BRY: Berry initialized, RAM used=3432
00:00:00.119 BRY: no 'preinit.be'
00:00:00.123 Project tasmota - Tasmota Version 10.1.0.1(zig_bridge32)-2_0_1_2(2021-12-12T22:12:09)
00:00:00.159 BRY: no 'autoexec.be'
00:00:00.228 WIF: WifiManager active for 3 minutes
00:00:00.986 HTP: Web server active on tasmota-E35F04-7940 with IP address 192.168.4.1
00:00:06.952 QPC: Reset
<<<

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-12-13:
As expected that it is NOT working since your board is lacking needed hardware

closed 2021-12-13 by Jason2866

sim0njo, 2021-12-13:
Ok you got me there, I should have known as I saw that part of the code last week.

Alternatively flashed Tasmota32.bin but the same result

>>>
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.003 HDW: ESP32-D0WD-V3 (PSRAM)
00:00:00.031 UFS: FlashFS mounted with 308 kB free
00:00:00.099 CFG: Loaded from File, Count 92
00:00:00.115 QPC: Count 2
00:00:00.187 BRY: Berry initialized, RAM used=3685
00:00:00.206 BRY: no 'preinit.be'
00:00:00.214 Project tasmota - Tasmota Version 10.1.0.1(tasmota)-2_0_1_1(2021-12-12T20:09:43)
00:00:00.304 BRY: no 'autoexec.be'
00:00:00.423 WIF: WifiManager active for 3 minutes
00:00:01.132 HTP: Web server active on tasmota-E35F04-7940 with IP address 192.168.4.1
00:00:06.822 QPC: Reset
<<<

barbudor, 2021-12-13:
Flashed Tasmota 10.1.0.1 with the web installer on a WROOM module:
```
ÿcãy[16]òets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.002-217/71 HDW: ESP32-D0WD 
00:00:00.034-215/71 UFS: FlashFS mounted with 308 kB free
00:00:00.097 CFG: Loaded from File, Count 6
00:00:00.114 QPC: Count 2
00:00:00.184 BRY: Berry initialized, RAM used=3685
00:00:00.202 BRY: no 'preinit.be'
00:00:00.209 Project tasmota - Tasmota Version 10.1.0.1(tasmota)-2_0_1_1(2021-12-13T18:46:27)
00:00:00.274 BRY: no 'autoexec.be'
00:00:00.477 WIF: WifiManager active for 3 minutes
00:00:01.149 HTP: Web server active on tasmota-F7B9D0-6608 with IP address 192.168.4.1
00:00:06.874 QPC: Reset
00:01:17.633 WIF: Connecting to AP BNETWLAN as tasmota-F7B9D0-6608 ...
00:01:22.973 WIF: SSId1 XXXXXXXXXXXX: Connected - IP Address 192.168.168.51
00:01:36.723 APP: Restarting
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.003 HDW: ESP32-D0WD 
00:00:00.009 UFS: FlashFS mounted with 308 kB free
00:00:00.025 CFG: Loaded from File, Count 8
00:00:00.041 QPC: Count 1
00:00:00.086 BRY: Berry initialized, RAM used=3685
00:00:00.096 BRY: no 'preinit.be'
00:00:00.103 Project tasmota - Tasmota Version 10.1.0.1(tasmota)-2_0_1_1(2021-12-13T18:46:27)
00:00:00.128 BRY: no 'autoexec.be'
00:00:00.743 WIF: Connecting to AP1 XXXXXXXXXXXXin mode 11n as tasmota-F7B9D0-6608...
00:00:06.752 WIF: Connected
00:00:06.961 HTP: Web server active on tasmota-F7B9D0-6608 with IP address 192.168.168.51
00:00:06.975 QPC: Reset
21:56:46.070 RSL: INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"10.1.0.1(tasmota)","FallbackTopic":"cmnd/DVES_F7B9D0_fb/","GroupTopic":"cmnd/tasmotas/"}}
21:56:46.084 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-F7B9D0-6608","IPAddress":"192.168.168.51"}}
21:56:46.089 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
21:56:49.964 RSL: STATE = {"Time":"2021-12-13T21:56:49","Uptime":"0T00:00:12","UptimeSec":12,"Heap":156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":127,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":34},"Wifi":{"AP":1,"SSId":"XXXXXXXXXXXX","BSSId":"88:C3:97:F2:25:EC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:07"}}
21:56:49.993 RSL: SENSOR = {"Time":"2021-12-13T21:56:49","ESP32":{"Temperature":60.0},"TempUnit":"C"}
```

I don't see any specific log when the phone connects to the wifi or open the web page, but I can configure the Wifi wihtout any problem, then Tamsota reboots as expected