arendst/Tasmota #14036: Tasmota32 does not allow any wifi connection after boot

sim0njo, 2021-12-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Since you prematurely closed 14035... here my next problem.

I alternatively flashed Tasmota32.bin to a DevkitC, I assume there is no specific hardware required, but still the same result after boot: not possible to connect to module's AP.

>>>
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.003 HDW: ESP32-D0WD-V3 (PSRAM)
00:00:00.031 UFS: FlashFS mounted with 308 kB free
00:00:00.099 CFG: Loaded from File, Count 92
00:00:00.115 QPC: Count 2
00:00:00.187 BRY: Berry initialized, RAM used=3685
00:00:00.206 BRY: no 'preinit.be'
00:00:00.214 Project tasmota - Tasmota Version 10.1.0.1(tasmota)-2_0_1_1(2021-12-12T20:09:43)
00:00:00.304 BRY: no 'autoexec.be'
00:00:00.423 WIF: WifiManager active for 3 minutes
00:00:01.132 HTP: Web server active on tasmota-E35F04-7940 with IP address 192.168.4.1
00:00:06.822 QPC: Reset
<<<

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sim0njo, 2021-12-13:
I use my laptop and see tasmota-E35F04-7940 as network available. After some trying it seems to receive an IP@ 192.168.4.2 after >10 seconds, but most of the times connecting fails. 

Trying to access 192.168.4.1 with Chrome does not result in the main page, it keeps trying and after maybe 10 seconds I see the Wifi connection to the module is lost.

I am flashing the module with esptool.py according to the instructions from the ESP32 section:
esptool.py --chip esp32 --port COM13 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32.bin

I download all bin files few hours ago, so they are all fresh.

Yesterday I also tried flashing with the WebInstaller, same result.

When I flash that same DevkitC with my own project from within ESP-IDF then all works fine, so the module is in working order.

barbudor, 2021-12-13:
Did you fully erased the flash before flashing Tasmota ?
With esptool this is a separated command to issue before flashing
With the web installer, it is a box to check

I just replied on your other (closed)) issue with a log of me flashing a WROOM module with the web installer without any problem and accessing the configuration page.

ascillato, 2021-12-13:
Are you powering the board with a power source of 1A or more? Without a good power source, these types of issues show up.

sim0njo, 2021-12-13:
Just retried the webinstaller with erase option but still the same result, not able to access the module via it's AP.

Next I tried esptool.py with --erase-all + reflash -> still same problem

sim0njo, 2021-12-13:
Powering from my laptop USB port, never had a problem with that, otherwise I would not be able to have a COM port...

ascillato, 2021-12-13:
Please, try to flash latest Tasmota from http://ota.tasmota.com/tasmota32/tasmota32.bin
Maybe your firmware file is corrupted.

sim0njo, 2021-12-13:
And how should I download from ota.tasmota.com ??? 

I re-downloaded tasmota32.bin from install/development and flash with esptool.py -> same problem remains.

Next step: use a WROOM instead -> same issue

Then, try another USB port and reflash -> problem remains

ascillato, 2021-12-14:
Can't reproduce it. Sorry.

Have you tried any other firmware or example code? Maybe your board is faulty.

Jason2866, 2021-12-14:
Powering via USB can be problematic too. Try a good! 5V Power Supply and a good USB cable.

Jason2866, 2021-12-14:
Closing since the issue is not reproducable. Sorry.

closed 2021-12-14 by Jason2866

njh, 2023-08-29:
I experienced similar problem when trying to install Tasmota 13.1.0 on a Sonoff THR316 (which uses a ESP32-D0WD-V3). I was powering the device using a little FT232R board, with a built-in 3.3v regulator.
I didn't want to connect up mains electricity to it at the same time.

It turned out that flashing was successful, it just didn't have enough power to boot Tasmota.
After disconnecting the UART, then powering it via mains, it booted fine 👍