arendst/Tasmota #14039: MHZ-19B with ESP32 show always -10°C

peterwup, 2021-12-14:
### PROBLEM DESCRIPTION
I connected a MHZ-19B to an ESP 32 in two different ways: 
1. try
TX MHZ-19B to Rx GPIO3 ESP32
RX MHZ-19B to Tx GPIO1 ESP32 
2. try
TX MHZ-19B to Rx GPI16 ESP32
RX MHZ-19B to Tx GPI17 ESP32 

In both cases the value "MHZ19B Temperature  -10.0 °C" is fix shown. 
The value "MHZ19B Carbon dioxide" is shown and toggling around between 2990ppm and 1500ppm within 20 seconds. 

</body>
</html>

00:00:00.003 HDW: ESP32-D0WDQ6 
00:00:00.052 UFS: FlashFS mounted with 304 kB free
00:00:00.130 CFG: Loaded from File, Count 24
00:00:00.146 QPC: Count 1
00:00:00.226 BRY: Berry initialized, RAM used=3849
00:00:00.247 BRY: no 'preinit.be'
00:00:00.254 Project tasmota - TasmotaCo2 Version 10.1.0(tasmota)-2_0_1_1(2021-12-08T14:50:20)
00:00:00.407 BRY: no 'autoexec.be'
00:00:00.994 WIF: Connecting to AP1

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ESP32 DEV Kit V1
- [x] Tasmota binary firmware version number used: Version 10.1.0(tasmota)-2_0_1_1(2021-12-08T14:50:20)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Esp-flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```lua
  Configuration output here:
12:02:14.294 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
12:02:14.516 RSL: RESULT = {"Module":{"0":"ESP32-DevKit"}}
12:02:14.775 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"1440":"MHZ Rx"},"GPIO17":{"1408":"MHZ Tx"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:31:06.218 CMD: Status 0
12:31:06.225 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"TasmotaCo2","FriendlyName":["TasmotaCo2"],"Topic":"tasmota_FD38C0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:31:06.262 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:06:08","StartupUTC":"2021-12-14T11:24:58","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2021-12-13T20:19:28","SaveCount":26}}
12:31:06.288 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:50:20","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"388/699"}}
12:31:06.313 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HoPiJaBox",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
12:31:06.348 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1315,"Free":1856,"Heap":158,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52,59","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,127"}}
12:31:06.396 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-FD38C0-6336","IPAddress":"192.168.0.64","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"3C:71:BF:FD:38:C0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:31:06.422 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FD38C0","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:31:06.438 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-12-14T11:31:06","Local":"2021-12-14T12:31:06","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:36","Sunset":"16:53"}}
12:31:06.469 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-12-14T12:31:06","MHZ19B":{"Model":"B","CarbonDioxide":2999,"Temperature":-10.0},"ESP32":{"Temperature":36.1},"TempUnit":"C"}}
12:31:06.490 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-12-14T12:31:06","Uptime":"0T00:06:08","UptimeSec":368,"Heap":157,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":48},"Wifi":{"AP":1,"SSId":"HoPiJaBox","BSSId":"5C:49:79:C4:88:91","Channel":6,"Mode":"11n","RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
12:32:56.532 RSL: RESULT = {"WebLog":4}
12:32:57.449 CFG: Saved, Count 27, Bytes 4096
12:33:01.842 DMP: FF 86 0B B8 1E 00 00 00 99
12:33:08.378 WIF: Checking connection...
12:33:09.800 DMP: FF 86 0B B8 1E 00 00 00 99
12:33:17.845 DMP: FF 86 0B B8 1E 00 00 00 99
12:33:25.807 DMP: FF 86 0B B8 1E 00 00 00 99
12:33:28.422 WIF: Checking connection...
12:33:33.814 DMP: FF 86 0B B8 1E 00 00 00 99

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
A valid temperature is shown and a more or less stable CO2 value

### SCREENSHOTS
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-12-14:
The displayed temperature is coherent with the value received on the serial (0x1E - 40 => -10)
The CO2 value received is 0xBB8 = 3000
I don't see that as a problem with Tasmota. 
Are you sure your MHZ is working fine ?

sfromis, 2021-12-14:
As a test, I just hooked a MH-Z19B up to an ESP32, configured `MHZ Tx` and `MHZ Rx` on TX/RX respectively, and saw no such trouble. The temperature shown was entirely consistent with other temperature sensors in the same room. CarbonDioxide value was not, possibly due to calibration lost after a long time on the shelf. 

One difference I see is that you're using gpio 16/17 instead of tx/rx. Those two pins are the last I'd use, due to the risk of interference with PSRAM polling at boot time. Also, your CO2 values seem very very high, presuming a normal indoor location.

peterwup, 2021-12-14:
Thank you for the quick reaction. As described before GPIO1/GPIO3 lead to the same result. As I only have one sensor I can not check if the sensor is damage. I will order a new one and try it again. 
Thank you.

closed 2021-12-14 by peterwup

peterwup, 2021-12-15:
Problem of power supply on my side. Connected 5V power supply to Vout instead of Vin. Now it seems to works. 
Sorry for the request.