arendst/Tasmota #14043: 14 hours after reflash with 10.1.0.1 (#14032) no web access, serial log shows Exception 9 and 0

Hans-Sh, 2021-12-14:
### PROBLEM DESCRIPTION
Approx. 14 hours after reflashing Sonoff Basic R2 with 10.1.0.1 (fix for #14032), which solved Wifi issues and connected fast and easy to Asus network in N mode, I was unable to connect to this device via Web. Serial log showed it had problems to connect to ASUS (2,4 and 5 Ghz SSID's separated) in N mode. It could connect after Wifi 3 command but it showed Execption 9 and 0. See the serial log.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [v ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [v ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [v ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- v[ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [v ] Device used (e.g., Sonoff Basic): Basic R2_
- [ ] Tasmota binary firmware version number used: 10.1.0.1
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Tasmotizer 1.2 for Windows
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:36:28.729 CMD: Backlog Template; Module; GPIO 255
14:36:28.731 SRC: WebConsole from 192.168.2.34
14:36:28.734 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
14:36:28.791 SRC: Backlog
14:36:28.793 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
14:36:28.800 MQT: HLso7/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:36:29.007 SRC: Backlog
14:36:29.010 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
14:36:29.015 MQT: HLso7/RESULT = {"Module":{"1":"Sonoff Basic"}}
14:36:29.162 DOM: idx 157, nvalue 0
14:36:29.258 SRC: Backlog
14:36:29.261 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
none

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
For test purposes now MQTT active, The Web access and Excepptions happened without MQTT was active.

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FA, Count 58
00:00:00.055 QPC: Count 1
00:00:00.064 Project tasmota - Tasmota Version 10.1.0.1(tasmota)-2_7_4_9(2021-12-12T20:09:35)
00:00:04.503 WIF: Connecting to AP2 HLinc2 Channel 6 BSSId 18:31:BF:36:A0:98 in mode 11n as HLso7...
00:00:06.525 QPC: Reset
00:00:13.800 WIF: Connect failed with AP timeout
00:00:26.768 WIF: Connect failed with AP timeout
00:00:32.124 WIF: Connecting to AP2 HLinc2 Channel 6 BSSId 18:31:BF:36:A0:98 in mode 11n as HLso7...
00:00:41.760 WIF: Connect failed with AP timeout
00:00:54.781 WIF: Connect failed with AP timeout
00:01:00.131 WIF: Connecting to AP2 HLinc2 Channel 6 BSSId 18:31:BF:36:A0:98 in mode 11n as HLso7...
00:01:09.770 WIF: Connect failed with AP timeout
00:01:22.789 WIF: Connect failed with AP timeout
00:01:28.147 WIF: Connecting to AP2 HLinc2 Channel 6 BSSId 18:31:BF:36:A0:98 in mode 11n as HLso7...
00:01:37.768 WIF: Connect failed with AP timeout
00:01:50.789 WIF: Connect failed with AP timeout
00:01:56.146 WIF: Connecting to AP2 HLinc2 Channel 6 BSSId 18:31:BF:36:A0:98 in mode 11n as HLso7...
00:02:05.770 WIF: Connect failed with AP timeout
00:02:18.800 WIF: Connect failed with AP timeout
00:02:24.104 WIF: Connecting to AP2 HLinc2 Channel 6 BSSId 18:31:BF:36:A0:98 in mode 11n as HLso7...
00:02:33.762 WIF: Connect failed with AP timeout
00:02:46.782 WIF: Connect failed with AP timeout
00:02:52.141 WIF: Connecting to AP2 HLinc2 Channel 6 BSSId 18:31:BF:36:A0:98 in mode 11n as HLso7...
reset 3
00:02:56.333 CMD: reset 3
[... 379 lines cut ...]
00:00:06.453 WIF: Sending Gratuitous ARP
00:00:06.455 NTP: Synch time...
00:00:06.480 RTC: UTC 2021-12-14T12:38:29, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
13:38:29.000 NTP: Synched
13:38:30.038 RSL: INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"10.1.0.1(tasmota)","FallbackTopic":"cmnd/DVES_AF718E_fb/","GroupTopic":"cmnd/tasmotas/"}}
13:38:30.043 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"HLso7","IPAddress":"192.168.2.90"}}
13:38:30.052 RSL: INFO3 = {"Info3":{"RestartReason":"Hardware Watchdog"}}
13:38:30.059 RSL: RESULT = {"POWER":"ON"}
13:38:30.062 RSL: POWER = ON

```

### TO REPRODUCE
Restarted, now with MQTT active

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-12-14:
Hi,

There is no need to open new issues for the same problem. You can continue in your own issue previous original issue. That is good because in separated issues, it got complex for others issues to follow your problem.

Exception 0 sometimes is because the power source is not enough or it is a very noisy one. Just for testing and so as to discard a faulty regulator in your device, please, can you try disconnecting your device from mains AC and powering the device directly to the 5Vcc rail or to the 3.3Vcc rail with a good power source?

Jason2866, 2021-12-14:
Do you have a second (spare) one? The issues you encounter looks like a defect device.
Sonoff Basic devices are known for dying young...
Most fail with a bad power supply, second most reason is faulty flash chip.

Hans-Sh, 2021-12-14:
> Hi,
> 
> There is no need to open new issues for the same problem. You can continue it in your own previous original issue. Doing that is good because in separated issues, it got complex for other users to follow your problem when doing a search.
> 
> Exception 0 sometimes is because the power source is not enough or it is a very noisy one. Just for testing and so as to discard a faulty regulator in your device, please, can you try disconnecting it from mains AC and powering directly to the 5Vcc rail or to the 3.3Vcc rail with a good power source?

Txs, The log provided is when it was(still is) connected Serial via PC USB bus AND the Sonoff it self is powered by a breadboard 3,3 Volt 1 Amp Power Supply.

Hans-Sh, 2021-12-14:
> Ok,  will setup a Basic R3 to test in same conditions. Will keep you posted.

Hans-Sh, 2021-12-14:
Have flashed (and erased) a Basic R3. Using Wifi Manager to set AP to HLinc2 (ASUS) failed. 6x power off/on to restart Wifi Manager. Set AP to HLinc (TP-Link network) and it connected easy to in in N mode. 
Using the web interface to set AP2 to HLinc2 (ASUS) as this is my preferred setup to have a Wifi fallback in case of..
Restart, Device connect to initial used AP1 (Hlinc TP-Link).  Dit set So56 and So57 to 1 to get the wished behaviour.
Device did find the strongest AP (HLinc2 ASUS) as it is 1,5 meter away from it but times out. AND KEEPS TRYING TO CONNECT TO the strongest AP found.And fails, and fails, enless.
This gave me the opportunity to fiddle around with my ASUS router:
- Disable 5 Ghz,  Device keeps trying but can't connect
- Reboot Asus, and it connects ..............  

Back to the reported problem my thought is, as I configured the 2 AP's as above (my preferred setup) It will connect to ASUS if ithe router is fresh started. If after hours it want/has to reconnect it ends up in this retry loop without success.
Wish,  Is it possible to let it connect after X tries to the next AP?

barbudor, 2021-12-14:
> Wish, Is it possible to let it connect after X tries to the other AP?

[`wificonfig 4`](https://tasmota.github.io/docs/Commands/#wificonfig)
but AFAIK you can configure the number of retries before attempting the other one

ascillato, 2021-12-14:
> Please let me know what to test /setup. I have a TP-Link 3x Deco M5 mesh and a Asus Mesh (AC2900 main/Rt68U client), several Sonoffs R2, R3, POW, HT16. I would be very pleased to be of any help in this annoying Wifi /Asus issue.

Sorry, ASUS routers have issues with the wifi routines of Espressif SDK (that is closed source). This SDK is used by the Arduino Core over which Tasmota is built.

Please, check the discussion https://github.com/arendst/Tasmota/discussions/13623 for extensive information about this and how to workaround it.

Hans-Sh, 2021-12-14:
> > Please let me know what to test /setup. I have a TP-Link 3x Deco M5 mesh and a Asus Mesh (AC2900 main/Rt68U client), several Sonoffs R2, R3, POW, HT16. I would be very pleased to be of any help in this annoying Wifi /Asus issue.
> 
> Sorry, ASUS routers have issues with the wifi routines of Espressif SDK (that are closed source). This SDK is used by the Arduino Core and Tasmota is built on top of all that.
> 
> Please, check the discussion #13623 for extensive information about this and how to workaround it.

Mmm, pitty, I do not use any Ardino, Just a lot (20+) of Sonoffs.  Is my wish (connect to other AP after x failed tries) part of Tasmota or also this SDK?

ascillato2, 2021-12-14:
Arduino is the name of the Core that uses Tasmota (Not the board). Arduino is the code that have several low functions. No matter which device you use, if you use Tasmota, under Tasmota, the functions are Arduino Core for ESP8266 board. This Core uses the radio routines provided by the manufacturer of the ESP8266 chip. Those are closed source and we can't fix them. So, all the issue with ASUS comes from the SDK (between other things is that the SDK don't have WMM that ASUS enforces as a MUST for 11n mode - other routers are relaxed about WMM requirements and that is why they work without issues).

Hans-Sh, 2021-12-14:
Thanks for the explanation.  
So my wish is for the Arduino SDK then? I think it is strange to have 2 AP's configure options and no fallback mechanism or is it how Tasmota uses the SDK?

ascillato, 2021-12-14:
What Tasmota does with WIFICONFIG being set to 4, is to try to connect to AP1. if AP1 is not available (meaning that it is not in the air), it will try to connect to AP2. Then, if Tasmota is connected to any AP and it disconnects, It will try to connect to the other configured AP.

With ASUS, the issue is that the initial handshake while trying to connect is performed but then ASUS asks for WMM and the SDK don't answer anything. So, from Tasmota side, it actually connects but in reality it doesn't.

So, the intended behavior for WIFICONFIG 4 is what you want, but the handshake (being managed in the SDK) is not correct.

You can:

* Set your router to allow b/g/n
* disable WMM
* put a different SSID to 5Ghz and disable band steering
* Set the command `wifi 3` in Tasmota to disable mode 11n
* Disable So56 and So57 and set Tasmota to connect to the router you already know has the higher strength.

Also, you can check everything discussed in #13623 for more workarounds.

Hans-Sh, 2021-12-14:
Txs for this very clear explanation. 
This helps to configure my setup in a way to keep it reliable. Will configure it timorrow and it acts as described i wiil close this issue.

Hans-Sh, 2021-12-15:
> 

Sorry @Jason2866. It wasn't a hardware issue. It was caused by the known Arduino SDK / ASUS router Problem. 
Workaround for me is to use Wifi 3 (if 9.5.0 and up) and wait for:
https://github.com/esp8266/Arduino/pull/8319 and the Tasmota development thereafter.
Thanks for all who helped to analyse my issue and to have it related to the existing mentioned Arduino SDK (2.5.2 and higher) without doubts.

Jason2866, 2021-12-15:
The PR https://github.com/esp8266/Arduino/pull/8319 does not solve the Asus issues.
See tests in #13623 So at the end it is a hardware issue (replace ASUS) when you want to solve.