arendst/Tasmota #14051: Backlog not working to return multiple values

nospam2k, 2021-12-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If I send http://192.168.0.94/cm?user=admin&password=pass&cmnd=Ssid1 I get back "{"SSId1":"ssid"}"
If I send http://ip/cm?user=admin&password=pass&cmnd=Backlog%20Ssid1%3bPassword1" I get back {}

Is there another way to get multiple values back in one web response? Using Tasmota 9.5.0?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:09:20.502 MQT: tasmota/lights/wifi-light-test/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
21:09:20.727 MQT: tasmota/lights/wifi-light-test/RESULT = {"Module":{"1":"Sonoff Basic"}}
21:09:20.981 MQT: tasmota/lights/wifi-light-test/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:10:08.947 MQT: tasmota/lights/wifi-light-test/STATUS = {"Status":{"Module":1,"DeviceName":"Wifi Light New","FriendlyName":["Wifi Light New"],"Topic":"wifi-light-test","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:10:08.956 MQT: tasmota/lights/wifi-light-test/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:27:48","StartupUTC":"2021-12-14T19:42:20","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2021-12-14T19:47:32","SaveCount":16,"SaveAddress":"F4000"}}
21:10:08.964 MQT: tasmota/lights/wifi-light-test/STATUS2 = {"StatusFWR":{"Version":"9.5.0(lite)","BuildDateTime":"2021-06-17T08:26:41","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"367/699"}}
21:10:08.971 MQT: tasmota/lights/wifi-light-test/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dafhome1","dafhome1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
21:10:08.982 MQT: tasmota/lights/wifi-light-test/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":28,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F000783","04108001","00000006","010001C0","00000000","00004004","00000000","00000020"],"Drivers":"1,2,3,4,9,10,16,20,21,37","Sensors":"3"}}
21:10:08.990 MQT: tasmota/lights/wifi-light-test/STATUS5 = {"StatusNET":{"Hostname":"wifi-light-test","IPAddress":"192.168.0.94","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"84:F3:EB:31:3B:01","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:10:08.996 MQT: tasmota/lights/wifi-light-test/STATUS6 = {"StatusMQT":{"MqttHost":"homectl","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_313B01","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:10:09.002 MQT: tasmota/lights/wifi-light-test/STATUS7 = {"StatusTIM":{"UTC":"2021-12-14T20:10:08","Local":"2021-12-14T21:10:08","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:36","Sunset":"16:53"}}
21:10:09.010 MQT: tasmota/lights/wifi-light-test/STATUS10 = {"StatusSNS":{"Time":"2021-12-14T21:10:09"}}
21:10:09.019 MQT: tasmota/lights/wifi-light-test/STATUS11 = {"StatusSTS":{"Time":"2021-12-14T21:10:09","Uptime":"0T00:27:49","UptimeSec":1669,"Vcc":3.044,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"dafhome1","BSSId":"08:62:66:BE:6B:5A","Channel":11,"Mode":"11n","RSSI":100,"Signal":-28,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
21:12:28.053 CMD: Backlog Ssid1;Password1
21:12:28.055 SRC: WebConsole from 192.168.0.253
21:12:28.058 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 15, Data 'Ssid1;Password1'
21:12:28.070 SRC: Backlog
21:12:28.073 CMD: Grp 0, Cmnd 'SSID', Idx 1, Len 0, Data ''
21:12:28.078 MQT: tasmota/lights/wifi-light-test/RESULT = {"SSId1":"dafhome1"}
21:12:28.298 SRC: Backlog
21:12:28.301 CMD: Grp 0, Cmnd 'PASSWORD', Idx 1, Len 0, Data ''
21:12:28.306 MQT: tasmota/lights/wifi-light-test/RESULT = {"Password1":"****"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
If I send http://192.168.0.94/cm?user=admin&password=pass&cmnd=Ssid1 I get back "{"SSId1":"ssid"}"
If I send http://ip/cm?user=admin&password=pass&cmnd=Backlog%20Ssid1%3bPassword1" I get back {}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Return {"SSId1":"dafhome1", "Password1":"****"}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-12-14:
> Is there another way to get multiple values back in one web response?

No. You need to query them separately.

Backlog executes each command with a delay between them. Also each command can cause more delay too. So, as you are doing an HTTP request, that one can not wait indefinitely, hence this is not supported, sorry.

ascillato2, 2021-12-14:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-12-14 by ascillato2

nospam2k, 2021-12-14:
ok, thx.