arendst/Tasmota #14067: Rainsensor Hydreon-RG-15 causes Restart Exception28

tremichl, 2021-12-16:
### PROBLEM DESCRIPTION
Rainsensor RG15 is connected to GPIO4 and GPIO5 on Wemos D1 mini or (for testing) on ESP12F. If there is no rain it is almost stable but randomly restarting. If rain is detected it starts rebooting. To reproduce:

sensor90 K
sensor90 R

you get an error, and ESP ist restarting with RestartReason":{"Exception":28,"Reason":"Exception","EPC":["4021a74f","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4000050c","4021a7a3","40286044","402265c7","401007e8","4024419e","40100cee","402441d8","40100cee","40226671","4021aafc","402341bc","402341f0","401007e8","4024421c","40101981"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP12F and Wemos D1 mini
- [x] Tasmota binary firmware version number used: 10.1.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmocompiler
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:20:06.581 MQT: stat/DVES_9BC0C9/RESULT = {"NAME":"Hydreon-RG-15","GPIO":[1,1,288,1,7424,7392,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:21:44.016 MQT: stat/DVES_9BC0C9/RESULT = {"Module":{"0":"Hydreon-RG-15"}}
12:22:44.480 MQT: stat/DVES_9BC0C9/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"288":"Led1"},"GPIO3":{"0":"None"},"GPIO4":{"7424":"HRG15 Rx"},"GPIO5":{"7392":"HRG15 Tx"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:23:48.630 CMD: status 0
12:23:48.638 MQT: stat/DVES_9BC0C9/STATUS = {"Status":{"Module":0,"DeviceName":"Regensensor","FriendlyName":["Regensensor"],"Topic":"DVES_9BC0C9","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:23:48.668 MQT: stat/DVES_9BC0C9/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Exception","Uptime":"0T00:09:14","StartupUTC":"2021-12-16T11:14:34","Sleep":50,"CfgHolder":4617,"BootCount":115,"BCResetTime":"2021-12-14T20:10:05","SaveCount":193,"SaveAddress":"3FB000"}}
12:23:48.706 MQT: stat/DVES_9BC0C9/STATUS2 = {"StatusFWR":{"Version":"10.1.0.1(TasmoCompiler-esp82664M)","BuildDateTime":"2021-12-16T11:12:01","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"382/699"}}
12:23:48.726 MQT: stat/DVES_9BC0C9/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TNet01","TNet04"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00008080","00006000","00004000"]}}
12:23:48.758 MQT: stat/DVES_9BC0C9/STATUS4 = {"StatusMEM":{"ProgramSize":591,"Free":1456,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F80458A","04260001","00000003","00000000","C000E981","00000084","00201000","00001000"],"Drivers":"1,2,4,6,8,9,10,18,19,22,24,26,27,29,30,35,37,38,45,50","Sensors":"1,2,90"}}
12:23:48.792 MQT: stat/DVES_9BC0C9/STATUS5 = {"StatusNET":{"Hostname":"DVES-9BC0C9-0201","IPAddress":"10.152.52.209","Gateway":"10.152.52.254","Subnetmask":"255.255.255.0","DNSServer1":"10.152.52.254","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:9B:C0:C9","Webserver":2,"HTTP_API":1,"WifiConfig":5,"WifiPower":17.0}}
12:23:48.816 MQT: stat/DVES_9BC0C9/STATUS6 = {"StatusMQT":{"MqttHost":"10.152.52.111","MqttPort":1993,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9BC0C9","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:23:48.840 MQT: stat/DVES_9BC0C9/STATUS7 = {"StatusTIM":{"UTC":"2021-12-16T11:23:48","Local":"2021-12-16T12:23:48","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:38","Sunset":"16:53"}}
12:23:48.860 MQT: stat/DVES_9BC0C9/STATUS10 = {"StatusSNS":{"Time":"2021-12-16T12:23:48","RG-15":{"Active":0.00, "Event":0.00, "Total":0.90, "FlowRate":0.00}}}
12:23:48.874 MQT: stat/DVES_9BC0C9/STATUS11 = {"StatusSTS":{"Time":"2021-12-16T12:23:48","Uptime":"0T00:09:14","UptimeSec":554,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"TNet04","BSSId":"34:31:C4:7C:9F:F9","Channel":1,"Mode":"11n","RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:05"}}}
12:23:48.908 MQT: stat/DVES_9BC0C9/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["4021a74f","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40103416","4021a7a3","40286044","402265c7","4010168b","401007e8","4024419e","40100cee","402441d8","40100cee","40226671","4021aafc","402341bc","402341f0","401007e8","4024421c","40101981"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8266EX
00:00:00.009 UFS: FlashFS mounted with 1992 kB free
00:00:00.075 CFG: Loaded from File, Count 194
00:00:00.081 CFG: CR 382/699, Busy 0
00:00:00.085 ROT: Mode 1
00:00:00.086 SRC: Restart
00:00:00.087 Project tasmota - Regensensor Version 10.1.0.1(TasmoCompiler-esp82664M)-2_7_4_9(2021-12-16T11:12:01)
00:00:00.194 HRG:SerialReceived = Acc  0.00 mm, EventAcc  0.00 mm, TotalAcc  0.90 mm, RInt  0.00 mmph
00:00:00.283 HRG:SerialReceived = Acc  0.00 mm, EventAcc  0.00 mm, TotalAcc  0.90 mm, RInt  0.00 mmph
00:00:00.286 RSL: SENSOR = {"Time":"1970-01-01T00:00:00","RG-15":{"Active":0.00, "Event":0.00, "Total":0.90, "FlowRate":0.00}}
00:00:00.390 WIF: Checking connection...
00:00:00.391 WIF: Attempting connection...
00:00:00.727 WIF: Connecting to AP2 TNet04 Channel 1 BSSId 34:31:C4:7C:9F:F9 in mode 11n as DVES-9BC0C9-0201...
00:00:01.753 WIF: Checking connection...
00:00:01.754 WIF: Attempting connection...
00:00:02.754 WIF: Checking connection...
00:00:02.754 WIF: Attempting connection...
[... 42 lines cut ...]
12:28:12.423 CMD: Grp 0, Cmnd 'SENSOR', Idx 90, Len 1, Data 'R'
12:28:12.937 MQT: stat/DVES_9BC0C9/RESULT = {"Command":"Error"}

and than restarting

```

### TO REPRODUCE
Sensor90 R
output OK

sensor90 K
sensor90 R
error

### EXPECTED BEHAVIOUR
Stable transmitting data

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-12-16:
Please try with hardware serial gpio1 and 3

tremichl, 2021-12-16:
Thanks Jason2866!

That was my first try:

Serial_out---TXD
Serial_in----RXD
not booting (because TXD pulled LOW from Sensor)

Serial_out---RXD GPIO3
Serial_in----TXD GPIO1
same behavior (restart) AND additional: Module Configuration is erased

arendst, 2021-12-17:
Looking at the code I see there is expected to receive serial text without a check for valid data. This is likely to cause problems on code like:
```
    char rg15_buffer[255];
    if (Rg15ReadLine(rg15_buffer)) {
        Response_P(PSTR("{\"" D_JSON_SERIALRECEIVED "\":%s\"}"), rg15_buffer);
        Rg15Process(rg15_buffer);
    }
```
If rg15_buffer is not a ascii data the "%s" will result in exception. The coder should not assume he receives valid strings but instead should always use a buffer dump in this kind of situations. I'll go over the code and fix numerous locations of these assumptions.

arendst, 2021-12-17:
Pls try latest dev and report back.

tremichl, 2021-12-17:
Have compiled development via tasmocompiler version 10.1.0.1
unfortunately it seems nothing has changed. Still randomly rebooting
Have I got the latest dev this way?

arendst, 2021-12-17:
I guess so. 

If rebooting pls show some logging with level 4 around the reboot time (use syslog).

tremichl, 2021-12-17:
That is the result after sending 
sensor90 K
sensor90 R
hope that helps

`Dec 17 17:43:43 DVES-4E80CA-0202 ESP-CMD: sensor90 K
Dec 17 17:43:43 DVES-4E80CA-0202 ESP-SRC: WebConsole from 10.152.52.62
Dec 17 17:43:43 DVES-4E80CA-0202 ESP-CMD: Grp 0, Cmnd 'SENSOR', Idx 90, Len 1, Data 'K'
Dec 17 17:43:43 DVES-4E80CA-0202 ESP-MQT: stat/DVES_4E80CA/RESULT = {"Sensor":"Done"}
Dec 17 17:43:44 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 17:43:45 DVES-4E80CA-0202 ESP-HRG: Received 'Reset D'
Dec 17 17:43:45 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/SENSOR = {"Time":"2021-12-17T17:43:44","RG-15":{"Active":0.00,"Event":0.00,"Total":0.00,"FlowRate":0.00}}
Dec 17 17:43:46 DVES-4E80CA-0202 ESP-CMD: sensor90 R
Dec 17 17:43:46 DVES-4E80CA-0202 ESP-SRC: WebConsole from 10.152.52.62
Dec 17 17:43:46 DVES-4E80CA-0202 ESP-CMD: Grp 0, Cmnd 'SENSOR', Idx 90, Len 1, Data 'R'
Dec 17 17:43:46 DVES-4E80CA-0202 ESP-MQT: stat/DVES_4E80CA/RESULT = {"Command":"Error"}
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-HDW: ESP8266EX
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-UFS: FlashFS mounted with 1992 kB free
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-CFG: Loaded from File, Count 46
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-CFG: CR 395/699, Busy 0
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-ROT: Mode 1
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-SRC: Restart
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-Project tasmota - Regensensor Version 10.1.0.1(TasmoCompiler-esp82664M)-2_7_4_9(2021-12-17T15:33:35)
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-HRG: Received 'Acc ▒#002j▒▒ EventAcc  0.00 mm, TotalAcc  1.25 mm, RInt  0.00 mmph'
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-HRG: Received 'Acc  0.00 mm, EventAcc  0.00 mm, TotalAcc  1.25 mm, RInt  0.00 mmph'
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-RSL: SENSOR = {"Time":"1970-01-01T00:00:00","RG-15":{"Active":0.00,"Event":0.00,"Total":1.25,"FlowRate":0.00}}
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-WIF: Connecting to AP1 TNet01 Channel 5 BSSId 24:5A:4C:13:C1:D5 in mode 11n as DVES-4E80CA-0202...
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-WIF: Connected
Dec 17 17:43:52 DVES-4E80CA-0202 ESP-HTP: Web server active on DVES-4E80CA-0202 with IP address 10.152.52.24
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-NTP: Synch time...
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-RTC: UTC 2021-12-17T16:43:52, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-NTP: Synched
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-MQT: Attempting connection...
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-MQT: Connected
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/LWT = Online (retained)
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-MQT: cmnd/DVES_4E80CA/POWER =
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/DVES_4E80CA/#
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/tasmotas/#
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/DVES_4E80CA_fb/#
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO1 = {"Info1":{"Module":"Hydreon-RG-15","Version":"10.1.0.1(TasmoCompiler-esp82664M)","FallbackTopic":"cmnd/DVES_4E80CA_fb/","GroupTopic":"cmnd/tasmotas/"}}
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"DVES-4E80CA-0202","IPAddress":"10.152.52.24"}}
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["402236cb","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4022371f","402ab311","40235a2b","40101e2b","40100d08","4022ff70","4010148e","4025fe10","4010148e","40235ad5","40224008","4024493c","4024496c","40100d08","4025fe54","40102121"]}}}
Dec 17 17:43:53 DVES-4E80CA-0202 ESP-TFS: File 'autoexec.bat' not found
Dec 17 17:43:58 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/STATE = {"Time":"2021-12-17T17:43:57","Uptime":"0T00:00:09","UptimeSec":9,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"Wifi":{"AP":1,"SSId":"TNet01","BSSId":"24:5A:4C:13:C1:D5","Channel":5,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
Dec 17 17:43:58 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/SENSOR = {"Time":"2021-12-17T17:43:57","RG-15":{"Active":0.00,"Event":0.00,"Total":1.25,"FlowRate":0.00}}
Dec 17 17:43:58 DVES-4E80CA-0202 ESP-APP: Boot Count 20
Dec 17 17:43:59 DVES-4E80CA-0202 ESP-CFG: Saved to flash at 3FB, Count 47, Bytes 4096
`

if you also need log of randomly reboot I'll try to catch it

arendst, 2021-12-17:
Thx. Will have a look tomorrow.

tremichl, 2021-12-17:
Thanks, no hurry. Here is a log of random restart

`Dec 17 21:38:38 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/STATE = {"Time":"2021-12-17T21:38:37","Uptime":"0T00:55:12","UptimeSec":3312,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"TNet04","BSSId":"34:31:C4:7C:9F:F9","Channel":1,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:05"}}
Dec 17 21:38:38 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/SENSOR = {"Time":"2021-12-17T21:38:37","RG-15":{"Active":0.00,"Event":0.00,"Total":1.25,"FlowRate":0.00}}
Dec 17 21:38:51 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:39:11 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:39:28 DVES-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 17 21:39:31 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:39:51 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:40:11 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:40:28 DVES-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 17 21:40:31 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:40:51 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:41:11 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:41:28 DVES-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 17 21:41:31 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:41:51 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:42:11 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:42:28 DVES-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 17 21:42:31 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:42:51 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:43:11 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-UFS: FlashFS mounted with 1992 kB free
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-CFG: Loaded from File, Count 52
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-CFG: CR 395/699, Busy 0
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-ROT: Mode 1
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-SRC: Restart
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-Project tasmota - Regensensor Version 10.1.0.1(TasmoCompiler-esp82664M)-2_7_4_9(2021-12-17T15:33:35)
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-HRG: Received 'Acc  0.00 mm, EventAcc  0.00 mm, TotalAcc  1.25 mm, RInt  0.00 mmph'
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-HRG: Received 'Acc  0.00 mm, EventAcc  0.00 mm, TotalAcc  1.25 mm, RInt  0.00 mmph'
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-RSL: SENSOR = {"Time":"1970-01-01T00:00:00","RG-15":{"Active":0.00,"Event":0.00,"Total":1.25,"FlowRate":0.00}}
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-WIF: Connecting to AP2 TNet04 Channel 1 BSSId 34:31:C4:7C:9F:F9 in mode 11n as DVES-4E80CA-0202...
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-WIF: Connected
Dec 17 21:43:23 DVES-4E80CA-0202 ESP-HTP: Web server active on DVES-4E80CA-0202 with IP address 10.152.52.24
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-NTP: Synch time...
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-RTC: UTC 2021-12-17T20:43:23, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-NTP: Synched
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-MQT: Attempting connection...
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-MQT: Connected
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/LWT = Online (retained)
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-MQT: cmnd/DVES_4E80CA/POWER =
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/DVES_4E80CA/#
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/tasmotas/#
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/DVES_4E80CA_fb/#
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO1 = {"Info1":{"Module":"Hydreon-RG-15","Version":"10.1.0.1(TasmoCompiler-esp82664M)","FallbackTopic":"cmnd/DVES_4E80CA_fb/","GroupTopic":"cmnd/tasmotas/"}}
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"DVES-4E80CA-0202","IPAddress":"10.152.52.24"}}
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["402236cb","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["402236e2","4022373d","402ab311","40235a2b","40203030","4025ed79","40101ba3","40101e2b","40100d08","4022ff70","4010148e","4025fe10","4010148e","40235ad5","40224008","4024493c","4024496c","40100d08","4025fe54","40102121"]}}}
Dec 17 21:43:24 DVES-4E80CA-0202 ESP-TFS: File 'autoexec.bat' not found
Dec 17 21:43:27 DVES-4E80CA-0202 ESP-APP: Boot Count 24
Dec 17 21:43:28 DVES-4E80CA-0202 ESP-CFG: Saved to flash at 3FB, Count 53, Bytes 4096
Dec 17 21:43:29 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/STATE = {"Time":"2021-12-17T21:43:28","Uptime":"0T00:00:11","UptimeSec":11,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"MqttCount":1,"Wifi":{"AP":2,"SSId":"TNet04","BSSId":"34:31:C4:7C:9F:F9","Channel":1,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:05"}} [... cut ...]
Dec 17 21:53:29 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/STATE = {"Time":"2021-12-17T21:53:28","Uptime":"0T00:10:11","UptimeSec":611,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"TNet04","BSSId":"34:31:C4:7C:9F:F9","Channel":1,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:05"}}
Dec 17 21:53:29 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/SENSOR = {"Time":"2021-12-17T21:53:28","RG-15":{"Active":0.00,"Event":0.00,"Total":1.25,"FlowRate":0.00}}
Dec 17 21:53:43 DVES-4E80CA-0202 ESP-WIF: Checking connection...`

arendst, 2021-12-18:
Pls retry.

tremichl, 2021-12-19:
a firs view with
sensor90 K
sensor90 R
results also in a Restart....sorry

`Dec 19 16:23:25 DVES-4E80CA-0202 ESP-APP: Boot Count 66
Dec 19 16:23:26 DVES-4E80CA-0202 ESP-CFG: Saved to flash at 3FB, Count 96, Bytes 4096
Dec 19 16:23:32 DVES-4E80CA-0202 ESP-CMD: sensor90 R
Dec 19 16:23:32 DVES-4E80CA-0202 ESP-SRC: WebConsole from 10.152.52.62
Dec 19 16:23:32 DVES-4E80CA-0202 ESP-CMD: Grp 0, Cmnd 'SENSOR', Idx 90, Len 1, Data 'R'
Dec 19 16:23:32 DVES-4E80CA-0202 ESP-MQT: stat/DVES_4E80CA/RESULT = {"SerialReceived":"Acc  0.00 mm, EventAcc  0.00 mm, TotalAcc  1.25 mm, RInt  0.00 mmph"}
Dec 19 16:23:39 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 16:23:41 DVES-4E80CA-0202 ESP-CMD: sensor90 K
Dec 19 16:23:41 DVES-4E80CA-0202 ESP-SRC: WebConsole from 10.152.52.62
Dec 19 16:23:41 DVES-4E80CA-0202 ESP-CMD: Grp 0, Cmnd 'SENSOR', Idx 90, Len 1, Data 'K'
Dec 19 16:23:41 DVES-4E80CA-0202 ESP-MQT: stat/DVES_4E80CA/RESULT = {"Sensor":"Done"}
Dec 19 16:23:43 DVES-4E80CA-0202 ESP-HRG: Received 52657365742044
Dec 19 16:23:43 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/SENSOR = {"Time":"2021-12-19T16:23:42","RG-15":{"Active":0.00,"Event":0.00,"Total":0.00,"FlowRate":0.00}}
Dec 19 16:23:45 DVES-4E80CA-0202 ESP-HRG: Received
Dec 19 16:23:45 DVES-4E80CA-0202 ESP-HRG: Received 3B2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2AFFFFFF
Dec 19 16:23:45 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/SENSOR = {"Time":"2021-12-19T16:23:44","RG-15":{"Active":0.00,"Event":0.00,"Total":0.00,"FlowRate":0.00}}
Dec 19 16:23:46 DVES-4E80CA-0202 ESP-CMD: sensor90 R
Dec 19 16:23:46 DVES-4E80CA-0202 ESP-SRC: WebConsole from 10.152.52.62
Dec 19 16:23:46 DVES-4E80CA-0202 ESP-CMD: Grp 0, Cmnd 'SENSOR', Idx 90, Len 1, Data 'R'
Dec 19 16:23:46 DVES-4E80CA-0202 ESP-MQT: stat/DVES_4E80CA/RESULT = {"Command":"Error"}
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-HDW: ESP8266EX
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-UFS: FlashFS mounted with 1992 kB free
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-CFG: Loaded from File, Count 96
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-CFG: CR 395/699, Busy 0
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-ROT: Mode 1
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-SRC: Restart
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-Project tasmota - Regensensor Version 10.1.0.1(TasmoCompiler-esp82664M)-2_7_4_9(2021-12-19T15:19:25)
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-HRG: Received 4163632020302E3030206D6D2C204576656E744163632020302E3030206D6D2C20546F74616C4163632020312E3235206D6D2C2052496E742020302E3030206D6D7068
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-HRG: Received 4163632020302E3030206D6D2C204576656E744163632020302E3030206D6D2C20546F74616C4163632020312E3235206D6D2C2052496E742020302E3030206D6D7068
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-RSL: SENSOR = {"Time":"1970-01-01T00:00:00","RG-15":{"Active":0.00,"Event":0.00,"Total":1.25,"FlowRate":0.00}}
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-WIF: Connecting to AP2 TNet04 Channel 1 BSSId 34:31:C4:7C:9F:F9 in mode 11n as DVES-4E80CA-0202...
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-WIF: Connected
Dec 19 16:23:51 DVES-4E80CA-0202 ESP-HTP: Web server active on DVES-4E80CA-0202 with IP address 10.152.52.24
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-NTP: Synch time...
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-RTC: UTC 2021-12-19T15:23:51, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-NTP: Synched
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-MQT: Attempting connection...
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-MQT: Connected
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/LWT = Online (retained)
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-MQT: cmnd/DVES_4E80CA/POWER =
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/DVES_4E80CA/#
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/tasmotas/#
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/DVES_4E80CA_fb/#
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO1 = {"Info1":{"Module":"Hydreon-RG-15","Version":"10.1.0.1(TasmoCompiler-esp82664M)","FallbackTopic":"cmnd/DVES_4E80CA_fb/","GroupTopic":"cmnd/tasmotas/"}}
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"DVES-4E80CA-0202","IPAddress":"10.152.52.24"}}
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["40223653","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["402236a7","402ab281","40235945","40100d04","4022fe90","4010148a","4025fd78","4010148a","402359f1","40223f8c","402448b4","402448e4","40100d04","4025fdbc","4010211d"]}}}
Dec 19 16:23:52 DVES-4E80CA-0202 ESP-TFS: File 'autoexec.bat' not found
Dec 19 16:23:57 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/STATE = {"Time":"2021-12-19T16:23:56","Uptime":"0T00:00:09","UptimeSec":9,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"Wifi":{"AP":2,"SSId":"TNet04","BSSId":"34:31:C4:7C:9F:F9","Channel":1,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:03"}}
[... cut ...]
Dec 19 16:24:42 DVES-4E80CA-0202 ESP-CMD: Grp 0, Cmnd 'SENSOR', Idx 90, Len 1, Data 'R'
Dec 19 16:24:42 DVES-4E80CA-0202 ESP-MQT: stat/DVES_4E80CA/RESULT = {"SerialReceived":"Acc  0.00 mm, EventAcc  0.00 mm, TotalAcc  1.25 mm, RInt  0.00 mmph"}
Dec 19 16:24:50 DVES-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 19 16:24:51 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 16:25:11 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 16:25:31 DVES-4E80CA-0202 ESP-WIF: Checking connection...
`

tremichl, 2021-12-19:
....and rain detected

`Dec 19 17:15:35 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:15:55 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:16:15 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-HDW: ESP8266EX
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-UFS: FlashFS mounted with 1992 kB free
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-CFG: Loaded from File, Count 99
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-CFG: CR 395/699, Busy 0
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-ROT: Mode 1
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-SRC: Restart
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-Project tasmota - Regensensor Version 10.1.0.1(TasmoCompiler-esp82664M)-2_7_4_9(2021-12-19T15:19:25)
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-WIF: Connecting to AP2 TNet04 Channel 1 BSSId 34:31:C4:7C:9F:F9 in mode 11n as DVES-4E80CA-0202...
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-HRG: Received 8120302E3031206D6D2C204576656E744163632020302E3738206D6D2C20546F74616C4163632020322E3033206D6D2C2052496E742020312E3832206D6D704163632020302E3030206D6D2C204576656E744163632020302E3738206D6D2C20546F74616C4163632020322E3033206D6D2C2052496E742020312E3832206D6D7068
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-RSL: SENSOR = {"Time":"1970-01-01T00:00:01","RG-15":{"Active":0.78,"Event":0.78,"Total":2.03,"FlowRate":1.82}}
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-WIF: Connected
Dec 19 17:16:23 DVES-4E80CA-0202 ESP-HTP: Web server active on DVES-4E80CA-0202 with IP address 10.152.52.24
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-HDW: ESP8266EX
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-UFS: FlashFS mounted with 1992 kB free
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-CFG: Loaded from File, Count 99
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-FRC: Some settings have been reset (2)
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-CFG: CR 395/699, Busy 0
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-ROT: Mode 1
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-SRC: Restart
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-Project tasmota - Regensensor Version 10.1.0.1(TasmoCompiler-esp82664M)-2_7_4_9(2021-12-19T15:19:25)
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-HRG: Received 41636320738282026AB5B1204576656E744163632020302E3738206D6D2C20546F74616C4163632020322E3033206D6D2C2052496E742020312E3837206D6D7068
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-HRG: Received 4163632020302E3030206D6D2C204576656E744163632020302E3738206D6D2C20546F74616C4163632020322E3033206D6D2C2052496E742020312E3837206D6D7068
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-RSL: SENSOR = {"Time":"1970-01-01T00:00:00","RG-15":{"Active":0.00,"Event":0.78,"Total":2.03,"FlowRate":1.87}}
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-WIF: Connecting to AP2 TNet04 Channel 1 BSSId 34:31:C4:7C:9F:F9 in mode 11n as DVES-4E80CA-0202...
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-WIF: Checking connection...
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-WIF: Connected
Dec 19 17:16:28 DVES-4E80CA-0202 ESP-HTP: Web server active on DVES-4E80CA-0202 with IP address 10.152.52.24
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-NTP: Synch time...
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-RTC: UTC 2021-12-19T16:16:28, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-NTP: Synched
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-MQT: Attempting connection...
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-MQT: Connected
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/LWT = Online (retained)
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-MQT: cmnd/DVES_4E80CA/POWER =
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/DVES_4E80CA/#
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/tasmotas/#
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-MQT: Subscribe to cmnd/DVES_4E80CA_fb/#
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO1 = {"Info1":{"Module":"Hydreon-RG-15","Version":"10.1.0.1(TasmoCompiler-esp82664M)","FallbackTopic":"cmnd/DVES_4E80CA_fb/","GroupTopic":"cmnd/tasmotas/"}}
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"DVES-4E80CA-0202","IPAddress":"10.152.52.24"}}
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["40223653","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4022366a","402236c5","402ab281","40235945","40261d00","40101b9f","40101e27","4020bf78","4010148a","402359f1","40223f8c","402448b4","402448e4","40100d04","4025fdbc","4010211d"]}}}
Dec 19 17:16:29 DVES-4E80CA-0202 ESP-TFS: File 'autoexec.bat' not found
Dec 19 17:16:32 DVES-4E80CA-0202 ESP-APP: Boot Count 70
[... cut ...]
Dec 19 17:16:34 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/STATE = {"Time":"2021-12-19T17:16:33","Uptime":"0T00:00:11","UptimeSec":11,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Wifi":{"AP":2,"SSId":"TNet04","BSSId":"34:31:C4:7C:9F:F9","Channel":1,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:05"}}
Dec 19 17:16:34 DVES-4E80CA-0202 ESP-MQT: tele/DVES_4E80CA/SENSOR = {"Time":"2021-12-19T17:16:33","RG-15":{"Active":0.00,"Event":0.78,"Total":2.03,"FlowRate":1.87}}
Dec 19 17:16:48 DVES-4E80CA-0202 ESP-WIF: Checking connection...
`

arendst, 2021-12-22:
To further assist you pls make a "default" Tasmota image with only the HRG-155 added.

Your current binary seems to be tailored with a larger memory model (2M) and UFS which makes debugging more difficult.

tremichl, 2021-12-22:
OK, have compiled esp8266generic with only HRG-15. Same behavior. What else can Ido?

`00:00:00.001 HDW: ESP8266EX
00:00:00.049 CFG: Loaded from flash at F4, Count 16
00:00:00.055 QPC: Count 1
00:00:00.063 Project tasmota - Tasmota Version 10.1.0.1(TasmoCompiler-esp8266generic)-2_7_4_9(2021-12-22T16:28:02)
00:00:00.262 RSL: SENSOR = {"Time":"1970-01-01T00:00:00","RG-15":{"Active":0.00,"Event":0.00,"Total":2.03,"FlowRate":0.00}}
00:00:00.703 WIF: Connecting to AP1 TNet01 Channel 5 BSSId 24:5A:4C:13:C8:3D in mode 11n as tasmota-4E80CA-0202...
00:00:06.303 QPC: Reset
00:00:11.797 WIF: Connected
00:00:12.049 HTP: Web server active on tasmota-4E80CA-0202 with IP address 10.152.52.24
17:47:51.049 MQT: Attempting connection...
17:47:51.071 MQT: Connected
17:47:51.075 MQT: tele/tasmota_4E80CA/LWT = Online (retained)
17:47:51.078 MQT: cmnd/tasmota_4E80CA/POWER = 
17:47:51.088 MQT: tele/tasmota_4E80CA/INFO1 = {"Info1":{"Module":"Generic","Version":"10.1.0.1(TasmoCompiler-esp8266generic)","FallbackTopic":"cmnd/DVES_4E80CA_fb/","GroupTopic":"cmnd/tasmotas/"}}
17:47:51.099 MQT: tele/tasmota_4E80CA/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-4E80CA-0202","IPAddress":"10.152.52.24"}}
17:47:51.120 MQT: tele/tasmota_4E80CA/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["40215ec7","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40201ca1","40215f1b","40215fa8","4010168f","4023fdc8","40101407","4010168f","4027b79a","40236c2e","40236fdd","4023fd85","40100cf2","40220a02","40216274","402738c2","402735ac","40216dfc","402735ac","40217580","402735ae","40273454","40221bb6","402078a0","4023fdc8","40101407","4010168f","40278db5","40236c2e","40236fdd","40230dac","40221db0"]}}}
17:47:55.272 MQT: tele/tasmota_4E80CA/STATE = {"Time":"2021-12-22T17:47:55","Uptime":"0T00:00:19","UptimeSec":19,"Heap":32,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"TNet01","BSSId":"24:5A:4C:13:C8:3D","Channel":5,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:13"}}
17:47:55.298 MQT: tele/tasmota_4E80CA/SENSOR = {"Time":"2021-12-22T17:47:55","RG-15":{"Active":0.00,"Event":0.00,"Total":2.03,"FlowRate":0.00}}`

arendst, 2021-12-22:
Try again with the latest dev branch.

Enable full logging (option 4).

tremichl, 2021-12-22:
Thanks!

no more exception28 but:

sensor90 K had no effect, did not erase values

after sensing rain, device was not reachable anymore. Even no AP, needed power reset  

`Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-Project tasmota - Rainsensor Version 10.1.0.1(TasmoCompiler-esp8266generic)-2_7_4_9(2021-12-22T17:29:55)
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-HRG: Received
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-HRG: Received 2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-HRG: Received 4576656E74
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-HRG: Received 535720312E30303020323032302E30372E3036
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-HRG: Received
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-HRG: Received 4163632020302E3030206D6D2C204576656E744163632020302E3030206D6D2C20546F74616C4163632020322E3033206D6D2C2052496E742020302E3030206D6D7068
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-HRG: Received 4163632020302E3030206D6D2C204576656E744163632020302E3030206D6D2C20546F74616C4163632020322E3033206D6D2C2052496E742020302E3030206D6D7068
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-WIF: Checking connection...
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-WIF: Attempting connection...
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-WIF: Connecting to AP1 TNet01 Channel 5 BSSId 24:5A:4C:13:C8:3D in mode 11n as tasmota-4E80CA-0202...
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-WIF: Checking connection...
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-WIF: Connected
Dec 22 18:58:46 tasmota-4E80CA-0202 ESP-HTP: Web server active on tasmota-4E80CA-0202 with IP address 10.152.52.24
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-NTP: Synch time...
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-RTC: UTC 2021-12-22T17:58:46, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-NTP: Synched
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-MQT: Attempting connection...
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-MQT: Connected
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-MQT: tele/tasmota_4E80CA/LWT = Online (retained)
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-MQT: cmnd/tasmota_4E80CA/POWER =
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-MQT: Subscribe to cmnd/tasmota_4E80CA/#
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-MQT: Subscribe to cmnd/tasmotas/#
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-MQT: Subscribe to cmnd/DVES_4E80CA_fb/#
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-MQT: tele/tasmota_4E80CA/INFO1 = {"Info1":{"Module":"Generic","Version":"10.1.0.1(TasmoCompiler-esp8266generic)","FallbackTopic":"cmnd/DVES_4E80CA_fb/","GroupTopic":"cmnd/tasmotas/"}}
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-MQT: tele/tasmota_4E80CA/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-4E80CA-0202","IPAddress":"10.152.52.24"}}
Dec 22 18:58:47 tasmota-4E80CA-0202 ESP-MQT: tele/tasmota_4E80CA/INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
Dec 22 18:58:50 tasmota-4E80CA-0202 ESP-QPC: Reset
Dec 22 18:58:52 tasmota-4E80CA-0202 ESP-MQT: tele/tasmota_4E80CA/STATE = {"Time":"2021-12-22T18:58:51","Uptime":"0T00:00:09","UptimeSec":9,"Heap":32,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"TNet01","BSSId":"24:5A:4C:13:C8:3D","Channel":5,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}
Dec 22 18:58:52 tasmota-4E80CA-0202 ESP-MQT: tele/tasmota_4E80CA/SENSOR = {"Time":"2021-12-22T18:58:51","RG-15":{"Active":0.00,"Event":0.00,"Total":2.03,"FlowRate":0.00}}
Dec 22 18:58:52 tasmota-4E80CA-0202 ESP-HTP: Main Menu
Dec 22 18:58:52 tasmota-4E80CA-0202 ESP-APP: Boot Count 12
Dec 22 18:58:53 tasmota-4E80CA-0202 ESP-CFG: Saved to flash at F4, Count 28, Bytes 4096
Dec 22 18:59:06 tasmota-4E80CA-0202 ESP-WIF: Checking connection...
Dec 22 18:59:15 tasmota-4E80CA-0202 ESP-HTP: Console
Dec 22 18:59:26 tasmota-4E80CA-0202 ESP-WIF: Checking connection...
Dec 22 18:59:39 tasmota-4E80CA-0202 ESP-CMD: sensor90 R
Dec 22 18:59:39 tasmota-4E80CA-0202 ESP-SRC: WebConsole from 10.152.52.62
Dec 22 18:59:39 tasmota-4E80CA-0202 ESP-CMD: Grp 0, Cmnd 'SENSOR', Idx 90, Len 1, Data 'R'
Dec 22 18:59:39 tasmota-4E80CA-0202 ESP-HRG: Received 4163632020302E3030206D6D2C204576656E744163632020302E3030206D6D2C20546F74616C4163632020322E3033206D6D2C2052496E742020302E3030206D6D7068
Dec 22 18:59:39 tasmota-4E80CA-0202 ESP-MQT: stat/tasmota_4E80CA/RESULT = {"SerialReceived":"Acc  0.00 mm, EventAcc  0.00 mm, TotalAcc  2.03 mm, RInt  0.00 mmph"}
Dec 22 18:59:45 tasmota-4E80CA-0202 ESP-WIF: Sending Gratuitous ARP
Dec 22 18:59:46 tasmota-4E80CA-0202 ESP-WIF: Checking connection...
Dec 22 18:59:48 tasmota-4E80CA-0202 ESP-CMD: sensor90 K
Dec 22 18:59:48 tasmota-4E80CA-0202 ESP-SRC: WebConsole from 10.152.52.62
Dec 22 18:59:48 tasmota-4E80CA-0202 ESP-CMD: Grp 0, Cmnd 'SENSOR', Idx 90, Len 1, Data 'K'
Dec 22 18:59:48 tasmota-4E80CA-0202 ESP-MQT: stat/tasmota_4E80CA/RESULT = {"Sensor":"Done"}
Dec 22 18:59:50 tasmota-4E80CA-0202 ESP-HRG: Received 52657365742044
Dec 22 18:59:52 tasmota-4E80CA-0202 ESP-HRG: Received
Dec 22 18:59:52 tasmota-4E80CA-0202 ESP-HRG: Received 3B2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2A2AD6A32C3105
Dec 22 18:59:56 tasmota-4E80CA-0202 ESP-CMD: sensor90 R
Dec 22 18:59:56 tasmota-4E80CA-0202 ESP-SRC: WebConsole from 10.152.52.62
Dec 22 18:59:56 tasmota-4E80CA-0202 ESP-CMD: Grp 0, Cmnd 'SENSOR', Idx 90, Len 1, Data 'R'
[... cut ...]
Dec 22 19:16:53 tasmota-4E80CA-0202 ESP-MQT: stat/tasmota_4E80CA/RESULT = {"Command":"Error"}
Dec 22 19:16:55 tasmota-4E80CA-0202 ESP-HRG: Received 41636320302E30303020696E2C204576656E7441636320302E30303020696E2C20546F74616C41636320302E30373320696E2C2052496E7420302E303030203F01
Dec 22 19:16:56 tasmota-4E80CA-0202 ESP-WIF: Checking connection...
Dec 22 19:17:16 tasmota-4E80CA-0202 ESP-WIF: Checking connection...
Dec 22 19:17:18 tasmota-4E80CA-0202 ESP-HTP: Main Menu
Dec 22 19:17:25 tasmota-4E80CA-0202 ESP-HTP: Console
`

tremichl, 2021-12-23:
@ascillato2 , sorry work is in progress and bug is still NOT fixed! 
It`s right, thanks Theo exception28 is gone but sensor in conjunction with Tasmota is unfortunately unusable at the moment. Please see comment and logs above

sensor90 K had no effect, did not erase values
after sensing rain, device was not reachable anymore. Even no AP, needed power reset
sensor90 K sensor90 R results in an error

arendst, 2021-12-23:
The original programmer made some assumptions resulting in exceptions. He also choose for reporting an error message when a "R" command was enetered and no data was available.

These are typical serial comms programming errors. Also the R error message should not be returned to the user as it did consume the R command but failed to report valid data.

I will remove the error message from the R command and leave it here up to any programmer willing to provide more usable code without calling function ``MqttPublishSensor()`` at the wrong locations resulting in exceptions.

tremichl, 2021-12-23:
The error massage itself is not a problem. The real problem now is, that device is not reachable if sensing rain (sending data to Tasmota) and after some hours running the same. Only power reset helps. You can see it also in syslog, after receiving data even syslog is not written anymore. In previews version we had an exception at that point and rebooting, now the device seems to be dead.

`Dec 22 19:01:13 tasmota-4E80CA-0202 ESP-HRG: Received 3B2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D
Dec 22 19:01:13 tasmota-4E80CA-0202 ESP-HRG: Received 4576656E74
Dec 22 19:01:13 tasmota-4E80CA-0202 ESP-HRG: Received 535720312E3030302032302D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D2D
root@Debian-Server:/home/trebuch#`

arendst, 2021-12-23:
As said, up to the original programmer to fix his user input checks and validity of returned received data instead of just assuming it always receives valid data.

Jason2866, 2021-12-23:
@tremichl Difficult to debug without the device. It is a try and fail game. All not compliant Tasmota code is removed. As Theo wrote someone with the sensor and coding skills has to look at.

tremichl, 2021-12-23:
Yes, I can understand that. Maybe the original coder (with a sensor in hands) could improve the code. Thanks to you all.

Jason2866, 2021-12-23:
@arendst  Considering to remove the driver from Tasmota? It is not complete and non working.

ascillato, 2021-12-23:
## Just for reference

Initial PR that adds supports to this sensor: https://github.com/arendst/Tasmota/pull/12974
Author: @wooti
Docs: https://github.com/tasmota/docs/pull/793

arendst, 2021-12-24:
@tremichl Pls try latest dev branch. I re-enabled the sensor messages.

arendst, 2021-12-24:
BTW I orderd a HRG-15 and will solve the issues once received.

tremichl, 2021-12-25:
stability seems to bee a big improvement now. Thank you Theo. Will do some more testing tomorrow and report.

tremichl, 2021-12-28:
Rainsensor on a Wemos D1mini and Tasmota 10.1.0.2 is up and stable running for several days now. No more rebooting on serial data exchange Sensor/Tasmota. Therefor that issue is solved. Thanks!

Other things I have observed:
Flowrate is not updated on website when sensing rain. Only after forcing read data (sensor90 R) it is updated
sensor90 K (kill) Restarts sensor, and response is only "done". Would be nice to see header as console output with FW-Version and so on

@arendst: Maybe you can check that, once you have got your sensor

arendst, 2021-12-30:
Just released an updated driver in the dev branch. Wait two hours and then try this one and let me know if it does what it should.

It works fine on my (in house) sensor.

tremichl, 2021-12-31:
Can get now all needed data on Web Interface and via MQTT and send successfully commands to sensor. Great, thanks! Sensor also still in house. Had 2 restarts within 12 hours. Will reactivate syslog...

arendst, 2022-01-03:
I've updated the driver thoroughly and my RG-15 went into production today after adding a Wemos into the box and power the whole thing with a USB power supply.

Let's close this issue. If anything pops up again pls open a new one.

closed 2022-01-03 by arendst

tremichl, 2022-01-03:
No more reboots for 3 days. Have mounted an ESP12F also into sensor case. And now waiting for rain...Thanks again!