arendst/Tasmota #14073: LedPwmMode causes core panic - ESP32 Solo1

ElksInNC, 2021-12-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

When attempting to use the LedPwmMode command on an ESP Solo1 based WIFI Smart Light Switch, the system crashes and reboots after a core panic

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): LINKIND ESP32 Solo1 Single Pole Switch
- [X] Tasmota binary firmware version number used: 10.1.0.1(tasmota) 2021-12-16T19:15:26 2_0_1_1/v4.4-dev-3728-g7cac8278e
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: _____ WebUI / Firmware Upgrade / Upgrade From File - File downloaded from: http://ota.tasmota.com/tasmota32/
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:25:24.247 MQT: stat/TESTBENCH/RESULT = {"NAME":"Linkind Switch ESP32Solo WS240010008","GPIO":[0,0,0,0,0,224,1,1,0,0,320,0,1,1,0,0,0,0,0,0,0,576,289,0,0,0,0,0,33,32,0,0,0,0,0,0],"FLAG":0,"BASE":1}
23:25:24.456 MQT: stat/TESTBENCH/RESULT = {"Module":{"0":"Linkind Switch ESP32Solo WS240010008"}}
23:25:24.712 MQT: stat/TESTBENCH/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"224":"Relay1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"320":"Led_i1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"576":"LedLink_i"},"GPIO26":{"289":"Led2"},"GPIO27":{"0":"None"},"GPIO32":{"33":"Button2"},"GPIO33":{"32":"Button1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
23:26:08.251 CMD: Backlog Rule1; Rule2; Rule3
23:26:08.340 MQT: stat/TESTBENCH/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
23:26:08.589 MQT: stat/TESTBENCH/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
23:26:08.836 MQT: stat/TESTBENCH/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
23:26:47.342 CMD: Status 0
23:26:47.362 MQT: stat/TESTBENCH/STATUS = {"Status":{"Module":0,"DeviceName":"TESTBENCH","FriendlyName":["TESTBENCH"],"Topic":"TESTBENCH","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
23:26:47.398 MQT: stat/TESTBENCH/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:11:03","StartupUTC":"2021-12-17T04:15:44","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-12-17T04:54:37","SaveCount":13}}
23:26:47.439 MQT: stat/TESTBENCH/STATUS2 = {"StatusFWR":{"Version":"10.1.0.1(tasmota)","BuildDateTime":"2021-12-16T19:15:26","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":80,"Hardware":"ESP32-U4WDH","CR":"446/699"}}
23:26:47.479 MQT: stat/TESTBENCH/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008109","2805C80001000600003C5A0A000000000000","000002A0","00006000","00004000"]}}
23:26:47.520 MQT: stat/TESTBENCH/STATUS4 = {"StatusMEM":{"ProgramSize":1306,"Free":1856,"Heap":166,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52,59","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,127"}}
23:26:47.558 MQT: stat/TESTBENCH/STATUS5 = {"StatusNET":{"Hostname":"TESTBENCH","IPAddress":"10.10.10.175","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.10.1","DNSServer2":"0.0.0.0","Mac":"B8:F0:09:XX:XX:XX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:26:47.592 MQT: stat/TESTBENCH/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.50.25","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_701F60","MqttUser":"XXXXX","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:26:47.627 MQT: stat/TESTBENCH/STATUS7 = {"StatusTIM":{"UTC":"2021-12-17T04:26:47","Local":"2021-12-16T23:26:47","StartDST":"2021-03-14T02:00:00","EndDST":"2021-11-07T02:00:00","Timezone":99,"Sunrise":"02:38","Sunset":"10:53"}}
23:26:47.661 MQT: stat/TESTBENCH/STATUS10 = {"StatusSNS":{"Time":"2021-12-16T23:26:47","ESP32":{"Temperature":147.0},"TempUnit":"F"}}
23:26:47.704 MQT: stat/TESTBENCH/STATUS11 = {"StatusSTS":{"Time":"2021-12-16T23:26:47","Uptime":"0T00:11:03","UptimeSec":663,"Heap":166,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":53},"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXX","BSSId":"E6:63:DA:XX:XX:XX","Channel":1,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:06"}}}
23:26:47.744 MQT: stat/TESTBENCH/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"LoadProhibited","EPC":"40090a17","EXCVADDR":"00000038","CallChain":["40090a14","4013f07f","400d43ba","400e67e4","400e69e5","400e6a3c","400fe9ff","401041a6","401043ba","40106149","40117135","40117144","40140e25"]}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
SerialLog provided as the crash data is not captured in Weblog

23:27:36.978 CMD: seriallog 4
23:27:36.996 MQT: stat/TESTBENCH/RESULT = {"SerialLog":{"4":{"Active":"4"}}}
23:27:37.885 CFG: Saved, Count 15, Bytes 4096
23:27:54.807 WIF: Checking connection...
23:27:59.069 CMD: LedPwmMode1 1
23:27:59.071 SRC: Serial
23:27:59.074 CMD: Grp 0, Cmnd 'LEDPWMMODE', Idx 1, Len 1, Data '1'
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40090a17  PS      : 0x00060933  A0      : 0x8013f082  A1      : 0x3ffb2510  
A2      : 0x00000000  A3      : 0x0000004c  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x000003ff  A7      : 0x3ffcbafc  A8      : 0x80090a14  A9      : 0x3ffb24f0  
A10     : 0x0000004c  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000000  
[... 24 lines cut ...]
 backlog0 mqtthost 192.168.50.25; topic TESTBENCH; mqttuser xxxxx; mqttpassword xxxxxx; 

Set LedPwmMode
 LedPwmMode1 1

CRASH

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

LedPwmMode1 set to ON and the ability to use LedPwmOff and LedPwmOn to set brightness values for LED

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Have been extensively testing this switch with berry and BLE using this DEVs code.
https://github.com/tony-fav/tasmota-blerry

Only attempting to set brightness of LED when ON to a lower setting.  All above results were without BLE and without any berry code loaded to UFS.

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2021-12-21:
Same effect on ESP32-C3

arendst, 2021-12-22:
The issue is caused by lack of call to ``analogAttach()`` for used LED gpio.

Implementing this is tricky as there are 8 possible "attach point" of which the first x (up to 5) are being used by configured PWM channels. The last attach point is reserved for the buzzer. 

Working on a safe solution ...

arendst, 2021-12-22:
Fixed in latest dev branch

Jason2866, 2021-12-23:
Fix confirmed

closed 2021-12-23 by Jason2866