arendst/Tasmota #14075: Rule execution problem

smokersonly, 2021-12-17:
### PROBLEM DESCRIPTION
Rule does not get executed

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Zbbridge
- [x] Tasmota binary firmware version number used: 10.1.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:21:43.781 MQT: stat/tasmota_ZBBridge2/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:21:44.006 MQT: stat/tasmota_ZBBridge2/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
11:21:44.209 MQT: stat/tasmota_ZBBridge2/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
11:24:10.564 RUL: Rule1 +zbreceived#0x2548#temperature do publish2 tele/zigbee/hobbykamer/temperature %value% endon Rule1 1
11:24:10.568 MQT: stat/tasmota_ZBBridge2/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":1089,"Free":276,"Rules":"on zbreceived#0x8EE4#humidity do publish2 tele/zigbee/living/humidity %value% endon   on zbreceived#0x8EE4#temperature do publish2 tele/zigbee/living/temperature %value% endon   on zbreceived#0xD596#humidity do publish2 tele/zigbee/buro/humidity %value% endon   on zbreceived#0xD596#temperature do publish2 tele/zigbee/buro/temperature %value% endon   on zbreceived#0x2016#humidity do publish2 tele/zigbee/badkamer/humidity %value% endon   on zbreceived#0x2016#temperature do publish2 tele/zigbee/badkamer/temperature %value% endon   on zbreceived#0x3C55#humidity do publish2 tele/zigbee/toilet/humidity %value% endon   on zbreceived#0x3C55#temperature do publish2 tele/zigbee/toilet/temperature %value% endon   on zbreceived#0xF9CB#humidity do publish2 tele/zigbee/slaapkamer/humidity %value% endon   on zbreceived#0xF9CB#temperature do publish2 tele/zigbee/slaapkamer/temperature %value% endon   on zbreceived#0x2548#humidity do publish2 tele/zigbee/hobbykamer/humidity %value% endon   on zbreceived#0x2548#temperature do publish2 tele/zigbee/hobbykamer/temperature %value% endon Rule1 1"}}
11:24:10.807 RUL: Rule2 on zbinfo#0x8EE4#lastseen do publish2 tele/zigbee/living/lastseen %value% endon   on zbinfo#0x8EE4#batterypercentage do publish2 tele/zigbee/living/batterypercentage %value% endon   on zbinfo#0xD596#lastseen do publish2 tele/zigbee/buro/lastseen %value% endon   on zbinfo#0xD596#batterypercentage do publish2 tele/zigbee/buro/batterypercentage %value% endon   on zbinfo#0x2016#lastseen do publish2 tele/zigbee/badkamer/lastseen %value% endon   on zbinfo#0x2016#batterypercentage do publish2
11:24:10.809 RUL: Rule2 +tele/zigbee/badkamer/batterypercentage %value% endon   on zbinfo#0x3C55#lastseen do publish2 tele/zigbee/toilet/lastseen %value% endon   on zbinfo#0x3C55#batterypercentage do publish2 tele/zigbee/toilet/batterypercentage %value% endon   on zbinfo#0xF9CB#lastseen do publish2 tele/zigbee/slaapkamer/lastseen %value% endon   on zbinfo#0xF9CB#batterypercentage do publish2 tele/zigbee/slaapkamer/batterypercentage %value% endon   on zbinfo#0x2548#lastseen do publish2 tele/zigbee/hobbykamer/lastseen %value%
11:24:10.809 RUL: Rule2 +endon   on zbinfo#0x2548#batterypercentage do publish2 tele/zigbee/hobbykamer/batterypercentage %value% endon Rule2 1
11:24:10.814 MQT: stat/tasmota_ZBBridge2/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":1113,"Free":278,"Rules":"on zbinfo#0x8EE4#lastseen do publish2 tele/zigbee/living/lastseen %value% endon   on zbinfo#0x8EE4#batterypercentage do publish2 tele/zigbee/living/batterypercentage %value% endon   on zbinfo#0xD596#lastseen do publish2 tele/zigbee/buro/lastseen %value% endon   on zbinfo#0xD596#batterypercentage do publish2 tele/zigbee/buro/batterypercentage %value% endon   on zbinfo#0x2016#lastseen do publish2 tele/zigbee/badkamer/lastseen %value% endon   on zbinfo#0x2016#batterypercentage do publish2 tele/zigbee/badkamer/batterypercentage %value% endon   on zbinfo#0x3C55#lastseen do publish2 tele/zigbee/toilet/lastseen %value% endon   on zbinfo#0x3C55#batterypercentage do publish2 tele/zigbee/toilet/batterypercentage %value% endon   on zbinfo#0xF9CB#lastseen do publish2 tele/zigbee/slaapkamer/lastseen %value% endon   on zbinfo#0xF9CB#batterypercentage do publish2 tele/zigbee/slaapkamer/batterypercentage %value% endon   on zbinfo#0x2548#lastseen do publish2 tele/zigbee/hobbykamer/lastseen %value% endon   on zbinfo#0x2548#batterypercentage do publish2 tele/zigbee/hobbykamer/batterypercentage %value% endon Rule2 1"}}
11:24:11.059 MQT: stat/tasmota_ZBBridge2/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":40,"Free":471,"Rules":"on Time#Minute|5 do ZbInfo endon Rule3 1"}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:25:00.469 MQT: stat/tasmota_ZBBridge2/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_ZBBridge2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:25:00.495 MQT: stat/tasmota_ZBBridge2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:38:58","StartupUTC":"2021-12-17T08:46:02","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2021-11-27T16:49:16","SaveCount":44,"SaveAddress":"FD000"}}
11:25:00.519 MQT: stat/tasmota_ZBBridge2/STATUS2 = {"StatusFWR":{"Version":"10.1.0.1(zbbridge)","BuildDateTime":"2021-12-16T19:15:17","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"371/699"}}
11:25:00.540 MQT: stat/tasmota_ZBBridge2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["abcxyz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008209","2805C80001000600003C5A00000000000000","00000080","00006000","00004000"]}}
11:25:00.566 MQT: stat/tasmota_ZBBridge2/STATUS4 = {"StatusMEM":{"ProgramSize":671,"Free":1120,"Heap":20,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","00000000"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50","Sensors":"1"}}
11:25:00.592 MQT: stat/tasmota_ZBBridge2/STATUS5 = {"StatusNET":{"Hostname":"tasmota-ZBBridge2-5411","IPAddress":"192.168.9.151","Gateway":"192.168.9.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.9.222","DNSServer2":"192.168.9.1","Mac":"C8:2B:96:04:D5:23","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:25:00.620 MQT: stat/tasmota_ZBBridge2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.9.222","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_04D523","MqttUser":"thuis","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:25:00.643 MQT: stat/tasmota_ZBBridge2/STATUS7 = {"StatusTIM":{"UTC":"2021-12-17T10:25:00","Local":"2021-12-17T11:25:00","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:38","Sunset":"16:53"}}
[... cut ...]
Connect a zigbee device and create rule2 and rule3

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
When Rule3 triggers, Rule2 should be triggered by the output of command in Rule3.
FYI When I manually enter the command in Rule3 (Zbinfo), then it does trigger Rule2

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-12-17:
Since the complete rule can not be seen, it is hard to help.
Please post the complete rules.

smokersonly, 2021-12-17:
Rule2
  on zbinfo#0x8EE4#lastseen do publish2 tele/zigbee/living/lastseen %value% endon
  on zbinfo#0x8EE4#batterypercentage do publish2 tele/zigbee/living/batterypercentage %value% endon
  on zbinfo#0xD596#lastseen do publish2 tele/zigbee/buro/lastseen %value% endon
  on zbinfo#0xD596#batterypercentage do publish2 tele/zigbee/buro/batterypercentage %value% endon
  on zbinfo#0x2016#lastseen do publish2 tele/zigbee/badkamer/lastseen %value% endon
  on zbinfo#0x2016#batterypercentage do publish2 tele/zigbee/badkamer/batterypercentage %value% endon
  on zbinfo#0x3C55#lastseen do publish2 tele/zigbee/toilet/lastseen %value% endon
  on zbinfo#0x3C55#batterypercentage do publish2 tele/zigbee/toilet/batterypercentage %value% endon
  on zbinfo#0xF9CB#lastseen do publish2 tele/zigbee/slaapkamer/lastseen %value% endon
  on zbinfo#0xF9CB#batterypercentage do publish2 tele/zigbee/slaapkamer/batterypercentage %value% endon
  on zbinfo#0x2548#lastseen do publish2 tele/zigbee/hobbykamer/lastseen %value% endon
  on zbinfo#0x2548#batterypercentage do publish2 tele/zigbee/hobbykamer/batterypercentage %value% endon
Rule2 1

Rule3
  on Time#Minute|5 do ZbInfo endon
Rule3 1

Rule2
12:42:04.551 RUL: Rule2 on zbinfo#0x8EE4#lastseen do publish2 tele/zigbee/living/lastseen %value% endon   on zbinfo#0x8EE4#batterypercentage do publish2 tele/zigbee/living/batterypercentage %value% endon   on zbinfo#0xD596#lastseen do publish2 tele/zigbee/buro/lastseen %value% endon   on zbinfo#0xD596#batterypercentage do publish2 tele/zigbee/buro/batterypercentage %value% endon   on zbinfo#0x2016#lastseen do publish2 tele/zigbee/badkamer/lastseen %value% endon   on zbinfo#0x2016#batterypercentage do publish2
12:42:04.554 RUL: Rule2 +tele/zigbee/badkamer/batterypercentage %value% endon   on zbinfo#0x3C55#lastseen do publish2 tele/zigbee/toilet/lastseen %value% endon   on zbinfo#0x3C55#batterypercentage do publish2 tele/zigbee/toilet/batterypercentage %value% endon   on zbinfo#0xF9CB#lastseen do publish2 tele/zigbee/slaapkamer/lastseen %value% endon   on zbinfo#0xF9CB#batterypercentage do publish2 tele/zigbee/slaapkamer/batterypercentage %value% endon   on zbinfo#0x2548#lastseen do publish2 tele/zigbee/hobbykamer/lastseen %value%
12:42:04.555 RUL: Rule2 +endon   on zbinfo#0x2548#batterypercentage do publish2 tele/zigbee/hobbykamer/batterypercentage %value% endon Rule2 1
12:42:04.559 MQT: stat/tasmota_ZBBridge2/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":1113,"Free":278,"Rules":"on zbinfo#0x8EE4#lastseen do publish2 tele/zigbee/living/lastseen %value% endon   on zbinfo#0x8EE4#batterypercentage do publish2 tele/zigbee/living/batterypercentage %value% endon   on zbinfo#0xD596#lastseen do publish2 tele/zigbee/buro/lastseen %value% endon   on zbinfo#0xD596#batterypercentage do publish2 tele/zigbee/buro/batterypercentage %value% endon   on zbinfo#0x2016#lastseen do publish2 tele/zigbee/badkamer/lastseen %value% endon   on zbinfo#0x2016#batterypercentage do publish2 tele/zigbee/badkamer/batterypercentage %value% endon   on zbinfo#0x3C55#lastseen do publish2 tele/zigbee/toilet/lastseen %value% endon   on zbinfo#0x3C55#batterypercentage do publish2 tele/zigbee/toilet/batterypercentage %value% endon   on zbinfo#0xF9CB#lastseen do publish2 tele/zigbee/slaapkamer/lastseen %value% endon   on zbinfo#0xF9CB#batterypercentage do publish2 tele/zigbee/slaapkamer/batterypercentage %value% endon   on zbinfo#0x2548#lastseen do publish2 tele/zigbee/hobbykamer/lastseen %value% endon   on zbinfo#0x2548#batterypercentage do publish2 tele/zigbee/hobbykamer/batterypercentage %value% endon Rule2 1"}}

Rule3
12:40:11.356 MQT: stat/tasmota_ZBBridge2/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":40,"Free":471,"Rules":"on Time#Minute|5 do ZbInfo endon Rule3 1"}}

ascillato2, 2021-12-17:
Hi, your rules have syntax errors. For example, rule 3 has a command after endon and that invalidates it.

Rule3 on Time#Minute|5 do ZbInfo endon Rule3 1

It should be

Rules on Time#Minute|5 do ZbInfo endon

Please, re check your rules for correct syntax. Thanks.

smokersonly, 2021-12-20:
Hi, I have cleaned up my rules (basically removing the ending 'Rule2 1' and 'Rule3 1'

I still have the same issue.
13:15:48.551 MQT: stat/tasmota_ZBBridge2/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":32,"Free":479,"Rules":"on Time#Minute|5 do ZbInfo endon"}}

13:18:21.195 MQT: stat/tasmota_ZBBridge2/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":1083,"Free":286,"Rules":"on zbinfo#0x8EE4#lastseen do publish2 tele/zigbee/living/lastseen %value% endon on zbinfo#0x8EE4#batterypercentage do publish2 tele/zigbee/living/batterypercentage %value% endon on zbinfo#0xD596#lastseen do publish2 tele/zigbee/buro/lastseen %value% endon on zbinfo#0xD596#batterypercentage do publish2 tele/zigbee/buro/batterypercentage %value% endon on zbinfo#0x2016#lastseen do publish2 tele/zigbee/badkamer/lastseen %value% endon on zbinfo#0x2016#batterypercentage do publish2 tele/zigbee/badkamer/batterypercentage %value% endon on zbinfo#0x3C55#lastseen do publish2 tele/zigbee/toilet/lastseen %value% endon on zbinfo#0x3C55#batterypercentage do publish2 tele/zigbee/toilet/batterypercentage %value% endon on zbinfo#0xF9CB#lastseen do publish2 tele/zigbee/slaapkamer/lastseen %value% endon on zbinfo#0xF9CB#batterypercentage do publish2 tele/zigbee/slaapkamer/batterypercentage %value% endon on zbinfo#0x2548#lastseen do publish2 tele/zigbee/hobbykamer/lastseen %value% endon on zbinfo#0x2548#batterypercentage do publish2 tele/zigbee/hobbykamer/batterypercentage %value% endon"}}

pkkrusty, 2022-01-08:
try adding backlog before ZbInfo

sfromis, 2022-01-09:
Case should not matter, as rules generally do the processing after folding to upper case. 

Wrapping the ZbInfo command in Backlog is a pretty good suggestion :-)

smokersonly, 2022-01-10:
Wrapping up the ZbInfo in Backlog fixed the issue. THX for the suggestion!

closed 2022-01-10 by smokersonly