arendst/Tasmota #14082: Exception and reboot when using a VL53L1X

dbochicchio, 2021-12-17:
### PROBLEM DESCRIPTION

I'm using a VL53L1X with the latest version. When I set SDA/SCL, the NodeMCU makes two reboots and reset its configuration.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

17:35:42.031 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_5FB010","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:35:42.057 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-sensors.bin","RestartReason":"Software Watchdog","Uptime":"0T00:03:47","StartupUTC":"2021-12-17T16:31:55","Sleep":50,"CfgHolder":4617,"BootCount":74,"BCResetTime":"2020-10-11T17:05:52","SaveCount":208,"SaveAddress":"FA000"}}
17:35:42.089 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0.1(sensors)","BuildDateTime":"2021-12-17T10:32:25","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"375/699"}}
17:35:42.109 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
17:35:42.143 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":668,"Free":332,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
17:35:42.179 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-airfresh","IPAddress":"192.168.1.196","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.26","DNSServer2":"192.168.1.1","Mac":"EC:FA:BC:5F:B0:10","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:35:42.203 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5FB010","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:35:42.225 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-12-17T16:35:42","Local":"2021-12-17T17:35:42","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:38","Sunset":"16:53"}}
17:35:42.245 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-12-17T17:35:42"}}
17:35:42.251 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-12-17T17:35:42","Uptime":"0T00:03:47","UptimeSec":227,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"XX,"BSSId":"XX","Channel":11,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}
17:35:42.283 RSL: STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Software Watchdog","EPC":["401079b5","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40256f80","40256fdc","402571aa","402573e4","40246f7c","40246fa4","4024b5a6","4024b72a","4024bade","40204c18","4022039d","4010148e","402435cc","402222f8","4022dbaa","40256650","4010211d"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-12-17:
Please, can you share the output of `Backlog Template; Module; GPIO 255` ?

dbochicchio, 2021-12-20:
I've tried to catch them, but it's impossible, since the device reboot itself and resets it configuration.

So, output of the above command is like the sensor was resetted

```
18:52:54.194 CMD: Backlog Template; Module; GPIO 255
18:52:54.196 SRC: WebConsole from 192.168.1.24
18:52:54.199 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
18:52:54.273 SRC: Backlog
18:52:54.276 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
18:52:54.283 MQT: stat/tasmota-pellet/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:52:54.493 SRC: Backlog
18:52:54.496 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
18:52:54.501 MQT: stat/tasmota-pellet/RESULT = {"Module":{"18":"Generic"}}
18:52:54.743 SRC: Backlog
18:52:54.745 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
18:52:54.758 MQT: stat/tasmota-pellet/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

ascillato, 2021-12-20:
Ok. So which GPIOs are you using?

barbudor, 2021-12-20:
The NodeMCU provodes logs on the serial
Please set `SerialLog 3` and then use a serial terminal to get the detailled logs from when you configure SDA/SCL until end of reboots

dbochicchio, 2021-12-22:
> Ok. So which GPIOs are you using?

GPIO5 for SCL, GPIO4 for SDA. I'll try with a serial terminal later this week, since I'm out of home at the moment. Thanks.

sfromis, 2022-01-12:
Just did a small test, NodeMCU and VL53L1X, I2C SCL/SDA on gpio 5/4. No resets, reboots or other problems, sensor works as expected. Used a self-compiled binary to include support for the sensor.

Looks like you used tasmota-sensors.bin, the precompiled version of this does not include `USE_VL53L1X` for Tasmota support of this sensor. It does however include `USE_VL53L0X` for the older variant, but you need the specific driver to match the sensor chip. It may be a good plan to specifically disable I2C drivers you are not using, to avoid conflicts, especially with other drivers expecting a sensor on the same I2C address `0x29`.

dbochicchio, 2022-01-12:
@sfromis thanks for checking in. I've created a custom build with just 

```
#define USE_I2C
#define USE_VL53L1X
```

and it's still crashing. Can you please share your custom build settings? Thanks.

sfromis, 2022-01-12:
It may not be a question of settings in the build, but about disabling unneeded I2C drivers, as just suggested.

In addition to using the I2CDriver command for this, driver by driver, you can also do what I'm doing when creating builds, disabling all drivers:
```
#define I2CDRIVERS_0_31  0x00000000
#define I2CDRIVERS_32_63 0x00000000
#define I2CDRIVERS_64_95 0x00000000
```
And then I enable the ones I want after first boot, using the I2CDriver command. This is not "better", but tends to be less work.

dbochicchio, 2022-01-12:
Ok, no more crashing, but device is not detected after `I2CScan`. What's the best way to enable the device again? I'm sorry if those are very basic questions.

sfromis, 2022-01-12:
If I2CScan does not detect, triple-check your wiring and connections.

dbochicchio, 2022-01-12:
Wires as OK.  I've just reflashed tasmota-sensors.bin and output seems OK:

```{"I2CScan":"Device(s) found at 0x29"}```

So it's definitely something in the way I'm preparing the build. I've read a couple of things in the doc, but it's something very subtle that I'm missing.

trlafleur, 2022-01-25:
I'm also having issues with this sensor, if I add it to my build, with the sensor-enabled, I flash the code, let it run, set the I2C port in configuration, save it, it will hang the ESP32 at 2nd boot...
More to follow...

codebase: 2022.01.3(tasmota)-2_0_2_1
I2C on pins 22 and 23
~~~
#undef USE_VL53L0X    // these have same addres as VL53L1
#undef USE_TSL2561
#undef USE_TSL2591

#ifndef USE_VL53L1X 
#define USE_VL53L1X 
#endif
//#undef USE_VL53L1X

//#define I2CDRIVERS_0_31  0x00000000
//#define I2CDRIVERS_32_63 0x00000000
//#define I2CDRIVERS_64_95 0x00000000
~~~
~~~
Hard Resetting...
Done! Flashing is complete!
Showing logs:
[14:55:51]
[14:55:51]00:00:00.003-214/50 HDW: ESP32-D0WDQ6 
[14:55:51]./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
[14:55:51]00:00:00.202-211/50 UFS: FlashFS mounted with 312 kB free
[14:55:51]00:00:00.210-211/50 CFG: Use defaults
[14:55:51]00:00:00.364 QPC: Reset
[14:55:52]00:00:00.444 BRY: Berry initialized, RAM used=3930 bytes
[14:55:52]00:00:00.473 Project tasmota - Tasmota Version 2022.01.3(tasmota)-2_0_2_1(2022-01-25T11:23:51)
[14:55:52]00:00:00.326 WIF: Connecting to AP1 lafleur in mode 11n as tasmota-286814-2068...
[14:55:57]00:00:05.052 WIF: Connected
[14:55:57]00:00:05.312 HTP: Web server active on tasmota-286814-2068 with IP address 192.94.166.30
[14:55:59]23:55:58.390 MQT: Attempting connection...
[14:55:59]23:55:58.429 MQT: Connected
[14:55:59]23:55:58.437 MQT: tele/tasmota_286814/LWT = Online
[14:55:59]23:55:58.451 MQT: cmnd/tasmota_286814/POWER = 
[14:55:59]23:55:58.467 MQT: tele/tasmota_286814/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"2022.01.3(tasmota)","FallbackTopic":"cmnd/DVES_286814_fb/","GroupTopic":"cmnd/tasmotas/"}}
[14:55:59]23:55:58.490 MQT: tele/tasmota_286814/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-286814-2068","IPAddress":"192.94.166.30"}}
[14:55:59]23:55:58.508 MQT: tele/tasmota_286814/INFO3 = {"Info3":{"RestartReason":"Vbat power on reset"}}
[14:56:03]23:56:02.656 MQT: tele/tasmota_286814/STATE = {"Time":"2022-01-25T23:56:02","Uptime":"0T00:00:12","UptimeSec":12,"Heap":159,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":145,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":38},"Wifi":{"AP":1,"SSId":"lafleur","BSSId":"9C:C9:EB:9A:56:C1","Channel":1,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:06"}}
[14:56:03]23:56:02.709 MQT: tele/tasmota_286814/SENSOR = {"Time":"2022-01-25T23:56:02","ESP32":{"Temperature":58.9},"TempUnit":"C"}
[14:56:46]23:56:45.773 MQT: stat/tasmota_286814/RESULT = {"Module":{"0":"ESP32-DevKit"}}
[14:56:47]23:56:46.115 MQT: stat/tasmota_286814/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"640":"I2C SDA1"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO39":{"0":"None"}}
[14:56:47]23:56:46.736 APP: Restarting
[14:56:48]ets Jun  8 2016 00:22:57
[14:56:48]
[14:56:48]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:56:48]configsip: 0, SPIWP:0xee
[14:56:48]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:56:48]mode:DOUT, clock div:2
[14:56:48]load:0x3fff0018,len:4
[14:56:48]load:0x3fff001c,len:1044
[14:56:48]load:0x40078000,len:10124
[14:56:48]load:0x40080400,len:5828
[14:56:48]entry 0x400806a8
[14:56:48]
[14:56:48]00:00:00.003 HDW: ESP32-D0WDQ6 
[14:56:48]00:00:00.059 UFS: FlashFS mounted with 304 kB free
[14:56:48]00:00:00.161 CFG: Loaded from File, Count 6
[14:56:48]00:00:00.177 QPC: Count 1
[14:56:49]00:00:00.412 BRY: Berry initialized, RAM used=3930 bytes
[14:56:49]00:00:00.445 Project tasmota - Tasmota Version 2022.01.3(tasmota)-2_0_2_1(2022-01-25T11:23:51)

~~~

sfromis, 2022-01-26:
If I2CScan finds nothing, it is usually a question of not having appropriate wiring of the sensor (assuming that it is working, of course). Your output has `"GPIO21":{"608":"I2C SCL1"},"GPIO22":{"640":"I2C SDA1"}` which does **not** match your statement about *I2C on pins 22 and 23*.

trlafleur, 2022-01-26:
The wiring is correct, runs just fine under Arduino on the same hardware...

If I boot up, with the sensor disconnected, then re-add the sensor, do a scan, I get
`{"I2CScan":"Device(s) found at 0x29"}`  so Tasmota has access to device....
currently reviewing driver code....

sfromis, 2022-01-26:
With ESP32, `"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"}` I'm getting `"VL53L1X":{"Distance":23}`

Sometimes, it can be useful to disable unused I2C sensor drivers.

Jason2866, 2022-01-26:
Closing since the orig. issue and the later mentioned one can not be reproduced

closed 2022-01-26 by Jason2866

trlafleur, 2022-01-26:
Ok, I disable all the I2C drivers with #define below, I enabled the VL53L1
~~~
#define I2CDRIVERS_0_31  0x00000000
#define I2CDRIVERS_32_63 0x00000000
#define I2CDRIVERS_64_95 0x00000000

#undef USE_VL53L0X    // same address as VL53L1
#undef USE_TSL2561
#undef USE_TSL2591

#ifndef USE_VL53L1X 
#define USE_VL53L1X 
#endif
~~~
I check for the device on the bus:
`[07:28:22]16:28:21.987 MQT: stat/tasmota_0B9F50/RESULT = {"I2CScan":"Device(s) found at 0x29"}`

Then I enable the driver:
`[07:28:31]16:28:30.859 CMD: i2cdriver 77 1`
It rebooted... but still no driver active??
did I miss something??

~~~
[07:28:00]00:00:00.003 HDW: ESP32-D0WDQ6 
[07:28:00]00:00:00.059 UFS: FlashFS mounted with 304 kB free
[07:28:00]00:00:00.160 CFG: Loaded from File, Count 6
[07:28:00]00:00:00.176 QPC: Count 1
[07:28:01]00:00:00.408 BRY: Berry initialized, RAM used=3930 bytes
[07:28:01]00:00:00.441 Project tasmota - Tasmota Version 2022.01.3(tasmota)-2_0_2_1(2022-01-26T07:25:54)
[07:28:02]00:00:01.168 WIF: Connecting to AP1 lafleur Channel 1 BSSId 9C:C9:EB:9A:56:C1 in mode 11n as tasmota-0B9F50-8016...
[07:28:03]00:00:02.593 WIF: Connected
[07:28:04]00:00:03.248 HTP: Web server active on tasmota-0B9F50-8016 with IP address 192.94.166.30
[07:28:05]16:28:04.192 MQT: Attempting connection...
[07:28:05]16:28:04.223 MQT: Connected
[07:28:05]16:28:04.230 MQT: tele/tasmota_0B9F50/LWT = Online
[07:28:05]16:28:04.244 MQT: cmnd/tasmota_0B9F50/POWER = 
[07:28:05]16:28:04.259 MQT: tele/tasmota_0B9F50/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"2022.01.3(tasmota)","FallbackTopic":"cmnd/DVES_0B9F50_fb/","GroupTopic":"cmnd/tasmotas/"}}
[07:28:05]16:28:04.280 MQT: tele/tasmota_0B9F50/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-0B9F50-8016","IPAddress":"192.94.166.30"}}
[07:28:05]16:28:04.298 MQT: tele/tasmota_0B9F50/INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
[07:28:07]16:28:06.801 QPC: Reset
[07:28:09]16:28:08.813 MQT: tele/tasmota_0B9F50/STATE = {"Time":"2022-01-26T16:28:08","Uptime":"0T00:00:09","UptimeSec":9,"Heap":159,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":39},"Wifi":{"AP":1,"SSId":"lafleur","BSSId":"9C:C9:EB:9A:56:C1","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:04"}}
[07:28:09]16:28:08.860 MQT: tele/tasmota_0B9F50/SENSOR = {"Time":"2022-01-26T16:28:08","ESP32":{"Temperature":45.0},"TempUnit":"C"}
[07:28:22]16:28:21.954 CMD: i2cscan
[07:28:22]16:28:21.987 MQT: stat/tasmota_0B9F50/RESULT = {"I2CScan":"Device(s) found at 0x29"}
[07:28:31]16:28:30.859 CMD: i2cdriver 77 1
[07:28:31]16:28:30.874 MQT: stat/tasmota_0B9F50/RESULT = {"I2CDriver":"!7,!8,!9,!10,!11,!12,!13,!14,!15,!17,!18,!20,!24,!29,!31,!36,!41,!42,!44,!46,!48,!54,!58,!62"}
[07:28:33]16:28:32.619 APP: Restarting
[07:28:33]ets Jun  8 2016 00:22:57
[07:28:33]
[07:28:33]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[07:28:33]configsip: 0, SPIWP:0xee
[07:28:33]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[07:28:33]mode:DOUT, clock div:2
[07:28:33]load:0x3fff0018,len:4
[07:28:33]load:0x3fff001c,len:1044
[07:28:33]load:0x40078000,len:10124
[07:28:33]load:0x40080400,len:5828
[07:28:33]entry 0x400806a8
[07:28:34]
[07:28:34]00:00:00.003 HDW: ESP32-D0WDQ6 
[07:28:34]00:00:00.022 UFS: FlashFS mounted with 304 kB free
[07:28:34]00:00:00.057 CFG: Loaded from File, Count 8
[07:28:34]00:00:00.072 QPC: Count 1
[07:28:34]00:00:00.181 BRY: Berry initialized, RAM used=3930 bytes
[07:28:34]00:00:00.203 Project tasmota - Tasmota Version 2022.01.3(tasmota)-2_0_2_1(2022-01-26T07:25:54)
[07:28:35]00:00:00.963 WIF: Connecting to AP1 lafleur Channel 1 BSSId 9C:C9:EB:9A:56:C1 in mode 11n as tasmota-0B9F50-8016...
[07:28:37]00:00:02.893 WIF: Connected
[07:28:37]00:00:03.164 HTP: Web server active on tasmota-0B9F50-8016 with IP address 192.94.166.30
[07:28:38]16:28:38.139 MQT: Attempting connection...
[07:28:39]16:28:38.164 MQT: Connected
[07:28:39]16:28:38.172 MQT: tele/tasmota_0B9F50/LWT = Online
[07:28:39]16:28:38.186 MQT: cmnd/tasmota_0B9F50/POWER = 
[07:28:39]16:28:38.202 MQT: tele/tasmota_0B9F50/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"2022.01.3(tasmota)","FallbackTopic":"cmnd/DVES_0B9F50_fb/","GroupTopic":"cmnd/tasmotas/"}}
[07:28:39]16:28:38.225 MQT: tele/tasmota_0B9F50/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-0B9F50-8016","IPAddress":"192.94.166.30"}}
[07:28:39]16:28:38.242 MQT: tele/tasmota_0B9F50/INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
[07:28:41]16:28:40.954 QPC: Reset
[07:28:43]16:28:42.956 MQT: tele/tasmota_0B9F50/STATE = {"Time":"2022-01-26T16:28:42","Uptime":"0T00:00:09","UptimeSec":9,"Heap":159,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":39},"Wifi":{"AP":1,"SSId":"lafleur","BSSId":"9C:C9:EB:9A:56:C1","Channel":1,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:04"}}
[07:28:43]16:28:43.002 MQT: tele/tasmota_0B9F50/SENSOR = {"Time":"2022-01-26T16:28:42","ESP32":{"Temperature":45.0},"TempUnit":"C"}
[07:29:13]16:29:12.964 CMD: i2cdriver77 1
[07:29:13]16:29:12.980 MQT: stat/tasmota_0B9F50/RESULT = {"I2CDriver":"!7,!8,!9,!10,!11,!12,!13,!14,!15,!17,!18,!20,!24,!29,!31,!36,!41,!42,!44,!46,!48,!54,!58,!62"}
[07:29:14]16:29:14.437 APP: Restarting
[07:29:15]ets Jun  8 2016 00:22:57

~~~

sfromis, 2022-01-26:
`i2cdriver 77 1` is a bad command. You should have no space before the index, and VL53L1X is not [driver](https://tasmota.github.io/docs/I2CDEVICES/) number 77.

barbudor, 2022-01-26:
Tom, VL53L1X is I2C #54

(yes it is xsns_77 but i2c 54)

[image]

trlafleur, 2022-01-26:
Thank you for the help.  It’s all good now

trlafleur, 2022-01-26:
yes, I tried with and without space..  I now see from the table its 54.... confusing to a beginner....
```
#define XSNS_77     77   
xsns_77_vl53l1x.ino - VL53L1X sensor support for Tasmota
```
also, this issue should stay open, as there is an issue with this driver as it's not playing well with other I2C devices...  see  --> dbochicchio commented on Dec 17, 2021, at top of this issue

trlafleur, 2022-01-26:
Adding this to enable the driver at compile boot time..
Having some documentation on this device would be great
~~~
   #define I2CDRIVERS_0_31  0x00000000 
   #define I2CDRIVERS_32_63 0x00400000     // enable only device 54, the VL53L1X
   #define I2CDRIVERS_64_95 0x00000000
~~~

barbudor, 2022-01-26:
What documentation would like to see ? 
There is not much to configure from Tasmota : I2C GPIO, I2CDriver command to enabe it (and disable un-used drivers).
It's mostly plug'n'play (once you've right the basic configuration but that's nothing specific to this sensor 😂)

trlafleur, 2022-01-26:
As a minimum, you might add a supplement to `https://tasmota.github.io/docs/VL53L0x/` to show what is needed to get the VL53L1X working.. This would have saved me lots of time...

1) how to enable the driver in the codebase --> `tasmota/user_config_override.h`
~~~
  #ifndef USE_VL53L1X 
  #define USE_VL53L1X 
  #endif
~~~

2) That sometimes we need to disable other I2C devices for compatibility...
~~~
   #define I2CDRIVERS_0_31  0x00000000 
   #define I2CDRIVERS_32_63 0x00400000     // enable only device 54, the VL53L1X
   #define I2CDRIVERS_64_95 0x00000000
~~~

3) How to check if the device is available using I2CScan

` {"I2CScan":"Device(s) found at 0x29"}`

Thank you, it was a good learning experience!...