arendst/Tasmota #14088: after upgrade to Tasmota 10.1.0 - DHT22 don't work

mkonsel, 2021-12-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
after update to 10.0.1.0, the AM2301 don't work. it's a "NULL" value
debug logging:
12:32:35.888 DHT: Timeout waiting for start signal low pulse
the DHT22 is power up with GPIO13 set to "OUTPUT HI": this configuration always works with tasmota 10.0.0 (tasmota32.bin)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wroom32 board
- [ ] Tasmota binary firmware version number used: 10.1.0.1
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:29:11.472 MQT: asciugatrice/stat/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
12:29:11.675 MQT: asciugatrice/stat/RESULT = {"Module":{"1":"ESP32-DevKit"}}
12:29:11.891 MQT: asciugatrice/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"480":"Buzzer"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"1216":"AM2301"},"GPIO13":{"3840":"Output Hi"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"352":"Counter1"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"224":"Relay1"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"4896":"ADC CT Power1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:34:26.397 MQT: asciugatrice/stat/STATUS = {"Status":{"Module":1,"DeviceName":"wroom32","FriendlyName":["wroom32"],"Topic":"asciugatrice","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:34:26.446 MQT: asciugatrice/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"RTC Watch dog reset digital core and rtc module","Uptime":"0T00:20:29","StartupUTC":"2021-12-18T11:13:57","Sleep":50,"CfgHolder":4617,"BootCount":97,"BCResetTime":"2021-10-16T17:19:12","SaveCount":578}}
12:34:26.490 MQT: asciugatrice/stat/STATUS2 = {"StatusFWR":{"Version":"10.1.0.1(tasmota)","BuildDateTime":"2021-12-17T18:51:50","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"499/699"}}
12:34:26.529 MQT: asciugatrice/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wifi82",""],"TelePeriod":900,"Resolution":"558180C0","SetOption":["8000A009","0A05C80001000600003C5A0A000000000000","00800080","00006000","00004000"]}}
12:34:26.582 MQT: asciugatrice/stat/STATUS4 = {"StatusMEM":{"ProgramSize":1315,"Free":1856,"Heap":158,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52,59","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,127"}}
12:34:26.635 MQT: asciugatrice/stat/STATUS5 = {"StatusNET":{"Hostname":"wroom32","IPAddress":"192.168.2.169","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"3C:61:05:16:AE:F8","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:34:26.681 MQT: asciugatrice/stat/STATUS6 = {"StatusMQT":{"MqttHost":"kiwi","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_16AEF8","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:34:26.728 MQT: asciugatrice/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-12-18T11:34:26","Local":"2021-12-18T12:34:26","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"08:39","Sunset":"16:54"}}
12:34:26.814 MQT: asciugatrice/stat/STATUS10 = {"StatusSNS":{"Time":"2021-12-18T12:34:26","COUNTER":{"C1":0},"ANALOG":{"CTEnergy1":{"Energy":0.012,"Power":55,"Voltage":230,"Current":0.240}},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"ESP32":{"Temperature":53.3},"TempUnit":"C"}}
12:34:26.862 MQT: asciugatrice/stat/STATUS11 = {"StatusSTS":{"Time":"2021-12-18T12:34:26","Uptime":"0T00:20:29","UptimeSec":1229,"Heap":156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":35},"POWER":"OFF","Wifi":{"AP":1,"SSId":"wifi82","BSSId":"F0:B0:14:5B:03:00","Channel":1,"Mode":"11n","RSSI":32,"Signal":-84,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
12:32:35.888 DHT: Timeout waiting for start signal low pulse

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
wiring DHT22 sensor to board

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
normally, tasmota display temperature e humidity from DHT22

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mkonsel, 2021-12-18:
Downgrade to release 10.0.0. DHT22 works

sfromis, 2021-12-18:
Just checked, my DHT22 works fine with Tasmota 10.1.0. In my case, I allow it to be powered from the start, instead of powering up a bit later during the boot. 

During normal operation, some timeout messages does appear to be quite common, within a few seconds of enabling logging, I saw 
`DHT: Timeout waiting for start signal high pulse`

mkonsel, 2021-12-18:
with Tasmota 10.1.0 stable release, the DHT work at boot, after 2-3 minutes: null value . With latest developer release 10.1.0.1 , null value immediately after boot

sfromis, 2021-12-18:
Actually, my 10.1 build was specifically 10.1.0.1, and with a new build based on a fresh source download, I'm still seeing the sensor work as expected. As I pointed out already, an obvious difference is that you're delaying giving power to the sensor.

sfromis, 2021-12-18:
My version 10.1.0 download was already 10.1.0.1, but with a fresh source download and build (still 10.1.0.1), I'm now seeing null values too.

sfromis, 2021-12-28:
Playing a bit more with DHT22 due to another issue, I've found that while it has worked nicely with ESP8266, it is at best flaky with ESP32, and during the test with negative temperatures, it completely stopped working, all I could get was the timeout message I already posted. Only after switching to ESP8266, I could run the test to get debug with negative temperatures.

I tend to suspect a timing issue, as the inherent delays due to execution speed will no different from ESP8266 to ESP32.

arendst, 2021-12-29:
I'm afraid we need an ESP32 special driver for the DHTxx sensors like I did for the DS18B20. The timings are very critical so tinkering with the current driver while keeping it functional for the ESP8266 is impossible.

The quick and easy way is porting a DHTxx driver for the ESP32 in the wild to Tasmota.

arendst, 2021-12-29:
I'll have a go at it now.

arendst, 2021-12-29:
Just released a new DHT driver specifically for the ESP32 based on the Adafruit DHT library. If it still doesn't work blame Adafruit ;-)

BTW both drivers work fine with me; they loose a reading once or twice a minute but that shouldn't be a problem for a slow device like a temperature sensor.

sfromis, 2021-12-29:
Now, my DHT22 works a bit less bad on ESP32. Getting lots of "Invalid reading", and the outcome quickly changes to "null" for the values, but I've also seen it return to providing plausible values. Still pretty bad, I blame  the sensor a bit more than Adafruit :-) 

And the debugging messages with incoming data went away.

Jason2866, 2022-01-10:
Please try with latest DHT22 driver for ESP32

Jason2866, 2022-01-16:
Closing since no feedback.

closed 2022-01-16 by Jason2866

estibi, 2022-01-22:
I also get null values on ESP32 for AM2302, 10.1.0(tasmota) release.
"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null}
On ESP8266 the same sensor does work.
Where can I find the latest driver @Jason2866 ?

estibi, 2022-01-22:
I've just connected AM2302 to the ESP32-c3 with 10.1.0 and it does work here.
However it gets null values on ESP-WROOM-32 board.

barbudor, 2022-01-22:
You are using a version which is older than the implementation of the new driver
Use latest development version from http://ota.tasmota.com/tasmota32/