arendst/Tasmota #14090: Shelly 1PM Plus - new hardware revision - tasmota will not boot

AndreasAusD, 2021-12-18:
### PROBLEM DESCRIPTION
trying to flash Shelly 1PM Plus with tasmota 10.1.0, device does not start, compared to the pictures it looks liek a new hardware revision

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _Shelly 1PM Plus____
- [ ] Tasmota binary firmware version number used: _v10.1.0____
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: _esptool.py v3.0___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
flash device, power it up, led does not turn bright but stays dimmed

output from esptool :
```
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-U4WDH (revision 3)
Features: WiFi, BT, Single Core, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 44:17:93:94:c8:b0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 1339632 bytes to 902967...
Wrote 1339632 bytes (902967 compressed) at 0x00000000 in 15.7 seconds (effective 681.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
led on device would light up, new wifi would be visible with name tasmota_xxxxxx

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**
[image]
[image]

Jason2866, 2021-12-18:
what firmware do you have flashed and how?

sfromis, 2021-12-18:
If you flashed it with just `tasmota32.bin`, that's likely the cause of your troubles

AndreasAusD, 2021-12-18:
thank you @sfromis  for pointing me in the right direction.
I remember reading about the additional files the other day but forgot about it today when i tried to flash.

so for reference :
```
sudo esptool.py --port /dev/ttyUSB0 -b 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32solo1.bin
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32-U4WDH (revision 3)
Features: WiFi, BT, Single Core, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 44:17:93:91:b6:d4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15872 bytes to 10320...
Wrote 15872 bytes (10320 compressed) at 0x00001000 in 0.1 seconds (effective 994.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 1540.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4087.7 kbit/s)...
Hash of data verified.
Compressed 1339632 bytes to 902967...
Wrote 1339632 bytes (902967 compressed) at 0x00010000 in 14.3 seconds (effective 749.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```
Power calibration worked ok

only thing left : the main page reads
`ESP32 Temperature  95.6 °C`
which can't be right and leads to overtemp power off
I can adjust that with SetOption42, but can that temperature value be calibrated like voltage and current ?

Jason2866, 2021-12-18:
No, the internal temp sensor can not be calibrated.

closed 2021-12-18 by Jason2866

sfromis, 2021-12-18:
The ESP32 internal temperature readings are crap, my preference is just to shut it off to avoid those irrelevant numbers. 
`SetSensor127 0`

sfromis, 2022-11-30:
You need no additional files for flashing Tasmota to Shelly 1PM plus, as the [http://ota.tasmota.com/tasmota32/tasmota32solo1.factory.bin](http://ota.tasmota.com/tasmota32/tasmota32solo1.factory.bin) image merges the needed files into one. This file can be used for flashing via esptool.py, but the recommended tool for flashing is the [Tasmota Web Installer](https://tasmota.github.io/install/) where you can select the `Tasmota32 Solo1` image.

The reason for even discussing the additional files in this thread was attempting to flash the device without using a `.factory` image, in which case you'd need more files, but it is *really* not recommended to do anything but using a `.factory` image for easy flashing.