arendst/Tasmota #14093: Tilt while moving a blind resets position to zero -> reboot?

s0170071, 2021-12-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Blind position jumps to zero when a tilt command is issued while the blind is moving. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ____shelly25_
- [ x] Tasmota binary firmware version number used: ___10.1.0__
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:04:05.338 MQT: stat/Shelly25RaffsWest/RESULT = {"NAME":"Shelly 2.5 (buttons)","GPIO":[320,0,34,0,224,129,0,0,640,128,608,225,3456,4736],"FLAG":0,"BASE":18}
00:04:05.567 MQT: stat/Shelly25RaffsWest/RESULT = {"Module":{"0":"Shelly 2.5 (buttons)"}}
00:04:05.823 MQT: stat/Shelly25RaffsWest/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"34":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"129":"Button_in2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"128":"Button_in1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ1"},"GPIO17":{"4736":"ADC Temp"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:04:53.922 MQT: stat/Shelly25RaffsWest/STATUS = {"Status":{"Module":0,"DeviceName":"TasmotaRaffWest","FriendlyName":["TasmotaRaffWest",""],"Topic":"Shelly25RaffsWest","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:04:53.959 MQT: stat/Shelly25RaffsWest/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:04:54","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2021-12-14T21:43:26","SaveCount":48,"SaveAddress":"F8000"}}
00:04:53.989 MQT: stat/Shelly25RaffsWest/STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"396/699"}}
00:04:54.002 MQT: stat/Shelly25RaffsWest/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MNET",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A000000000000","40000080","00006000","00004000"]}}
00:04:54.035 MQT: stat/Shelly25RaffsWest/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
00:04:54.069 MQT: stat/Shelly25RaffsWest/STATUS5 = {"StatusNET":{"Hostname":"Shelly25RaffsWest-4307","IPAddress":"192.168.1.121","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"98:F4:AB:F2:D0:D3","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:04:54.096 MQT: stat/Shelly25RaffsWest/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.30","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F2D0D3","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:04:54.120 MQT: stat/Shelly25RaffsWest/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:04:54","Local":"1970-01-01T00:04:54","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:04:54.141 MQT: stat/Shelly25RaffsWest/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:04:54.161 MQT: stat/Shelly25RaffsWest/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:04:54","ANALOG":{"Temperature":54.7},"ENERGY":{"TotalStartTime":"2020-03-08T08:10:59","Total":0.013,"Yesterday":0.000,"Today":0.013,"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Frequency": 0,"Voltage": 0,"Current":[0.000,0.000]},"Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"TempUnit":"C"}}
00:04:54.207 MQT: stat/Shelly25RaffsWest/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:04:54","Uptime":"0T00:04:54","UptimeSec":294,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"MNET","BSSId":"74:42:7F:13:87:0A","Channel":11,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}
00:04:54.238 MQT: stat/Shelly25RaffsWest/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":590,"Close":590,"50perc":50,"Delay":0,"Opt":"0001","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- block external internet access.
- setup venetian blinds according to manual.
- ShutterTiltConfig1"0 0 90 17 0 90
- play with the shutter, move it up and down. At some point you will notice a lag. It will move with a delay. Apparently that's when a time server sync is pending - and failing. 
- move blind to close. 
- move it up. 
- while it moves up, change tilt. That should crash it.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
- either stop and set tilt or complete movement and ignore tilt. 
- if the issue is related to a reboot, moving the shutter all the way up or down for the configured amount of time should fix this.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I use a FritzBox, that does also provide the time. Setting NTP1 to 192.1.6.1.1 apparently fixes the issue.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-12-18:
Tasmota does NOT work correctly if it can't access a NTP server.
As designed.

closed 2021-12-18 by Jason2866

s0170071, 2021-12-19:
It must not crash.
 That's a bug, not by design. I understand that some functions are not available without ntp, but this is different.