arendst/Tasmota #14096: Problem with water leak sensor

holgerschochwitz, 2021-12-18:
Hi,
i have a problem with my warer leak sensor. 
23:09:08.377 MQT: /SonoffZigbeeBridge/898E/tele/SENSOR = {"ZbReceived":{"0x898E":{"Device":"0x898E","ModelId":"lumi.sensor_wleak.aq1","BatteryVoltage":3,"BatteryPercentage":100,**"Water":0**,"Endpoint":1,"LinkQuality":155}}}

The water status is not set when the event occurs, but the "0500_00" was changed (bolded text)

23:10:24.381 ZIG: ZbZCLRawReceived: {"0x898E":{"0500<00":"**01**0000FF0000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":110}}
23:10:24.386 MQT: /SonoffZigbeeBridge/898E/tele/SENSOR = {"ZbReceived":{"0x898E":{"Device":"0x898E","0500<00":"010000FF0000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":110}}}
23:10:33.936 ZIG: {"ZbEZSPReceived":"45000004010005010100010000DEFFE58E89FFFF09191B00000000FF000002"}
23:10:33.938 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0500","srcaddr":"0x898E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":157,"securityuse":0,"seqnumber":222,"fc":"0x19","frametype":1,"direction":1,"disableresp":1,"manuf":"0x0000","transact":27,"cmdid":"0x00","payload":"000000FF0000"}}
23:10:33.941 ZIG: ZbZCLRawReceived: {"0x898E":{"0500<00":"**00**0000FF0000","ZoneStatusChange":0,"Endpoint":1,"LinkQuality":157}}
23:10:33.945 MQT: /SonoffZigbeeBridge/898E/tele/SENSOR = {"ZbReceived":{"0x898E":{"Device":"0x898E","0500<00":"000000FF0000","ZoneStatusChange":0,"Endpoint":1,"LinkQuality":157}}}

closed 2021-12-18 by close-issue-app[bot]

rebel7580, 2021-12-19:
This message:
 ZbZCLRawReceived: {"0x898E":{"0500<00":"010000FF0000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":110}}
is what you need to look at. It's the one that tells you the water leak detector has sensed water.  NOT the SENSOR message with "water":1

rebel7580, 2021-12-19:
"ZoneStatusChange":1  == water now detected,
"ZoneStatusChange":0   == water no longer detected