arendst/Tasmota #14127: SONOFF Dual R3 V2 strange switches behaviour

Jjmphiphi, 2021-12-20:
### PROBLEM DESCRIPTION
when rocker switches are actioned and whatever the SwitchMode is selected , the relays never latch and stays closed for approximately 12 seconds and release.

### REQUESTED INFORMATION

- [ ] Device used: sonoff Dual R3 V2
- [ ] Tasmota binary firmware version number used: Tasmota32.bin  stable version and developpment tasmota32.bin 10.1.0.1
  - Pre-compiled
- [ ] Flashing tools used: ESP32 tool and OTA
- [ ] Provide the output of command: `Backlog
- [ ]  CMD: backlog template
14:34:54.117 SRC: WebConsole from 192.168.1.121
14:34:54.120 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 8, Data 'template'
14:34:54.179 SRC: Backlog
14:34:54.182 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
14:34:54.193 RSL: RESULT = {"NAME":"Sonoff Dual R3 v2","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,3200,8128,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
-[]CMD: module
14:36:59.410 SRC: WebConsole from 192.168.1.121
14:36:59.414 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
14:36:59.419 RSL: RESULT = {"Module":{"0":"Sonoff Dual R3 v2"}}
[]CMD: GPIO 255
14:38:29.173 SRC: WebConsole from 192.168.1.121
14:38:29.176 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
14:38:29.192 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"3200":"Serial Tx"},"GPIO26":{"8128":"BL0939 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"160":"Switch1"},"GPIO33":{"161":"Switch2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
14:38:41.281 WIF: Checking connection...
```lua
  Configuration output:  no load on relays
- [ ] No rules applied, 
- [ ] Provide the output of this command: `Status 0`:
CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
14:49:32.434 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_C6BBF4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,4,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:49:32.449 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:21:43","StartupUTC":"2021-12-20T13:27:49","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2021-12-17T10:58:04","SaveCount":262}}
14:49:32.455 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0.1(tasmota)","BuildDateTime":"2021-12-19T15:59:09","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3","CR":"378/699"}}
14:49:32.462 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Sables",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A000000000000","00000080","00006000","00000002"]}}
14:49:32.483 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1328,"Free":1856,"Heap":160,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","0015A001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,38,50,52,59","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,127"}}
14:49:32.492 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-C6BBF4-7156","IPAddress":"192.168.1.94","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.254","DNSServer2":"0.0.0.0","Mac":"94:3C:C6:C6:BB:F4","Webserver":2,"HTTP_API":0,"WifiConfig":4,"WifiPower":17.0}}
14:49:32.498 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-12-20T13:49:32","Local":"2021-12-20T14:49:32","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:40","Sunset":"16:54"}}
14:49:32.505 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:49:32.525 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-12-20T14:49:32","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2021-12-17T10:58:04","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Voltage": 0,"Current":[0.000,0.000],"BL09XX":{"Temperature":28.6}},"ESP32":{"Temperature":53.3},"TempUnit":"C"}}
14:49:32.542 BRY: GC from 6999 to 3524 bytes, objects freed 11/35 (in 1 ms)
14:49:32.550 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-12-20T14:49:32","Uptime":"0T00:21:43","UptimeSec":1303,"Heap":158,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":35},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Sables","BSSId":"68:A3:78:65:4F:BB","Channel":8,"Mode":"11n","RSSI":26,"Signal":-87,"LinkCount":1,"Downtime":"0T00:00:09"}}}
14:49:36.889 BL9: Checksum failure received 0x0D instead of 0xDD

- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
15:19:32.144 HTP: Main Menu
15:19:36.059 SRC: WebGui from 192.168.1.121
15:19:36.068 RSL: RESULT = {"POWER1":"ON"}
15:19:36.071 RSL: POWER1 = ON
15:19:36.475 CFG: Saved, Count 263, Bytes 4096
15:19:37.847 BL9: Voltage 238.490479, Temp 29.375002
15:19:39.854 BL9: Voltage 238.208176, Temp 29.185270
15:19:41.311 WIF: Checking connection...
15:19:41.841 BL9: Voltage 238.328918, Temp 29.185270
15:19:43.853 BL9: Voltage 238.277588, Temp 29.564735
15:19:45.854 BL9: Voltage 238.361328, Temp 29.185270
15:19:47.845 BL9: Voltage 238.379913, Temp 29.564735
15:19:48.099 SRC: PulseTimer
15:19:48.105 RSL: RESULT = {"POWER1":"OFF"}
15:19:48.108 RSL: POWER1 = OFF
15:19:49.303 CFG: Saved, Count 264, Bytes 4096
15:19:50.548 SRC: WebGui from 192.168.1.121
15:19:50.556 RSL: RESULT = {"POWER2":"ON"}
15:19:50.558 RSL: POWER2 = ON
15:19:51.284 CFG: Saved, Count 265, Bytes 4096
15:19:51.856 BL9: Voltage 238.180893, Temp 29.564735
15:19:53.846 BL9: Voltage 238.585876, Temp 30.323664
15:19:55.863 BL9: Voltage 238.451126, Temp 30.513395
15:19:57.860 BL9: Voltage 238.433060, Temp 29.185270
15:19:59.840 BL9: Voltage 238.477936, Temp 29.944199
15:20:01.317 WIF: Checking connection...
15:20:01.845 BL9: Voltage 238.835831, Temp 29.564735
15:20:02.649 SRC: PulseTimer
15:20:02.655 RSL: RESULT = {"POWER2":"OFF"}
15:20:02.661 RSL: POWER2 = OFF
15:20:03.308 CFG: Saved, Count 266, Bytes 4096
15:20:21.284 WIF: Checking connection...
15:20:34.901 BL9: Checksum failure received 0x00 instead of 0xFC
15:20:35.851 BL9: Checksum failure received 0x3D instead of 0x48
15:20:41.306 WIF: Checking connection...
15:20:56.897 BL9: Checksum failure received 0x3E instead of 0xA5

### TO REPRODUCE
Fairly easy to re produce just wire switches on input of the sonof module (see pictures), and operate switches .after 12 seconds the relays go to "off state". No matter the value of SwitchMode is set. This has been tested on two different unit of Sonoff dual R3 .

### EXPECTED BEHAVIOUR
rocker switches should set the relays On or Off. whatever the value of switchMode is set, the relays are released after 12 seconds. interlock commande doesn't operate neither

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**
![20211219_204425](https://user-images.githubusercontent.com/53942655/146784936-b5b41029-e19a-4246-a980-9145422206fc.jpg)
![20211219_204406](https://user-images.githubusercontent.com/53942655/146784983-020d0506-36a4-4e64-9e57-c752410c5be2.jpg)
![20211219_204219 (1)](https://user-images.githubusercontent.com/53942655/146785054-d2f8b323-33e2-4700-813b-53c7d1d07145.jpg)

sfromis, 2021-12-20:
If you do not want it to switch off automatically, then do not configure `PulseTime 12`. Your output clearly shows that this is what happens. `SRC: PulseTimer`.
You can reset it with `PulseTime 0`

Jjmphiphi, 2021-12-21:
Many tanks for your reply, and sorry for the burden. I'm quite New with
Tamosta.
Thanks again.
Merry Christmas
Jm