arendst/Tasmota #14138: Flash Size to big if you want OTA upgrade to 10.1.0 with actual minimal.bin

laserrichi, 2021-12-21:
### PROBLEM DESCRIPTION

if you want to upgrade OTA with 10.1.0 tasmota-minimal.bin  to 10.1.0 10.1.0 tasmota-DE.bin
is not possible, not enough space for flashing.

### REQUESTED INFORMATION

size of tasmota-minimal.bin and tasmota-DE.bin are over 1024kb 

### TO REPRODUCE
the free Flash wit tasmota-DE.bin is 384 kB
the 10.1.0  tasmota-minimal.bin size is 395k  and its to large.
Use an older tasmota-minimal.bin with 380 kB  and it works

### EXPECTED BEHAVIOUR

### SCREENSHOTS
no screenshot

### ADDITIONAL CONTEXT

sfromis, 2021-12-21:
The only issue you have is that you are not using the `.bin.gz` binary. You should **always** use those for OTA, the `.bin` ones are basically only for initial flashing.

Jason2866, 2021-12-21:
`tasmota-minimal.bin` is removed, since it is obsolete and `tasmota-minimal.bin.gz` has to be used always!

closed 2021-12-21 by Jason2866