arendst/Tasmota #14142: INA219 Power and Voltage only as integer

Radium78, 2021-12-21:
I've added the INA219 sensor to my NodeMCU with tasmota 10.1.0. It works fine, will be found by i2c scan but it shows only the current in float (3 decimals). Power an voltage is only shown as integers (no decimals).

The same issues are on firmware 9.3.1, seems not to be a firmware issue. And I've seen some pictures, where the INA219 sensors drops voltage and power values as floating number.

I've tested all breakout board with an arduino nano. Here i get 

Any ideas what's going wrong?

Here are some pictures...

That's what I can read from INA219 with an arduino nano:
[image]

That's what FHEM has recorded; voltages and power values are only integer steps:
[image]

That's what I get on the webpage:
[image]

closed 2021-12-21 by close-issue-app[bot]

barbudor, 2021-12-21:
https://tasmota.github.io/docs/Commands/#power-monitoring

VoltRes and the like are your friend

Radium78, 2021-12-21:
Thanks... that's the missing info I need ;)