arendst/Tasmota #14146: "IRSend":"Invalid RawData" error for rule based commands where same command is working console

github768484, 2021-12-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Each of the following commands work just find in the console
```
IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF7C03F"}
IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF7E01F"}
IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF740BF"}
```
However, the same commands crank out this error within a rule 
"IRSend":"Invalid RawData"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
Rule1 ON Event#ledson
        DO Backlog IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF7C03F"};
        Delay 5;
        IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF7E01F"};
        Publish stat/IRkitchen/stat ON
        endon
ON Event#ledsoff
        IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF740BF"};
        Publish stat/IRkitchen/stat OFF
        endon 

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:09:47.199 CMD: rule1"
15:09:47.201 SRC: WebConsole from 192.168.3.102
15:09:47.203 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 1, Data '"'
15:09:47.209 MQT: stat/IRkitchen/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:09:47.985 CFG: Saved to flash at F4, Count 40, Bytes 4096
15:09:48.602 CMD: Rule1 ON Event#ledson DO Backlog IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF7C03F"}; Delay 5; IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF7E01F"}; Publish stat/IRkitchen/stat ON endon ON Event#ledsoff IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF740BF"}; Publish stat/IRkitchen/stat OFF endon Rule1 1
15:09:48.625 SRC: WebConsole from 192.168.3.102
15:09:48.628 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 300, Data 'ON Event#ledson DO Backlog IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF7C03F"}; Delay 5; IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF7E01F"}; Publish stat/IRkitchen/stat ON endon ON Event#ledsoff IRSend {"Protocol":"NEC","Bits":32,"Data":"0xF740BF"}; Publish stat/IRkitchen/stat OFF endon Rule1 1'
15:09:48.640 RUL: Stored uncompressed, would compress from 300 to 148 (-51%)
15:09:48.646 MQT: stat/IRkitchen/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":300,"Free":211,"Rules":"ON Event#ledson DO Backlog IRSend {\"Protocol\":\"NEC\",\"Bits\":32,\"Data\":\"0xF7C03F\"}; Delay 5; IRSend {\"Protocol\":\"NEC\",\"Bits\":32,\"Data\":\"0xF7E01F\"}; Publish stat/IRkitchen/stat ON endon ON Event#ledsoff IRSend {\"Protocol\":\"NEC\",\"Bits\":32,\"Data\":\"0xF740BF\"}; Publish stat/IRkitchen/stat OFF endon Rule1 1"}}
15:09:48.008 CFG: Saved to flash at FB, Count 41, Bytes 4096
15:09:53.435 SRC: MQTT
15:09:53.438 CMD: Grp 0, Cmnd 'IRSEND', Idx 1, Len 6, Data 'ledson'
15:09:53.444 MQT: stat/IRkitchen/RESULT = {"IRSend":"Invalid RawData"}
15:09:55.499 SRC: MQTT
15:09:55.502 CMD: Grp 0, Cmnd 'IRSEND', Idx 1, Len 7, Data 'ledsoff'
15:09:55.508 MQT: stat/IRkitchen/RESULT = {"IRSend":"Invalid RawData"}`
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Add the Rule
issue the following MQTT command 
cmnd/IRkitchen/IRsend 
ledson or ladsoff

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The goal is to send the following MQTT message to my IR Controller for the kitchen LED lights with a simple switch

Topic: cmnd/IRkitchen/event
Payload: ledson or ledsoff

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-12-22:
This shows that the commands you send via MQTT are just bad.
```
15:09:53.435 SRC: MQTT
15:09:53.438 CMD: Grp 0, Cmnd 'IRSEND', Idx 1, Len 6, Data 'ledson'
15:09:53.444 MQT: stat/IRkitchen/RESULT = {"IRSend":"Invalid RawData"}
```
You are sending the command `IRSEND ledson` which cannot work.

Jason2866, 2021-12-22:
Closing. Wrong rule syntax. Discussed and solved in Discord

closed 2021-12-22 by Jason2866