arendst/Tasmota #14163: DS18x20 with ESP32

dknubben, 2021-12-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
No detection of DS18B20 with ESP32 devicess and builds

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): ESP32-DevKit, wemos D1
- [x ] Tasmota binary firmware version number used: 10.1.0.1(tasmota)
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool.py and ESP-Flasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
04:08:59.195 CMD: Backlog Template; Module; GPIO 255
04:08:59.198 SRC: WebConsole from 192.168.0.5
04:08:59.199 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
04:08:59.238 SRC: Backlog
04:08:59.239 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
04:08:59.247 MQT: stat/tasmota_CA4BB4/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
04:08:59.473 SRC: Backlog
04:08:59.474 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
04:08:59.482 MQT: stat/tasmota_CA4BB4/RESULT = {"Module":{"1":"ESP32-DevKit"}}
04:08:59.675 SRC: Backlog
04:08:59.677 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
04:08:59.690 MQT: stat/tasmota_CA4BB4/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"1312":"DS18x20"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"640":"I2C SDA1"},"GPIO27":{"608":"I2C SCL1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
04:08:11.491 CMD: STATUS 0
04:08:11.492 SRC: WebConsole from 192.168.0.5
04:08:11.493 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
04:08:11.500 MQT: stat/tasmota_CA4BB4/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_CA4BB4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
04:08:11.540 MQT: stat/tasmota_CA4BB4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:01:59","StartupUTC":"2021-12-25T03:06:12","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2021-12-24T17:12:00","SaveCount":20}}
04:08:11.581 MQT: stat/tasmota_CA4BB4/STATUS2 = {"StatusFWR":{"Version":"10.1.0.1(tasmota)","BuildDateTime":"2021.12.23 00:29:04","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6","CR":"433/699"}}
04:08:11.601 MQT: stat/tasmota_CA4BB4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["home_1.dd",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","01000080","00006000","00004000"]}}
04:08:11.636 MQT: stat/tasmota_CA4BB4/STATUS4 = {"StatusMEM":{"ProgramSize":1833,"Free":1856,"Heap":104,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000407","9FDAC787","0015A001","FFFFBFCF","05DB9BC0","60BE0DD7","000A40D2","20200000","0000482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,32,34,38,41,50,52,59,79","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,24,26,27,28,31,34,37,39,40,42,43,45,51,52,54,55,56,57,58,59,64,66,67,74,85,92,127"}}
04:08:11.691 MQT: stat/tasmota_CA4BB4/STATUS5 = {"StatusNET":{"Hostname":"tasmota-CA4BB4-2996","IPAddress":"192.168.0.29","Gateway":"192.168.0.11","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.10","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:CA:4B:B4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
04:08:11.722 MQT: stat/tasmota_CA4BB4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.72","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CA4BB4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
04:08:11.756 MQT: stat/tasmota_CA4BB4/STATUS7 = {"StatusTIM":{"UTC":"2021-12-25T03:08:11","Local":"2021-12-25T04:08:11","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"08:09","Sunset":"16:35"}}
04:08:11.783 MQT: stat/tasmota_CA4BB4/STATUS10 = {"StatusSNS":{"Time":"2021-12-25T04:08:11","BME280":{"Temperature":25.1,"Humidity":38.2,"DewPoint":9.8,"Pressure":932.5},"ESP32":{"Temperature":43.3},"PressureUnit":"hPa","TempUnit":"C"}}
04:08:11.799 BRY: GC from 7182 to 3853 bytes, objects freed 12/39 (in 1 ms)
04:08:11.807 MQT: stat/tasmota_CA4BB4/STATUS11 = {"StatusSTS":{"Time":"2021-12-25T04:08:11","Uptime":"0T00:01:59","UptimeSec":119,"Heap":103,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":39},"Wifi":{"AP":1,"SSId":"home_1.dd","BSSId":"38:D5:47:BB:73:6A","Channel":9,"Mode":"11n","RSSI":100,"Signal":-28,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.003-165/62 HDW: ESP32-D0WDQ6 
00:00:00.004-165/62 HDW: FoundPSRAM=0 CanUsePSRAM=0
00:00:00.028-163/61 UFS: FlashFS mounted with 304 kB free
00:00:00.076 CFG: Loaded from File, Anzahl 19
00:00:00.084 QPC: Count 1
00:00:00.086 CFG: CR 433/699, Busy 0
00:00:00.092 ROT: Mode 1
00:00:00.093 BRY: GC from 1435 to 1435 bytes, objects freed 0/1 (in 0 ms)
[... 40 lines cut ...]
04:06:25.058 MQT: tele/tasmota_CA4BB4/BLE = {"Time":"2021-12-25T04:06:25","BLEDevices":{"total":0}}
04:06:25.064 MQT: tele/tasmota_CA4BB4/BLE = {"Time":"2021-12-25T04:06:25","BLE":{"scans":0,"adverts":0,"devices":0,"resets":0}}
04:06:26.137 HTP: Konsole
04:06:29.950 MQT: tasmota/discovery/C8C9A3CA4BB4/config = {"ip":"192.168.0.29","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"tasmota-CA4BB4-2996","mac":"C8C9A3CA4BB4","md":"ESP32-DevKit","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"10.1.0.1","t":"tasmota_CA4BB4","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1} (beibehalten)
04:06:29.965 MQT: tasmota/discovery/C8C9A3CA4BB4/sensors = {"sn":{"Time":"2021-12-25T04:06:29","BME280":{"Temperature":24.9,"Humidity":38.2,"DewPoint":9.6,"Pressure":932.5},"ESP32":{"Temperature":42.8},"PressureUnit":"hPa","TempUnit":"C"},"ver":1} (beibehalten)
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
[... cut ...]
_A clear and concise description of what you expected to happen._
Shows the temperatur(s) of the sensors

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
With ESP8266EX (wemos D1 pro/non pro) Tasmota 10.1.0.1 (sensors) DS18B20 sensors are recognized, I2C (eg. BME280) works with ESP32 and normal GPIO (eg. relay)
Tests with ESP32 Devkit V1/V4, ESP32S2 (webmos), no success

**(Please, remember to close the issue when the problem has been addressed)**
yes

sfromis, 2021-12-25:
I just did a small test with 10.1.0.2, hooked a single DS18B20 up to gpio 19 (same pin as you used) on a dev board, including a pull-up resistor (my experience from earlier is that SetOption74 did not have any effect). Result:
`"DS18B20":{"Id":"030B97942847","Temperature":22.37}`

Jason2866, 2021-12-25:
Sensor Hardware issue. Many bad fake clones...

closed 2021-12-25 by Jason2866

dknubben, 2021-12-26:
Thanks for testing and sorry for makeing confusion. On ESP it's working with the resistor (SetOption74 on ESP32 has no effect, that was new for me), but still no DS18B20 on ESP32S2