arendst/Tasmota #14165: Problem with Sensor DHT22 when Temperature is below 0 degrees Celsius

flinty32, 2021-12-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Using the ESP 8266, WEMOS D1 and other ESP with ESP8266EX on Tasmota 10.1.0 (sensors)
If Temperatur is greater then 0 degrees Celsius the Tasmota Displays the correct Temp. with DHT22 (AM2301) It works fine.
But if the Temperature is below 0 Gegrees C It Display wrong Temperature : AM2301 Temperature -3276.3 °C

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP8266EX
- [X] Tasmota binary firmware version number used: 10.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: NodeMCU-PyFlasher-4.0-x86.exe
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
  00:25:07.421 RSL: SENSOR = {"Time":"1970-01-01T00:25:07","AM2301":{"Temperature":-3276.2,"Humidity":95.7,"DewPoint":-3373.4},"BME280":{"Temperature":0.9,"Humidity":100.0,"DewPoint":0.9,"Pressure":980.1,"SeaPressure":1004.2},"BH1750":{"Illuminance":0},"PressureUnit":"hPa","TempUnit":"C"}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
  00:25:07.421 RSL: SENSOR = {"Time":"1970-01-01T00:25:07","AM2301":{"Temperature":-3276.2,"Humidity":95.7,"DewPoint":-3373.4},"BME280":{"Temperature":0.9,"Humidity":100.0,"DewPoint":0.9,"Pressure":980.1,"SeaPressure":1004.2},"BH1750":{"Illuminance":0},"PressureUnit":"hPa","TempUnit":"C"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
connect a DHT22 sensor to the board. And look if Temperate is below Zero Celsius.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
If Temperatue is below 0 there was Display the wrong Temperature

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-12-25:
Try change line 152 in xsns_06_dht.ino from
`temperature = ((int16_t)(dht_data[2] & 0x7F) << 8 ) | dht_data[3];` to
`temperature = ((int16_t)(dht_data[2] << 8 ) | dht_data[3];`

sfromis, 2021-12-27:
Debugging output was added to the DHT driver https://github.com/arendst/Tasmota/commit/cb52c4cf35c06afe07fbbb91f1abd992cc02ced8

Try running with `Weblog 4` for both positive and negative temperatures, to get details of what happens.

Hooking a DHT22 up to an ESP8266, I was able to have negative temperatures work as expected.
`"Temperature":0.20 ` is parsed from `DHT: Read 0214000218`
`"Temperature":-0.20` is parsed from `DHT: Read 02208002A4`

With the two first bytes being RH, the next two being temperature, and the last byte being checksum, the handing of negative where the first bit of the temperature indicates negative, without using two's complement, is consistent with the coding in the driver, and as I'm getting sensible temperature readouts, it works for me.

Wondering if your DHT22 has a different way of sending negative temperatures. Since you see a "temperature" of 3275.5 this would correspond to a binary value of `0x7ff3`. Assuming a two's complement negative sign bit, this becomes `0xfff3` which is decimal -1.3, which could be a "plausible" temperature from your sensor. 

To me, this looks like a very different number representation than the original sensor, meaning that it is effectively a different sensor, needing a different algorithm for parsing temperatures.

What A Mess.

flinty32, 2021-12-27:
I am not a good Engineer, to make self the bin flashfiles for esp8266. 
Has everyone a new Bin File for (ESP 8266) Beta Tasmota 10.x.x  where the new Feature: Debugging output was added to the DHT driver cb52c4c was insert? 
Then I can test it new, and give feedback about and can report new WEBLogs.  

Additional Info to the Sensors:
All my DHT 22 Sensors are bougt from Reichelt Elektronic in Germany.  Called : DEBO DHT 22
And all they have the same Problem with it. It is not a defect on Sensor.

barbudor, 2021-12-27:
Development branch is automatically compiled and publish on http://ota.tasmota.com/tasmota a few hours after commit.

flinty32, 2021-12-27:
Ok, new Debug output with Tasmota 10.1.0.3:
14:00:07.406 DHT: Read 039B001DBB
14:00:08.419 RSL: SENSOR = {"Time":"2021-12-27T14:00:08","AM2301":{"Temperature":2.9,"Humidity":92.3,"DewPoint":1.8}
14:00:09.430 DHT: Read 039E001DBE
__________________________________________
14:03:55.403 DHT: Read 0233FFF72B
14:05:08.420 RSL: SENSOR = {"Time":"2021-12-27T14:05:08","AM2301":{"Temperature":-3275.3,"Humidity":52.9,"DewPoint":-5755.5}
14:05:09.431 DHT: Read 0211FFF103
14:05:11.433 DHT: Read 0210FFF102

barbudor, 2021-12-27:
Looks like @sfromis is right, some devices just use bit 31 set to one to specify a negatove value, so use 2-complements.
That should be fairly easy to support both for example by comparing the top nibble to 0xF
I can do it tonight if no-one take it before

barbudor, 2021-12-27:
@flinty32 @sfromis 
Could you please test with the attached tasmota-sensor ? I don't have a DHT with me but simulation looks ok.
[test_DHT.zip](https://github.com/arendst/Tasmota/files/7780599/test_DHT.zip)
If it works for both of you, I will PR

sfromis, 2021-12-27:
Looks good for my specs-compliant sensor.
```
15:53:57.512 DHT: Read 01C10002C4 "Temperature":0.2
15:54:27.500 DHT: Read 01C880024B "Temperature":-0.2
```

sfromis, 2021-12-27:
Unless the sensor is broken, there should be no risk of the `F` in first nibble changing for negative temperatures, as it would need to be below absolute zero for that. And the sensor does not go near as low anyway.

flinty32, 2021-12-27:
> @flinty32 @sfromis Could you please test with the attached tasmota-sensor ? I don't have a DHT with me but simulation looks ok. [test_DHT.zip](https://github.com/arendst/Tasmota/files/7780599/test_DHT.zip) If it works for both of you, I will PR

With new Version it will work fine. 
16:11:34.413 RSL: SENSOR = {"Time":"2021-12-27T16:11:34","AM2301":{"Temperature":-4.8,"Humidity":92.9,"DewPoint":-5.8}
16:11:35.426 DHT: Read 03A2FFD074
16:11:37.403 DHT: Read 03A2FFD175
16:11:39.468 DHT: Read 03A2FFD276

barbudor, 2021-12-27:
Thanks both of you
Wait for https://github.com/arendst/Tasmota/pull/14173 to be merged and new development version to be available on http://ota.tasmota.com/tasmota

sfromis, 2021-12-27:
Merged now, and new development binaries available

closed 2021-12-27 by ascillato2