arendst/Tasmota #14167: Status5 (NET) for ESP32 with ethernet and wifi should return data for two networks

urechi, 2021-12-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

urechi, 2021-12-27:
For `Status5` command, this is what we get now:
```
{
   "StatusNET":{
      "Hostname":"**B9**D2**40-4928-eth",
      "IPAddress":"192.168.1.35",
      "Gateway":"192.168.1.1",
      "Subnetmask":"255.255.255.0",
      "DNSServer1":"192.168.1.1",
      "DNSServer2":"0.0.0.0",
      "Mac":"**:B9:**:D2:**:43",
      "Webserver":2,
      "HTTP_API":1,
      "WifiConfig":4,
      "WifiPower":17.0
   }
}
```

And this is what would be nice to get in the future:
```
{
   "StatusNET":{
       "eth": {
          "Hostname":"**B9**D2**40-4928-eth",
          "IPAddress":"192.168.1.35",
          "Gateway":"192.168.1.1",
          "Subnetmask":"255.255.255.0",
          "DNSServer1":"192.168.1.1",
          "DNSServer2":"0.0.0.0",
          "Mac":"**:B9:**:D2:**:43"
      },
      "wifi": {
          "Hostname":"**B9**D2**42-4928-wifi",
          "IPAddress":"192.168.1.36",
          "Gateway":"192.168.1.1",
          "Subnetmask":"255.255.255.0",
          "DNSServer1":"192.168.1.1",
          "DNSServer2":"0.0.0.0",
          "Mac":"**:B9:**:D2:**:42"
      },
          "Webserver":2,
          "HTTP_API":1,
          "WifiConfig":4,
          "WifiPower":17.0,
          "ExternalIPAddress: "1.1.1.1"
   }
}
```

Please note the ExternalIPAddress too.

A huge thank you to everyone contributing to this project, you are all rock stars!

ascillato2, 2021-12-27:
Thanks for the idea.

Moving to ideas in discussions due to this is a feature request and not a bug report.