arendst/Tasmota #14171: Shelly RGBW2 on Tasmota won't go back to full brightness after being dimmed over MQTT

niwla23, 2021-12-27:
### PROBLEM DESCRIPTION
After dimming the Shelly RGBW2 over MQTT through openHAB to 0 and bringing it back up to a 100% the light is still heavily dimmed. (Maybe 5% Brightness or so). I am also unable to restore the original brightness throught the WebUI, but changing color works. The only way to get it back to normal brightness is to restart it.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly RGBW2
- [ ] Tasmota binary firmware version number used: 10.1.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: [mgos-to-tasmota](https://github.com/yaourdt/mgos-to-tasmota)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
13:44:54.477 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/RESULT = {"NAME":"ShellyRGBW2","GPIO":[0,0,544,0,419,199,0,0,416,32,418,417,0,4896],"FLAG":0,"BASE":18}
13:44:54.682 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/RESULT = {"Module":{"0":"ShellyRGBW2"}}
13:44:54.899 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"544":"LedLink"},"GPIO3":{"0":"None"},"GPIO4":{"419":"PWM4"},"GPIO5":{"199":"Switch_n8"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"416":"PWM1"},"GPIO13":{"32":"Button1"},"GPIO14":{"418":"PWM3"},"GPIO15":{"417":"PWM2"},"GPIO16":{"0":"None"},"GPIO17":{"4896":"ADC CT Power"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:45:49.463 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/STATUS = {"Status":{"Module":0,"DeviceName":"shelly-rgbw2-alwin-shelf-back","FriendlyName":["shelly-rgbw2-alwin-shelf-back"],"Topic":"shelly-rgbw2-alwin-shelf-back","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:45:49.473 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:07","StartupUTC":"2021-12-27T12:35:42","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2021-12-19T00:28:17","SaveCount":551,"SaveAddress":"FB000"}}
13:45:49.481 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/STATUS2 = {"StatusFWR":{"Version":"10.1.0.3(tasmota)","BuildDateTime":"2021-12-26T13:01:04","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"437/699"}}
13:45:49.489 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
13:45:49.503 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
13:45:49.512 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/STATUS5 = {"StatusNET":{"Hostname":"shelly-rgbw2-alwin-shelf-back-7","IPAddress":"192.168.178.134","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"94:B9:7E:00:BE:4A","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:45:49.519 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.33","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_00BE4A","MqttUser":"shelly-rgbw2-alwin-desk","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:45:49.528 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-12-27T12:45:49","Local":"2021-12-27T13:45:49","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:59"}}
13:45:49.575 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/STATUS10 = {"StatusSNS":{"Time":"2021-12-27T13:45:49","Switch8":"ON","ANALOG":{"CTEnergy":{"Energy":0.001,"Power":5,"Voltage":230,"Current":0.021}}}}
13:45:49.587 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/STATUS11 = {"StatusSTS":{"Time":"2021-12-27T13:45:49","Uptime":"0T00:10:07","UptimeSec":607,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":103,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"89FF1A00","HSBColor":"91,90,100","White":0,"Channel":[54,100,10,0],"Scheme":0,"Fade":"OFF","Speed":10,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lohrie","BSSId":"B0:BE:76:9D:EC:9A","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:03"}}}
13:45:51.414 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/tele/STATE = {"Time":"2021-12-27T13:45:51","Uptime":"0T00:10:09","UptimeSec":609,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"89FF1A00","HSBColor":"91,90,100","White":0,"Channel":[54,100,10,0],"Scheme":0,"Fade":"OFF","Speed":10,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lohrie","BSSId":"B0:BE:76:9D:EC:9A","Channel":11,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:45:51.464 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/tele/SENSOR = {"Time":"2021-12-27T13:45:51","Switch8":"ON","ANALOG":{"CTEnergy":{"Energy":0.001,"Power":5,"Voltage":230,"Current":0.021}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:48:34.508 SRC: MQTT
13:48:34.511 CMD: Grp 0, Cmnd 'HSBCOLOR', Idx 1, Len 8, Data '59,100,0'
13:48:34.515 SRC: Light
13:48:34.523 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/RESULT = {"POWER":"OFF","Dimmer":0,"Color":"00000000","HSBColor":"59,100,0","White":0,"Channel":[0,0,0,0]}
13:48:34.586 CFG: Saved to flash at F6, Count 564, Bytes 4096
13:48:34.597 SRC: MQTT
13:48:34.599 CMD: Grp 0, Cmnd 'SCHEME', Idx 1, Len 1, Data '1'
13:48:34.607 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/RESULT = {"Scheme":1}
13:48:34.947 SRC: MQTT
13:48:34.950 CMD: Grp 0, Cmnd 'HSBCOLOR', Idx 1, Len 8, Data '59,100,0'
13:48:34.958 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/RESULT = {"POWER":"OFF","Dimmer":0,"Color":"00000000","HSBColor":"59,100,0","White":0,"Channel":[0,0,0,0]}
13:48:34.997 SRC: MQTT
13:48:34.000 CMD: Grp 0, Cmnd 'SCHEME', Idx 1, Len 1, Data '1'
13:48:35.001 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/RESULT = {"Scheme":1}
13:48:35.595 CFG: Saved to flash at F5, Count 565, Bytes 4096
13:48:40.076 SRC: MQTT
13:48:40.078 CMD: Grp 0, Cmnd 'HSBCOLOR', Idx 1, Len 9, Data '59,100,62'
13:48:40.082 SRC: Light
13:48:40.090 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/RESULT = {"POWER":"ON","Dimmer":0,"Color":"00000000","HSBColor":"59,100,0","White":0,"Channel":[0,0,0,0]}
13:48:40.103 SRC: MQTT
13:48:40.105 CMD: Grp 0, Cmnd 'SCHEME', Idx 1, Len 1, Data '1'
13:48:40.114 MQT: alwin/tasmota/shelly-rgbw2-alwin-shelf-back/stat/RESULT = {"Scheme":1}
13:48:40.555 CFG: Saved to flash at F4, Count 566, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
flash shelly rgbw2, send 0 brightness command with mqtt and try to get it back to full brightness

### EXPECTED BEHAVIOUR
come back to full brightness

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-12-27:
I see only the command `scheme 1` sent to the device.

sfromis, 2021-12-27:
HSB means **H**ue **S**aturation **B**rightness
When I on a RGBCW bulb (PWM channels) go from `HSBColor 66,66,66` `HSBColor 66,66,99`, I stay at the same green color, as H/S is the same, but the brightness changes. This is reflected in how RGB and channels change to match the brightness.
```
HSBColor 66,66,66
{"POWER1":"ON","Dimmer1":66,"POWER2":"OFF","Dimmer2":100,"Color":"156,168,57,84,171","HSBColor":"66,66,66","White":100,"CT":386,"Channel":[61,66,22,33,67]}
HSBColor 66,66,99
{"POWER1":"ON","Dimmer1":99,"POWER2":"OFF","Dimmer2":100,"Color":"234,252,86,84,171","HSBColor":"66,66,99","White":100,"CT":386,"Channel":[92,99,34,33,67]}
```
Repeating the test after setting `Scheme 1` produced the same results, as the scheme is reset to 0, maybe due to me not having worked with settings for wakeup.
Using `HSBColor 66,66,0` resets the RGB channels to 0, and the prior brighess is "resurrected" after setting the B of RGBColor again.
```
HSBColor 66,66,0
{"POWER1":"OFF","Dimmer1":0,"POWER2":"OFF","Dimmer2":99,"Color":"0,0,0,83,169","HSBColor":"66,66,0","White":99,"CT":386,"Channel":[0,0,0,32,66]}
HSBColor 66,66,99
{"POWER1":"ON","Dimmer1":99,"POWER2":"OFF","Dimmer2":99,"Color":"234,252,86,83,169","HSBColor":"66,66,99","White":99,"CT":386,"Channel":[92,99,34,32,66]}
```
Having no Shelly RGW2, I can't do a specific test, but PWM devices tend to rather similar, and there seems to be no special handling for the Shelly device.

sfromis, 2021-12-27:
What looks odd in your weblog output is that setting HSBColor with a non-zero B still comes back with a B of zero. Could not repeat that.

ascillato2, 2022-01-10:
Hi, any news on this?