arendst/Tasmota #14190: MQTT TH postings

jeschofnik, 2021-12-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

temperature + humidity sensor regularly sends zero values via mqtt on a sonoff th16. Sensor is an original sonoff am2301 (conf as a SI7021).

I've already tried to solder the sensor directly to the pcb to rule out the plug. No difference

13:46:20.424 DHT: Timeout waiting for start signal low pulse

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)  --- talked to sfromis
- [ ] Device used (e.g., Sonoff Basic): Sonoff TH16
- [ ] Tasmota binary firmware version number used: 10.1.0.3
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:53:46.756 CMD: Backlog Template; Module; GPIO 255
13:53:47.070 MQT: domoticz/in/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
13:53:47.285 MQT: domoticz/in/RESULT = {"Module":{"4":"Sonoff TH"}}
13:53:47.545 MQT: domoticz/in/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"1248":"SI7021"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
13:54:39.281 CMD: Backlog Rule1; Rule2; Rule3
13:54:39.549 MQT: domoticz/in/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:54:39.790 MQT: domoticz/in/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:54:40.041 MQT: domoticz/in/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
`13:55:21.967 CMD: Status 0
13:55:21.977 MQT: domoticz/in/STATUS = {"Status":{"Module":4,"DeviceName":"Steckdose1","FriendlyName":["Steckdose1"],"Topic":"tasmota_9F84B6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:55:21.985 MQT: domoticz/in/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:12:34","StartupUTC":"2021-12-28T12:42:47","Sleep":50,"CfgHolder":4617,"BootCount":48,"BCResetTime":"2021-11-30T14:44:14","SaveCount":342,"SaveAddress":"F8000"}}
13:55:21.993 MQT: domoticz/in/STATUS2 = {"StatusFWR":{"Version":"10.1.0.3(tasmota)","BuildDateTime":"2021-12-28T11:24:38","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"326/699"}}
13:55:22.000 MQT: domoticz/in/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["als7",""],"TelePeriod":180,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
13:55:22.013 MQT: domoticz/in/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
13:55:22.019 MQT: domoticz/in/STATUS5 = {"StatusNET":{"Hostname":"tasmota-9F84B6-1206","IPAddress":"192.168.0.183","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:9F:84:B6","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:55:22.025 MQT: domoticz/in/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.42","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F84B6","MqttUser":"phil","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:55:22.034 MQT: domoticz/in/STATUS7 = {"StatusTIM":{"UTC":"2021-12-28T12:55:22","Local":"2021-12-28T13:55:22","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:59"}}
13:55:22.042 MQT: domoticz/in/STATUS10 = {"StatusSNS":{"Time":"2021-12-28T13:55:22","SI7021":{"Temperature":24.3,"Humidity":21.4,"DewPoint":0.8},"TempUnit":"C"}}
13:55:22.050 MQT: domoticz/in/STATUS11 = {"StatusSTS":{"Time":"2021-12-28T13:55:22","Uptime":"0T00:12:35","UptimeSec":755,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"als7","BSSId":"78:8A:20:2A:71:9A","Channel":6,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:04"}}}`

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

13:56:54.456 DHT: Read 00D900F1CA
--> 13:56:56.457 DHT: Timeout waiting for start signal low pulse
13:56:56.459 DHT: Read 00D900F1CA`
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

connect a sensor^^

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
transmit the right TH data or just no null values via mqtt (retain old value if it snt older than one hour or so)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-12-28:
I'm seeing a similar behavior from the same DHT driver, while getting "readings" of null, I'm seeing timeout messages, and reasonably-looking debug data.

ascillato2, 2021-12-28:
Hi,

This is the same issue as in https://github.com/arendst/Tasmota/issues/14088

Let's continue there. Thanks.

closed 2021-12-28 by ascillato2