arendst/Tasmota #14197: ESP32 devkit hangs at boot when US-100 plugged in and configured

LoloTheJeeper, 2021-12-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Hi ! I have been experimenting for the few last days with a US-100 (which is basically a SR04 distance sensor working in 3.3v).
I use it in HC-SR04 mode (jumper removed).
I have connected 3.3v and GND on a separate power supply. The ground of the ESP devkit and of the power supply are connected together.
I plugged the Echo Pin on GPIO19 and Trig on GPIO18
When I boot the ESP32, it hangs here :
```lua
`Using 'COM3' as serial port.
Showing logs:
[15:50:34]x00,hd_drv:0
[15:50:34]00:00:00.003-217/51 HDW: ESP32-D0WDQ6 
[15:50:35]00:00:00.077-215/50 UFS: FlashFS mounted with 304 kB free
[15:50:35]00:00:00.180 CFG: Loaded from File, Count 130
[15:50:35]00:00:00.196 QPC: Count 2
[15:50:35]00:00:00.287 BRY: Berry initialized, RAM used=3813
[15:50:35]00:00:00.313 BRY: no 'preinit.be'
[15:50:35]00:00:00.321 Project tasmota - Tasmota_Sensor Version 10.1.0.3(tasmota)-2_0_2(2021-12-26T13:01:18)
[15:50:35]00:00:00.536 BRY: no 'autoexec.be'`
```

The esp32 hangs until I unplug one of the three connections from the US-100:
- GND
- Trig
- Echo

Right after unplugging one of the pins above, the line "SR4: Mode 1" appears on the serial log 
```lua
`Using 'COM3' as serial port.
Showing logs:
[15:50:34]x00,hd_drv:0
[15:50:34]00:00:00.003-217/51 HDW: ESP32-D0WDQ6 
[15:50:35]00:00:00.077-215/50 UFS: FlashFS mounted with 304 kB free
[15:50:35]00:00:00.180 CFG: Loaded from File, Count 130
[15:50:35]00:00:00.196 QPC: Count 2
[15:50:35]00:00:00.287 BRY: Berry initialized, RAM used=3813
[15:50:35]00:00:00.313 BRY: no 'preinit.be'
[15:50:35]00:00:00.321 Project tasmota - Tasmota_Sensor Version 10.1.0.3(tasmota)-2_0_2(2021-12-26T13:01:18)
[15:50:35]00:00:00.536 BRY: no 'autoexec.be'
[15:53:15]00:02:40.355 SR4: Mode 1`
```

and after 30 seconds, the boot resumes...

If I reconnect the pin I disconnected, the sensor works perfectly until the next reboot.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _ESP32 devkit 30 pins____
- [X] Tasmota binary firmware version number used: __10.1.0.3___
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: _ESP_Flasher____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:04:56.590 CMD: Backlog Template; Module; GPIO 255
22:04:56.656 MQT: stat/sensor/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
22:04:56.978 MQT: stat/sensor/RESULT = {"Module":{"0":"ESP32-DevKit"}}
22:04:57.237 MQT: stat/sensor/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"1856":"SR04 Tri/TX"},"GPIO19":{"1888":"SR04 Ech/RX"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:05:31.451 CMD: status 0
22:05:31.465 MQT: stat/sensor/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota_Sensor","FriendlyName":["Tasmota_Sensor"],"Topic":"sensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:05:31.509 MQT: stat/sensor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:01:04","StartupUTC":"2021-12-28T21:04:27","Sleep":50,"CfgHolder":4617,"BootCount":57,"BCResetTime":"2021-12-26T21:31:53","SaveCount":135}}
22:05:31.552 MQT: stat/sensor/STATUS2 = {"StatusFWR":{"Version":"10.1.0.3(tasmota)","BuildDateTime":"2021-12-26T13:01:18","Core":"2_0_2","SDK":"v4.4-2.0.2","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"391/699"}}
22:05:31.575 MQT: stat/sensor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Get Out",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
22:05:31.619 MQT: stat/sensor/STATUS4 = {"StatusMEM":{"ProgramSize":1336,"Free":1856,"Heap":156,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","0015A001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,38,50,52,59","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,127"}}
22:05:31.677 MQT: stat/sensor/STATUS5 = {"StatusNET":{"Hostname":"sensor-3864","IPAddress":"192.168.88.55","Gateway":"192.168.88.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.88.1","DNSServer2":"192.168.88.1","Mac":"24:6F:28:B1:8F:18","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:05:31.714 MQT: stat/sensor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.88.110","MqttPort":1883,"MqttClientMask":"sensor","MqttClient":"sensor","MqttUser":"homemqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:05:31.739 MQT: stat/sensor/STATUS7 = {"StatusTIM":{"UTC":"2021-12-28T21:05:31","Local":"2021-12-28T22:05:31","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:59"}}
22:05:31.771 MQT: stat/sensor/STATUS10 = {"StatusSNS":{"Time":"2021-12-28T22:05:31","SR04":{"Distance":0.158},"ESP32":{"Temperature":35.0},"TempUnit":"C"}}
22:05:31.803 MQT: stat/sensor/STATUS11 = {"StatusSTS":{"Time":"2021-12-28T22:05:31","Uptime":"0T00:01:04","UptimeSec":64,"Heap":158,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":35},"Wifi":{"AP":1,"SSId":"Get Out","BSSId":"74:DA:88:4F:7E:73","Channel":9,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Shutdown the ESP32
Connect GND, 3.3V, Echo and Trig to ESP
Power on the ESP32

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The ESP32 should boot even with the echo and trig pins connected

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2022-01-10:
Hi, please, can you try latest development version? Thanks.

LoloTheJeeper, 2022-01-11:
Hi ! 
I uploaded the newest dev firmware on the board.
Same result with the US-100 plugged in at boot time. The ESP32 hangs here :
```
[20:28:50]ets Jun  8 2016 00:22:57
[20:28:50]
[20:28:50]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:28:50]configsip: 0, SPIWP:0xee
[20:28:50]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:28:50]mode:DOUT, clock div:2
[20:28:50]load:0x3fff0018,len:4
[20:28:50]load:0x3fff001c,len:1044
[20:28:50]load:0x40078000,len:10124
[20:28:50]load:0x40080400,len:5828
[20:28:50]entry 0x400806a8
[20:28:50]
[20:28:50]00:00:00.003-216/51 HDW: ESP32-D0WDQ6 
[20:28:51]00:00:00.076-215/50 UFS: FlashFS mounted with 304 kB free
[20:28:51]00:00:00.177 CFG: Loaded from File, Count 314
[20:28:51]00:00:00.194 QPC: Count 1
[20:28:51]00:00:00.469 BRY: Berry initialized, RAM used=3926
[20:28:51]00:00:00.504 Project tasmota - Tasmota_Garage Version 2022.01.2(tasmota)-2_0_2(2022-01-10T22:25:13)

```

If I unplug the GND wire from the US-100 at this point, the boot resumes...

```

[20:29:02]00:00:11.321 SR4: Reading CRC error.
[20:29:02]00:00:11.356 SR4: Reading CRC error.
[20:29:02]00:00:11.368 SR4: Mode 1
[20:29:08]00:00:17.557 WIF: Connecting to AP1 Get Out Channel 6 BSSId 74:DA:88:4F:7E:73 in mode 11n as Garage...
[20:29:10]00:00:19.762 WIF: Connected
[20:29:11]00:00:20.025 HTP: Web server active on Garage with IP address 192.168.88.55
[20:29:12]02:29:16.015 MQT: Attempting connection...
[20:29:12]02:29:16.058 MQT: Connected
[20:29:12]02:29:16.067 MQT: tele/garage/LWT = Online (retained)
[20:29:12]02:29:16.085 MQT: cmnd/garage/POWER = 
[20:29:12]02:29:16.114 MQT: tele/garage/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"2022.01.2(tasmota)","FallbackTopic":"cmnd/garage_fb/","GroupTopic":"cmnd/tasmotas/"}}
[20:29:12]02:29:16.132 MQT: tele/garage/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"Garage","IPAddress":"192.168.88.55"}}
[20:29:12]02:29:16.139 MQT: tele/garage/INFO3 = {"Info3":{"RestartReason":"Vbat power on reset"}}
[20:29:12]02:29:16.146 MQT: stat/garage/RESULT = {"POWER1":"OFF"}
[20:29:12]02:29:16.152 MQT: stat/garage/POWER1 = OFF
[20:29:12]02:29:16.158 MQT: stat/garage/RESULT = {"POWER2":"OFF"}
[20:29:12]02:29:16.163 MQT: stat/garage/POWER2 = OFF
[20:29:14]02:29:18.610 QPC: Reset
[20:29:16]02:29:20.626 MQT: tele/garage/STATE = {"Time":"2022-01-11T02:29:20","Uptime":"0T00:00:10","UptimeSec":10,"Heap":162,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":39},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Get Out","BSSId":"74:DA:88:4F:7E:73","Channel":6,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:04"}}
[20:29:16]02:29:20.677 MQT: tele/garage/SENSOR = {"Time":"2022-01-11T02:29:20","Switch1":"OFF","Switch2":"OFF","AM2301":{"Temperature":20.8,"Humidity":51.6,"DewPoint":10.4},"SR04":{"Distance":0.088},"ESP32":{"Temperature":33.9},"TempUnit":"C"}
[20:29:27]02:29:31.617 MQT: tele/garage/STATE = {"Time":"2022-01-11T02:29:31","Uptime":"0T00:00:21","UptimeSec":21,"Heap":162,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":39},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Get Out","BSSId":"74:DA:88:4F:7E:73","Channel":6,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:04"}}
[20:29:27]02:29:31.668 MQT: tele/garage/SENSOR = {"Time":"2022-01-11T02:29:31","Switch1":"OFF","Switch2":"OFF","AM2301":{"Temperature":20.8,"Humidity":51.5,"DewPoint":10.4},"SR04":{"Distance":0.105},"ESP32":{"Temperature":33.9},"TempUnit":"C"}
```

If I don't unplug anything from the US-100, the ESP32 hangs indefinitly.
If I unplug for example the GND on the US-100, let the boot process finish then replug the GND, the ESP32 and the sensor works just perfect.

If you want me to test something else, don't hesitate to ask ;)

Thank you (and BTW Tasmota is a REAL piece of work)

LoloTheJeeper, 2022-01-18:
I am sorry because English is not my mother tongue and I think "piece of work" doesn't mean what I thought. What I meant is that Tasmota is an impressive work with a super active community and that a lot of work has been put in development. Sorry if I hurt anybody. It was not what I intended. I would really be ungrateful. I have 10 dimmers and 8 power plugs all driven by Tasmota. Tasmota rules !!!

Jason2866, 2022-01-18:
Please try again, maybe the issue fixed with #14499 is the reason for your problem.

LoloTheJeeper, 2022-01-19:
Hi Jason2866, I just tried the build 2022.01.3.
Exact same result. The ESP freezes at the same place then resumes if I unplug the GND pin from the US-100.
Everything works if I plug back the sensor AFTER the boot

arendst, 2022-01-19:
Pls try two other GPIOs and report back.

LoloTheJeeper, 2022-01-19:
Hi !
I tried these GPIO with the exact same results :

TRI       ECH
25        26
19        18
33        32

Thank you !

LoloTheJeeper, 2022-01-20:
Thank you very much Theo.
On my side I also tested the HC-SR04 and it works as expected, with the same configuration and same GPIOs. Working with a US-100 is easier with the esp32 because it removes the need of the voltage divider resistors. Thanks again and let me know if I can be of any help ;)

arendst, 2022-01-20:
BTW I test my SR04 directly connected to GPIO18/19 and to the 5V from the ESP32 devkit. No dividers connected. Probably not a long term solution but currently it still works after an hour ;-)

arendst, 2022-01-20:
I ordered two US-100 for testing.

LoloTheJeeper, 2022-01-20:
Yep at first that's also what I did but I was concerned of health of the GPIO handling the echo signal at 5 v on the long term for the esp32... 
Have a great day. And thanks again :)

techman83, 2022-01-21:
@LoloTheJeeper - the pins are pretty tolerant in my experience. And they're only a few dollars to replace, so I've never really worried too much about it. Be different if selling a product or burying the device somewhere super hard to get to.

LoloTheJeeper, 2022-01-21:
Thanks @techman83 .
Id rather not take a chance because this esp32 will drive the opening and closing of my garage door and will be positioned in a hard to reach spot. I'll add the two resistors to lower the 5v to 3.3 on that particular GPIO input !

arendst, 2022-01-21:
Just received my US-100. Connected it to the ESP-Devkit at the same location a working SR04 was connected. Power cycled the device and presto:

```
00:00:00.003-218/51 HDW: ESP32-D0WDQ6 
00:00:00.064-216/51 UFS: FlashFS mounted with 304 kB free
00:00:00.166 CFG: Loaded from File, Count 424
00:00:00.182 QPC: Count 1
00:00:00.192 ROT: Mode 1
00:00:00.426 CFG: No '*.autoconf' file found
00:00:00.433 BRY: Berry initialized, RAM used=3930 bytes
00:00:00.457 BRY: No 'preinit.be'
00:00:00.462 SRC: Restart
00:00:00.467 Project tasmota - Esp32d Version 2022.01.3(theo)-2_0_2(2022-01-20T14:37:17)
00:00:00.469 ETH: No ETH MDC and/or ETH MDIO GPIO defined
00:00:00.487 DSB: Sensors found 0
00:00:00.498 SHT: Sensor did not ACK command
00:00:00.544 I2C: HTU21 found at 0x40
00:00:00.564 BME: Gas variant 0
00:00:00.571 I2C: BME680 found at 0x77
00:00:01.306 SR4: Mode 1
00:00:01.494 BRY: No 'autoexec.be'
00:00:01.911 WIF: Attempting connection...
00:00:02.446 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11n as esp32d...
00:00:03.081 WIF: Attempting connection...
00:00:04.708 WIF: Connected
00:00:04.830 RTC: UTC 2022-01-21T10:29:56, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
11:29:57.051 HTP: Web server active on esp32d with IP address 192.168.2.115
11:29:57.947 MQT: Attempting connection...
11:29:57.981 MQT: Connected
11:29:57.992 MQT: tele/esp32d/LWT = Online (retained)
11:29:58.005 MQT: cmnd/esp32d/POWER = 
11:29:58.007 MQT: Subscribe to cmnd/esp32d/#
11:29:58.011 MQT: Subscribe to cmnd/tasmotas/#
11:29:58.017 MQT: Subscribe to cmnd/DVES_6D2DB4_fb/#
11:29:58.035 MQT: tele/esp32d/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"2022.01.3(theo)","FallbackTopic":"cmnd/DVES_6D2DB4_fb/","GroupTopic":"cmnd/tasmotas/"}}
11:29:58.054 MQT: tele/esp32d/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"esp32d","IPAddress":"192.168.2.115"}}
11:29:58.060 MQT: tele/esp32d/INFO3 = {"Info3":{"RestartReason":"Vbat power on reset"}}
11:29:58.084 TFS: File 'autoexec.bat' not found
11:29:59.824 QPC: Reset
11:30:01.777 APP: Boot Count 359
11:30:03.001 CFG: Saved, Count 425, Bytes 4096
11:30:03.019 MQT: tele/esp32d/STATE = {"Time":"2022-01-21T11:30:03","Uptime":"0T00:00:12","UptimeSec":12,"Heap":163,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":39},"Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-24,"LinkCount":1,"Downtime":"0T00:00:04"}}
11:30:03.065 MQT: tele/esp32d/SENSOR = {"Time":"2022-01-21T11:30:03","HTU21":{"Temperature":22.6,"Humidity":38.0,"DewPoint":7.5},"BME680":{"Temperature":24.1,"Humidity":40.4,"DewPoint":9.8,"Pressure":1031.9,"Gas":2436.82},"SR04":{"Distance":66.877},"ESP32":{"Temperature":50.0},"PressureUnit":"hPa","TempUnit":"C"}
```
No hangs. Just works.

I suspect your hardware is wrong.

LoloTheJeeper, 2022-01-21:
Hum, this is really wired. I tested the same hardware with an ino script and with an esphome integration without any problem. I guess  I'm going to order a new US-100 and see. I'll keep you posted.
Again, thank you everyone for your time and dedication ! Tasmota really rocks !!!

LoloTheJeeper, 2022-01-23:
Hi @arendst , I tested the US-100 with another Devkit I had and I've got the exact same behavior.
I don't have another US-100 but as I said, it seems to work well with a ino script or with ESPHome...
Is there a way we could get more logs to try to know what makes the ESP hang ?
Thank you again !

arendst, 2022-01-24:
Try latest dev which makes the US-100 serial work. So insert the jumper and see what happens.

LoloTheJeeper, 2022-01-24:
Thank you @arendst, I'll test the new dev FW in "serial" mode on the US-100 tonight !

LoloTheJeeper, 2022-01-25:
I think you nailed it @arendst  ! 
It works perfect. SR4: Mode 3.
I don't know why the other mode wasn't working but I'll use it in serial com mode :)
Thank you soooo much for your time 
Again, congratulations for Tasmota ! It's a HUGE project that allows people without extended programming skills to interface hardware and home automation !

Jason2866, 2022-01-26:
Closing since it is fixed

closed 2022-01-26 by Jason2866