arendst/Tasmota #14198: platform compile error esp32c3

kskenyon, 2021-12-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

/Users/private/Tasmota/tasmota/support.ino:2031:22: error: 'class TwoWire' has no member named 'end'; did you mean 'read'?
   if (reinit) { Wire.end(); }
                      ^~~
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-12-28:
We have no problem building tasmota32c3
So you must do something wrong
Please details what you've done, what you changed and how exactly are you building?

Such problems can be easier to help on Discord #compiling channel

kskenyon, 2021-12-28:
I’ll try on Discord.  Thanks for the feedback.

This appears to be a library/coding error of some sort.  My build environment is working otherwise.

Kevin

barbudor, 2021-12-28:
There is no error in current code base. It builds perfectly well.
But without more details on your setup, it's not possible to what is wrong on your side.
- Insure you clean up your `.pio/` and `.cache/` folders
- Be sure to build `tasmota32c3` environement
   - either run `platformio run -e tasmota32c3`
   - or in VSCode select `tasmota32c3` in the selection menu and click the build "tick" 
  
[image]

kskenyon, 2021-12-28:
Compiling .pio/build/tasmota32c3/libff9/bearssl-esp8266/hash/sha2big.c.o
/Users/private/Tasmota/tasmota/support.ino: In function 'bool I2cBegin(int, int, uint32_t)':
/Users/private/Tasmota/tasmota/support.ino:2031:22: error: 'class TwoWire' has no member named 'end'; did you mean 'read'?
   if (reinit) { Wire.end(); }

                      ^~~
                      read

Linter
Severity	Provider	Description	Line
GitHub
Git

Checked out pull request
You are currently on your repository's default branch.
Checkout or create a new branch  to share your work with a pull request.
Open pull requests
Offline

You can see that I am building tasmota32c3 and I did a clean before building.  I just did a git pull before this build attempt.  This is my first tasmota32c3 device.

How could this error be related to my build environment? I don’t get it.

barbudor, 2021-12-28:
Did you pulled `master` or `development` ?
I'm not sure 32C3 was part of latest master release
Be sure to use development

Can you clarify how you are building ?
Are you using VSCode ? or command line platformio ?
I don't understand the lines after the error (the line with `read`)

kskenyon, 2021-12-28:
I’m using Atom with PlatformIO on a new Macbook Pro M1.  All updated.

It’s the Development branch of Tasmota.

It shows the location of the error and a possible correction.  That’s what the last line indicates.  I didn’t type that.

Thanks for hangin in there on this.

K

barbudor, 2021-12-28:
I have no experience with Atom
We are all using VScode on either Mac, Linux on Windows.

@Jason2866 is using a Macbook M1 but with VScode

kskenyon, 2021-12-28:
Deleted .platformio/platforms and .platformio/packages.  Auto reinstall did the trick.  Must have been an old library in there somehow.

Thanks.

K

closed 2021-12-29 by Jason2866