arendst/Tasmota #14203: AM2301-Sensor not working - always "null" values.

didi-j, 2021-12-29:
### PROBLEM DESCRIPTION
_AM2301-Sensor not working - always "null" values._

I tried two am2301-sensors on different GPIO-ports with and without 4k7 pullup resistor. The shown values are allways null.
On console I got "Timeout waiting for start signal high pulse". Checking the signal on GPIO with a DSO seems ok.
I tried also various configurations (am2301, DHT11, Si7021) without success.

I took a look in the code (xsns_06_dht.ino) and found signal-timings:
1.) 2000 µs low
[image]
[image]
[image]

2.) 50 µs hight
3.) wait max 100 µs for 0
4.) wait max 100 µs for 1  (maybe the timeout is here)
5.) wait max 100 µs for 0

on my DSO-signal, the time between  point 3 and 4 is about 400µs (see screenshots)
 
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU D1 Mini
- [x] Tasmota binary firmware version number used: 10.1.0.3(allsensors) 2021-12-28T15:29:49
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:12:10.744 CMD: Backlog Template; Module; GPIO 255
13:12:10.746 SRC: WebConsole from 10.0.0.139
13:12:10.748 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
13:12:10.763 SRC: Backlog
13:12:10.765 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
13:12:10.773 MQT: stat/tasmota_CD5762/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
13:12:10.990 SRC: Backlog
13:12:10.992 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
13:12:11.001 MQT: stat/tasmota_CD5762/RESULT = {"Module":{"18":"Generic"}}
13:12:11.193 DHT: Timeout waiting for start signal high pulse
13:12:11.195 DHT: Read 0000000000
13:12:11.365 SRC: Backlog
13:12:11.368 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
13:12:11.379 MQT: stat/tasmota_CD5762/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"1216":"AM2301"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"1856":"SR04 Tri/TX"},"GPIO14":{"0":"None"},"GPIO15":{"1888":"SR04 Ech/RX"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
13:16:02.673 CMD: Backlog Rule1; Rule2; Rule3
13:16:02.675 SRC: WebConsole from 10.0.0.139
13:16:02.677 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 19, Data 'Rule1; Rule2; Rule3'
13:16:02.797 SRC: Backlog
13:16:02.800 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
13:16:02.808 MQT: stat/tasmota_CD5762/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":70,"Free":441,"Rules":"on ANALOG#A0 do DisplayText [zs1c1l1]Analog A0:[s2l2c5]%value%  endon "}}
13:16:03.024 SRC: Backlog
13:16:03.026 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
13:16:03.036 MQT: stat/tasmota_CD5762/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:16:03.399 SRC: Backlog
13:16:03.402 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
13:16:03.409 MQT: stat/tasmota_CD5762/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:17:21.620 CMD: STATUS 0
13:17:21.622 SRC: WebConsole from 10.0.0.139
13:17:21.624 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
13:17:21.630 MQT: stat/tasmota_CD5762/STATUS = {"Status":{"Module":18,"DeviceName":"TasmotaTest2","FriendlyName":["TasmotaTest2"],"Topic":"tasmota_CD5762","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:17:21.638 MQT: stat/tasmota_CD5762/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/tasmota/install/raw/main/firmware/unofficial/tasmota-allsensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:04:37","StartupUTC":"2021-12-29T11:12:44","Sleep":50,"CfgHolder":4617,"BootCount":148,"BCResetTime":"2021-11-27T15:20:33","SaveCount":365,"SaveAddress":"F4000"}}
13:17:21.643 MQT: stat/tasmota_CD5762/STATUS2 = {"StatusFWR":{"Version":"10.1.0.3(allsensors)","BuildDateTime":"2021-12-28T15:29:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"406/699"}}
13:17:21.648 MQT: stat/tasmota_CD5762/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxl",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","2805C80001000600003C5A0A000000000000","00800080","00006000","00004000"]}}
13:17:21.667 MQT: stat/tasmota_CD5762/STATUS4 = {"StatusMEM":{"ProgramSize":711,"Free":292,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F908787","04146001","FFFFBFCF","8F7417FF","303E0AD3","1149785A","26204FFF","0008F0F3"],"Drivers":"1,2,3,4,7,8,9,10,12,14,15,20,21,24,28,29,34,43,47,50,56,60","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,24,26,27,28,29,30,31,32,33,38,40,41,42,44,45,46,50,51,54,55,56,57,58,59,63,64,65,66,67,70,71,72,73,74,76,78,79,81,82,85,86,88,89,90,91,92,93"}}
13:17:21.676 MQT: stat/tasmota_CD5762/STATUS5 = {"StatusNET":{"Hostname":"tasmota-CD5762-5986","IPAddress":"10.0.0.162","Gateway":"10.0.0.138","Subnetmask":"255.255.255.128","DNSServer1":"10.0.0.138","DNSServer2":"0.0.0.0","Mac":"A8:48:FA:CD:57:62","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:17:21.682 MQT: stat/tasmota_CD5762/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.140","MqttPort":1883,"MqttClientMask":"test2","MqttClient":"test2","MqttUser":"fhemMqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:17:21.689 MQT: stat/tasmota_CD5762/STATUS7 = {"StatusTIM":{"UTC":"2021-12-29T12:17:21","Local":"2021-12-29T13:17:21","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:00"}}
13:17:21.698 MQT: stat/tasmota_CD5762/STATUS10 = {"StatusSNS":{"Time":"2021-12-29T13:17:21","AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"SR04":{"Distance":141.737},"TempUnit":"C"}}
13:17:21.704 MQT: stat/tasmota_CD5762/STATUS11 = {"StatusSTS":{"Time":"2021-12-29T13:17:21","Uptime":"0T01:04:37","UptimeSec":3877,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxxxxxxx
![Screenshot from 2021-12-29 13-25-12](https://user-images.githubusercontent.com/96823649/147661921-64ad3787-5100-4229-9370-2d1d52c3417c.png)
![Screenshot from 2021-12-29 13-30-24](https://user-images.githubusercontent.com/96823649/147662401-d921bf67-c672-4c8f-ab0e-98f011330243.png)
![Screenshot from 2021-12-29 13-30-50](https://user-images.githubusercontent.com/96823649/147662418-50e7b352-3dd5-49bb-8aee-66cc764e4e4d.png)
","BSSId":"44:4E:6D:DA:7F:1D","Channel":6,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:17:59.226 DHT: Timeout waiting for start signal high pulse
13:17:59.227 DHT: Read 0000000000
13:18:01.233 DHT: Timeout waiting for start signal high pulse
13:18:01.235 DHT: Read 0000000000
13:18:03.239 DHT: Timeout waiting for start signal high pulse
13:18:03.240 DHT: Read 0000000000
13:18:05.234 DHT: Timeout waiting for start signal high pulse
13:18:05.235 DHT: Read 0000000000
13:18:07.229 DHT: Timeout waiting for start signal high pulse
13:18:07.231 DHT: Read 0000000000

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
please adjust timings (higher timeout in DhtWaitState() ??)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-12-29:
The history of the DHT driver is a long one.

I suggest you have a look at file ``xsns_06_dht.ino`` where at the beginning you find some links of changes made on request of users not being able to get a stable readout.

My AOSONG AM2301 works fine on both ESP8266 and ESP32.

[image]

I just made a change to the polling interval to stabilize the DHT/AM before a next readout.

Changing the file above feel free to play with the timings as you wish; you probably come up with values unworkable for others but usable for you.

sfromis, 2021-12-29:
DHT22 (and similar one-wire sensors) is a bad mess. I've got a DHT22 working reasonably well on ESP8266, but hardly on ESP32. Unless you just love them, I'd suggest sticking to I2C sensors where feasible, that communication protocol is much bore robust.

didi-j, 2021-12-30:
I got my sensors working!

I only changed the timeout from 100 to 400 µs in xsns_06_dht.ino at line 57

`unsigned long timeout = micros() + 400;     // before 100`

I think there should be no impact to now-working-sensors. 
@arendst: please check in the higher timeout for future releases

thanks

didi-j, 2021-12-30:
bad news

sensor stopped working after about 4 hours - I'm ddebugging ...

Jason2866, 2021-12-30:
@didi-j 
Famous last words 
>I think there should be no impact to now-working-sensors.

Every change (in the long history) of the DHT driver have had impacts.

ascillato2, 2022-01-03:
> I got my sensors working!

great to know that now it is working for you. Thanks.

closed 2022-01-03 by ascillato2

orlandperz, 2022-07-04:
Thanks [didi-j](https://github.com/didi-j) you solution work for me

sonybansi, 2024-10-07:
I am facing the same problem. Where can I change "unsigned long timeout = micros" values from 100 to 400? Can you provide the file location, please?

fabian727, 2024-10-07:
It was just a generic explenation. I dont kno ehich file/line within tasmota does the trick. I changed to a Hardware rx/tx pin

dobsonaron, 2025-07-29:
> I got my sensors working!
>
> I only changed the timeout from 100 to 400 µs in xsns_06_dht.ino at line 57
>
> `unsigned long timeout = micros() + 400; // before 100`
>
> I think there should be no impact to [...]

you are a lifesaver! This change made my am2301s work on a nodemcu v3.
Thank you very much for this!