arendst/Tasmota #14213: Wrong address definition in file support_command.ino

mdiazfer, 2021-12-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

File support_command.ino, line 68, I suspect the of array TasmotaCommand[] is wrong for the conditional USE_I2C part, as it include a variable, not the variable address.

It states: &CmndI2cScan, CmndI2cDriver,

It should state: &CmndI2cScan, &CmndI2cDriver,

Please, check.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): all
- [x] Tasmota binary firmware version number used: 1.10
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: platformio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here: Not apply

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: Not apply

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here: Not apply

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here: Not apply

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Check the file support_command.ino, line 68

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-12-30:
Good eye
However in that context, it doesn't make any difference. While the function name itself or & of function are of different types, then end up providing the same address.
Same for an array.
There are a few articles available on the Internet about that.

Still would be nicer to use the same syntax for code coherency.

Thanks

barbudor, 2021-12-30:
https://github.com/arendst/Tasmota/pull/14215
Found a 2nd one in Zigbee too

mdiazfer, 2021-12-30:
Good to know.
Thanks for the clarification.

closed 2021-12-30 by mdiazfer