arendst/Tasmota #14222: Zigbee2Tasmota: impossible to use a rule on Zigbee multipress buttons

pilaGit, 2021-12-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Zigbee button does not produce a single unique trigger in its ZbResponse so usage of a ZB multipress switch with more than one button in a rule is not possible

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff ZbBridge + Zigbee Aqara Opple 6 key remote switch
- [x] Tasmota binary firmware version number used: ncp-uart-nsw_6.7.9_115200.ota
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:33:34.280 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:33:34.486 RSL: RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
15:33:34.740 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
that is the root cause of the problem, described later

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

15:35:14.297 RSL: STATUS = {"Status":{"Module":75,"DeviceName":"Zigbee","FriendlyName":["Zigbee"],"Topic":"tasmota_9DC6F0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:35:14.301 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:28:31","StartupUTC":"2021-12-31T14:06:43","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2021-12-31T10:19:51","SaveCount":46,"SaveAddress":"1FB000"}}
15:35:14.306 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(zbbridge)","BuildDateTime":"2021-12-08T14:47:37","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"365/699"}}
15:35:14.309 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["db1","d"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A00000000000000","00000080","00046002","00004000"]}}
15:35:14.315 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":671,"Free":1120,"Heap":28,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","00000000"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50","Sensors":"1"}}
15:35:14.319 RSL: STATUS5 = {"StatusNET":{"Hostname":"Zigbee","IPAddress":"192.168.2.60","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:9D:C6:F0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:35:14.323 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-12-31T14:35:14","Local":"2021-12-31T15:35:14","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:36","Sunset":"16:19"}}
15:35:14.326 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-12-31T15:35:14"}}
15:35:14.331 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-12-31T15:35:14","Uptime":"0T00:28:31","UptimeSec":1711,"Vcc":3.499,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"db1","BSSId":"18:31:BF:E4:9B:E8","Channel":11,"Mode":"11n","RSSI":100,"Signal":-31,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Button does not produce a single unique trigger so usage of a ZB multipress switch in a rule is not possible

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
When a ZB button is pressed, resulting ZbResponse should include the single unique identifying action to be usable as a trigger in a rule for testing.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The rules currently cannot be applied to Zigbee multipress buttons and such behavior is not consistent with other Tasmota features. Any other form of such testing of a response is also quite cumbersome even when possible.

I am testing Zigbee2Tasmota using Sonoff ZbBridge. I have paired Zigbee Aqara Opple 6 key remote switch. Relevant ZB config: `Backlog ZbName 0x23D3,Opple; SetOption83 1; ZbNoPrefix 1`

While the switch is in `{"OppleMode":0}` it sends 6 single keys and they can be normally used in rules.

`on Opple#0006!01 Do Var1 2single Endon`
`on Opple#DimmerStepDown Do Var1 3single Endon`

In order to get multiple clicks and hold, the Aqara Opple 6 key remote switch must be switched to `{"OppleMode":1}`. When multipress `{"OppleMode":1}` is enabled, it becomes impossible to apply a rule to its ZbResponse.

There are two problems: a) not all clicks have action and b) action itself is neither unique nor tied to the corresponding button. The problem and proposed solution likely apply to all other switches with more than one button and multipress and action.

Here are all available actions for the Button 1:

```
SENSOR = {"Opple":{"Device":"0x23D3","Name":"Opple","MultiInValue":1,"Click":"single","click":"single","Endpoint":1,"LinkQuality":139}}
SENSOR = {"Opple":{"Device":"0x23D3","Name":"Opple","MultiInValue":2,"Click":"double","click":"double","Endpoint":1,"LinkQuality":136}}
SENSOR = {"Opple":{"Device":"0x23D3","Name":"Opple","MultiInValue":3,"Click":"triple","Endpoint":1,"LinkQuality":136}}
SENSOR = {"Opple":{"Device":"0x23D3","Name":"Opple","MultiInValue":0,"Click":"hold","action":"hold","Endpoint":1,"LinkQuality":139}}
SENSOR = {"Opple":{"Device":"0x23D3","Name":"Opple","MultiInValue":255,"Click":"release","action":"release","Endpoint":1,"LinkQuality":136}}
```

**Problems:**
1.	Single and double click have repeated "click" but no "action"
2.	Triple click has neither
3.	Hold and Release have the "action" but not unique to the button. The button is identified by "Endpoint":1 for the first one. In this case the "Endpoint" range is from 1 to 6 inclusive.

This means the switch is not correctly (in a consistent manner) translated by Z2T as evidenced by problems 1. and 2.

To apply a rule, we need to boil an event down to one single triggervalue we can test (either for a change or the contents). So, each button should have the action and each action should include the button number. Like we now have: Power1, Power2…

So a pressed button 1 ("Endpoint":1) should produce **"action":"single1"**
a double pressed button 1 ("Endpoint":1) should produce **"action":"double1"**
a pressed button 2 ("Endpoint":2) should produce **"action":"single2"**
a released button 3 ("Endpoint":3) should produce **"action":"release3"**

Then, ZbResponse for such multipress multipoint switch would be easily tested by a rule as:

`on Opple#action=double1 Do Var1 %value% Endon`

and/or:

`on Opple#action Do Var1 %value% Endon`

Since rules are size limited, it would be better if action is started with the number. We could then test if the action string begins with "1s" or "2d" or "3r" instead of being equal to "single1" or "double2" or "release3". I understand this format is inconsistent with Tasmota, but since we are talking about one Bridge to many ZB devices, I feel this deviation is appropriate here.

Then it would be shorter to test:

`on Opple#action$<1d Do Var1 %value% Endon`

**(Please, remember to close the issue when the problem has been addressed)**

pilaGit, 2022-01-04:
I just went on to find a workaround for this problem, and here it is:
```

rule3
on Opple2#Endpoint Do Var9 %value% Endon
on Opple2#Click Do Var2 %Var9%%value% Endon
on Var2#State$<2s Do WebSend [192.168.2.72] Power 2 Break
on Var2#State$<2d Do WebSend [192.168.2.72] Event Dim Break
```

This constructs 1single and 2 single in Var2. Then I test. Same can be done with 
```
rule3
on Opple2#Endpoint Do Var9 %value% Endon
on Opple2#MultiInValue Do Var2 %Var9%%value% Endon
on Var2#State=21 Do WebSend [192.168.2.72] Power 2 Break
on Var2#State=22 Do WebSend [192.168.2.72] Event Dim Break
```
The price is loosing the precious rules space for the first two lines to create usable test value. 

If this would have been working in accordance to my initial post, instead of the first two lines, we would do:

```
rule3
on Opple2#Action Do Var2 %value% Endon

```

ascillato2, 2022-01-10:
Moving to ideas as this is a feature request. Thanks.