arendst/Tasmota #14236: PN532 stops working after few hours on WEMOS D1 Mini

DarthWeber, 2022-01-02:
### PROBLEM DESCRIPTION
Using the tasmota sensor firmware and connection PN532 the sensor works and tasmota displays the uid, but after a few minutes or hours no cards will be detected. There is even no output in the console. Tasmote still works and can be restartet, which let the PN532 work again for a couple of time.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Generic(18) on wemos d1 mini
- [x] Tasmota binary firmware version number used: 10.1.0(sensors)
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:19:51.110 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_DE5BD9","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:19:51.136 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:20","StartupUTC":"2022-01-02T07:13:31","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2022-01-02T07:49:56","SaveCount":17,"SaveAddress":"FB000"}}
08:19:51.169 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(sensors)","BuildDateTime":"2021-12-08T14:47:41","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"362/699"}}
08:19:51.189 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TomSushi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
08:19:51.223 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":668,"Free":332,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
08:19:51.258 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-DE5BD9-7129","IPAddress":"192.168.178.27","Gateway":"192.168.178.2","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.2","DNSServer2":"0.0.0.0","Mac":"50:02:91:DE:5B:D9","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:19:51.283 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DE5BD9","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:19:51.306 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-01-02T07:19:51","Local":"2022-01-02T08:19:51","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:04"}}
08:19:51.325 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-02T08:19:51"}}
08:19:51.331 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-01-02T08:19:51","Uptime":"0T00:06:20","UptimeSec":380,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":41,"MqttCount":0,"Wifi":{"AP":1,"SSId":"TomSushi","BSSId":"E8:DF:70:25:DF:8C","Channel":1,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
fresh installation of tasmota sensor bin. Connected PN532 to GND,3.3V,GPIO4 and 5. Nothing else configured except wifi of course.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Noschvie, 2022-01-02:
Duplicate of #12334

DarthWeber, 2022-01-02:
> Duplicate of #12334

Obviosly NOT !!! There is **nothing** common except the PN532.

Jason2866, 2022-01-07:
My test PN532 connected to the same Gpios as yours on a Wemos D1 mini is still working after 4 days. I can not reproduce the issue. Maybe your setup is bad (wiring, power supply...) or has a hardware defect? Can you try spare ones?

ascillato2, 2022-01-10:
Hi, any news on this? We're you able to test if the hardware is broken?

closed 2022-01-10 by ascillato2

DarthWeber, 2022-01-11:
After soldering the connection instead of using breakout cables the connections seems to be stable. Thanks for all feedbacks.