arendst/Tasmota #14245: Analog Input PH Sensor toggle to stupid values

ChrisVonFrosta, 2022-01-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): NodeMCU 
- [ ] Tasmota binary firmware version number used: 10.1.0.1(sensors)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  17:12:38.651 CMD: Backlog Template; Module; GPIO 255
17:12:38.712 MQT: stat/tasmota_C5C718/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:12:38.912 MQT: stat/tasmota_C5C718/RESULT = {"Module":{"18":"Generic"}}
17:12:39.120 MQT: stat/tasmota_C5C718/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"608":"I2C SCL"},"GPIO2":{"0":"None"},"GPIO3":{"608":"I2C SCL"},"GPIO4":{"0":"None"},"GPIO5":{"417":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"1216":"AM2301"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"416":"PWM1"},"GPIO17":{"6336":"ADC pH"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
17:14:19.507 CMD: Backlog Rule1; Rule2; Rule3
17:14:19.583 MQT: stat/tasmota_C5C718/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
17:14:19.784 MQT: stat/tasmota_C5C718/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
17:14:19.986 MQT: stat/tasmota_C5C718/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
17:14:49.389 CMD: Status 0
17:14:49.396 MQT: stat/tasmota_C5C718/STATUS = {"Status":{"Module":18,"DeviceName":"Garten-Sensor-PWM-Board","FriendlyName":["Garten-Sensor-PWM-Board","Tasmota2"],"Topic":"tasmota_C5C718","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:14:49.430 MQT: stat/tasmota_C5C718/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/","RestartReason":"Software/System restart","Uptime":"0T00:04:39","StartupUTC":"2022-01-02T16:10:10","Sleep":0,"CfgHolder":4617,"BootCount":271,"BCResetTime":"2020-04-20T21:34:58","SaveCount":85493,"SaveAddress":"F7000"}}
17:14:49.464 MQT: stat/tasmota_C5C718/STATUS2 = {"StatusFWR":{"Version":"10.1.0.1(sensors)","BuildDateTime":"2021-12-22T17:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"514/699"}}
17:14:49.483 MQT: stat/tasmota_C5C718/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Phradio","WashTeam"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00040088","00006000","00004000"]}}
17:14:49.520 MQT: stat/tasmota_C5C718/STATUS4 = {"StatusMEM":{"ProgramSize":668,"Free":332,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
17:14:49.560 MQT: stat/tasmota_C5C718/STATUS5 = {"StatusNET":{"Hostname":"Garten-Sensor-PWM-Board","IPAddress":"192.168.169.84","Gateway":"192.168.169.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.169.1","DNSServer2":"0.0.0.0","Mac":"EC:FA:BC:C5:C7:18","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:14:49.586 MQT: stat/tasmota_C5C718/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.169.86","MqttPort":1883,"MqttClientMask":"Garten-Sensor-PWM-Board_%06X","MqttClient":"Garten-Sensor-PWM-Board_C5C718","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:14:49.612 MQT: stat/tasmota_C5C718/STATUS7 = {"StatusTIM":{"UTC":"2022-01-02T16:14:49","Local":"2022-01-02T17:14:49","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:04"}}
17:14:49.640 MQT: stat/tasmota_C5C718/STATUS10 = {"StatusSNS":{"Time":"2022-01-02T17:14:49","ANALOG":{"pH0":145411296.00},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}}
17:14:49.662 MQT: stat/tasmota_C5C718/STATUS11 = {"StatusSTS":{"Time":"2022-01-02T17:14:49","Uptime":"0T00:04:39","UptimeSec":279,"Heap":17,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":933,"MqttCount":1,"POWER1":"ON","Channel1":100,"POWER2":"OFF","Channel2":0,"Color":"FF00","Wifi":{"AP":1,"SSId":"Phradio","BSSId":"5C:49:79:23:C2:3B","Channel":6,"Mode":"11n","RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The value of the PH is flipping from "normal" values to crazy values, eg: -145411312.0. Sometimes negativ. But also "normal" values are jumping from 10 to 5 and back to 10 without moving in the fresh water.
The Probe was calibrated 1 week ago.

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-02:
Tasmota is just applying formulas on the voltage measured by the ADC
If values are wrong, you have connection problems with your connection to the probe or probe might be defective.
You can set the ADC back to 'Analog Input' and check if the values are coherent with the expect voltage.

If it has been working before and not working as expected anymore, the problem is hardware

ChrisVonFrosta, 2022-01-02:
you are sure, i have to check the hardware output. But my oszilloscope and multimeter are to far awy for the moment.

i did set the ADC back to 'Analog Input' and i never saw "crazy" values. The analog value does not jumps on the website and never exceed the adc range. So i can't explain the (-)145411296 value.

When i start to answer this, the website shows an value of 145411296 again. It jumps to 12.44 after a second, a second later to 4.88. It seems to be randomly.

When set it to 'adc input', i got low flowting value between 250 and 260.
So i don't see why tasmota caluclate those strange values.

barbudor, 2022-01-02:
Can you type the AdcParam command to confirm the calibration values ?
May be someting went wrong with the setttings

barbudor, 2022-01-02:
Also enable `Weblog 3` so there should be logs that show both the raw ADC value and the calculated pH
That may help us understand why values are going so much out of range

ChrisVonFrosta, 2022-01-02:
1:08:16.415 MQT: tele/tasmota_C5C718/SENSOR = {"Time":"2022-01-02T21:08:16","ANALOG":{"pH0":10.81},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}
21:08:16.422 DHT: Timeout waiting for start signal low pulse
21:08:17.090 APP: Analog pH read. ADC-RAW: 323, cal-low(pH=ADC): 4.00=282, cal-high(pH=ADC): 9.18=435
21:08:18.382 DHT: Timeout waiting for start signal low pulse
21:08:19.528 APP: Analog pH read. ADC-RAW: 525, cal-low(pH=ADC): 4.00=282, cal-high(pH=ADC): 9.18=435
21:08:20.382 DHT: Timeout waiting for start signal low pulse
21:08:21.965 APP: Analog pH read. ADC-RAW: 312, cal-low(pH=ADC): 4.00=282, cal-high(pH=ADC): 9.18=435
21:08:22.382 DHT: Timeout waiting for start signal low pulse
21:08:24.382 DHT: Timeout waiting for start signal low pulse
21:08:24.410 APP: Analog pH read. ADC-RAW: 485, cal-low(pH=ADC): 4.00=282, cal-high(pH=ADC): 9.18=435
21:08:26.386 MQT: tele/tasmota_C5C718/STATE = {"Time":"2022-01-02T21:08:26","Uptime":"0T01:16:11","UptimeSec":4571,"Heap":22,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"ON","Channel1":100,"POWER2":"OFF","Channel2":0,"Color":"FF00","Wifi":{"AP":1,"SSId":"Phradio","BSSId":"5C:49:79:23:C2:3B","Channel":6,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:08:26.397 APP: Analog pH read. ADC-RAW: 353, cal-low(pH=ADC): 4.00=282, cal-high(pH=ADC): 9.18=435
21:08:26.405 MQT: tele/tasmota_C5C718/SENSOR = {"Time":"2022-01-02T21:08:26","ANALOG":{"pH0":6.40},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}
21:08:26.412 DHT: Timeout waiting for start signal low pulse
21:08:26.835 APP: Analog pH read. ADC-RAW: 500, cal-low(pH=ADC): 4.00=282, cal-high(pH=ADC): 9.18=435
21:08:28.382 DHT: Timeout waiting for start signal low pulse

ChrisVonFrosta, 2022-01-02:
on the startpage i can still notice value "flipping" but not in the log?!

ChrisVonFrosta, 2022-01-02:
after set analog input from ph to raw 'analog input', it seems to be more consistent

ChrisVonFrosta, 2022-01-02:
23:18:27.338 MQT: tele/tasmota_C5C718/STATE = {"Time":"2022-01-02T23:18:27","Uptime":"0T00:02:00","UptimeSec":120,"Heap":22,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"ON","Channel1":100,"POWER2":"OFF","Channel2":0,"Color":"FF00","Wifi":{"AP":1,"SSId":"Phradio","BSSId":"5C:49:79:23:C2:3B","Channel":6,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:18:27.412 MQT: tele/tasmota_C5C718/SENSOR = {"Time":"2022-01-02T23:18:27","ANALOG":{"A0":280},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}
23:18:37.337 MQT: tele/tasmota_C5C718/STATE = {"Time":"2022-01-02T23:18:37","Uptime":"0T00:02:10","UptimeSec":130,"Heap":22,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"ON","Channel1":100,"POWER2":"OFF","Channel2":0,"Color":"FF00","Wifi":{"AP":1,"SSId":"Phradio","BSSId":"5C:49:79:23:C2:3B","Channel":6,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:18:37.412 MQT: tele/tasmota_C5C718/SENSOR = {"Time":"2022-01-02T23:18:37","ANALOG":{"A0":263},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}
23:18:47.338 MQT: tele/tasmota_C5C718/STATE = {"Time":"2022-01-02T23:18:47","Uptime":"0T00:02:20","UptimeSec":140,"Heap":22,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"ON","Channel1":100,"POWER2":"OFF","Channel2":0,"Color":"FF00","Wifi":{"AP":1,"SSId":"Phradio","BSSId":"5C:49:79:23:C2:3B","Channel":6,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:18:47.412 MQT: tele/tasmota_C5C718/SENSOR = {"Time":"2022-01-02T23:18:47","ANALOG":{"A0":260},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}
23:18:57.338 MQT: tele/tasmota_C5C718/STATE = {"Time":"2022-01-02T23:18:57","Uptime":"0T00:02:30","UptimeSec":150,"Heap":22,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"ON","Channel1":100,"POWER2":"OFF","Channel2":0,"Color":"FF00","Wifi":{"AP":1,"SSId":"Phradio","BSSId":"5C:49:79:23:C2:3B","Channel":6,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:18:57.411 MQT: tele/tasmota_C5C718/SENSOR = {"Time":"2022-01-02T23:18:57","ANALOG":{"A0":255},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}

ChrisVonFrosta, 2022-01-03:
I did recalibrate.
Maybee it depends on the "inverted" output of the adc?
if i do calbration for low PH value with 4.0 solution.
'''AdcParam 9 , 4.0, 682, 7.0, 518 
the ph value is doing the wrong direction and do not exceed 7.0.

when is change the directions.
'''AdcParam 9,7.0, 518 , 4.0, 682

i got crazy values.

ChrisVonFrosta, 2022-01-03:
10:15:01.407 APP: Analog pH read. ADC-RAW: 495, cal-low(pH=ADC): 7.00=518, cal-high(pH=ADC): 4.00=682
10:15:01.413 MQT: tele/tasmota_C5C718/SENSOR = {"Time":"2022-01-03T10:15:01","ANALOG":{"pH0":-78566464.00}}
10:15:02.464 APP: Analog pH read. ADC-RAW: 494, cal-low(pH=ADC): 7.00=518, cal-high(pH=ADC): 4.00=682
1

arendst, 2022-01-03:
It seems the formula expects the low value as parameter 2 and the high value as parameter 4 like your first entry from above.

This is the code that does the calculation:
```
float AdcGetPh(uint32_t idx) {
  int adc = AdcRead(Adc[idx].pin, 2);

  float y1 = Adc[idx].param1 / ANALOG_PH_DECIMAL_MULTIPLIER;
  uint32_t x1 = Adc[idx].param2;
  float y2 = Adc[idx].param3 / ANALOG_PH_DECIMAL_MULTIPLIER;
  uint32_t x2 = Adc[idx].param4;

  float m = (y2 - y1) / (x2 - x1);
  float ph = m * (adc - x1) + y1;

  char phLow_chr[6];
  char phHigh_chr[6];
  dtostrfd(y1, 2, phLow_chr);
  dtostrfd(y2, 2, phHigh_chr);
  AddLog(LOG_LEVEL_DEBUG, PSTR(D_LOG_APPLICATION "Analog pH read. ADC-RAW: %d, cal-low(pH=ADC): %s=%d, cal-high(pH=ADC): %s=%d"), adc, phLow_chr, x1, phHigh_chr,x2);

  return ph;
}
```

ChrisVonFrosta, 2022-01-03:
Thanks for pointing the formula.

could u take a look to the log from my first entry?
the meassured analog input is 485 but it calcutlate it to 7 PH, while the 7 PH is calibrated with 518

ChrisVonFrosta, 2022-01-03:
i tried with other ph solution:
`AdcParam 9, 4.0, 684, 9.23, 396`

also crayz effects.
he probe is still in the PH 4.0 solution. website shows 4.0 after refresh and then 9.22. It toggled with F5 (website refresh) between this both values.
In the Weblog i never see flipping raw values.

19:57:54.424 APP: Analog pH read. ADC-RAW: 680, cal-low(pH=ADC): 4.00=684, cal-high(pH=ADC): 9.22=396
19:57:54.431 MQT: tele/tasmota_C5C718/SENSOR = {"Time":"2022-01-03T19:57:54","ANALOG":{"pH0":9.22}}

barbudor, 2022-01-03:
It looks like the formula is prefers to have the Low ADC value first because it's using unsigned int

I tried to make it more proof to inversions
Can you with the attached binary ?
[tastmota-test-ph.zip](https://github.com/arendst/Tasmota/files/7803704/tastmota-test-ph.zip)

ChrisVonFrosta, 2022-01-04:
it seems to work.
Great. Thx for your time!

ChrisVonFrosta, 2022-01-04:
should i close this or do this someone else?

ChrisVonFrosta, 2022-01-04:
If you like to point out your changes in the code, i would be happy to see that

barbudor, 2022-01-04:
You can see the change in the PR : https://github.com/arendst/Tasmota/pull/14275
Once merged, this will be part of the next Development builds until next official release

closed 2022-01-04 by arendst