arendst/Tasmota #14261: Shelly EM not reporting data

ebarretopr, 2022-01-03:
### PROBLEM DESCRIPTION
After Flashing Shelly EM with Tasmota over the air, there is no power monitoring information being provided just relay control I have used the following two templates with option 21 turn on and off without any avail

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Shelly EM_____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: OTA_____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:01:01.832 RSL: RESULT = {"NAME":"Shelly EM","GPIO":[0,0,288,0,32,0,0,0,640,3456,608,224,0,1],"FLAG":0,"BASE":18}
23:01:02.038 RSL: RESULT = {"Module":{"0":"Shelly EM"}}
23:01:02.294 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"288":"Led1"},"GPIO3":{"0":"None"},"GPIO4":{"32":"Button1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"3456":"ADE7953 IRQ1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
23:01:38.015 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
23:01:38.225 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
23:01:38.474 RUL: Stored uncompressed, would compress from 9 to 10 (-0%)
23:01:38.477 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":9,"Free":502,"Rules":"`: ```lua"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

23:01:57.611 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"ShellyEM","FriendlyName":["ShellyEM"],"Topic":"tasmota_B0C44E","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
23:01:57.637 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:11","StartupUTC":"2022-01-03T21:55:46","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2022-01-03T22:22:11","SaveCount":44,"SaveAddress":"F6000"}}
23:01:57.670 RSL: STATUS2 = {"StatusFWR":{"Version":"10.1.0(tasmota)","BuildDateTime":"2021-12-08T14:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"343/699"}}
23:01:57.690 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Panorama Command",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
23:01:57.722 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1620C2","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
23:01:57.752 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-B0C44E-1102","IPAddress":"192.168.253.59","Gateway":"192.168.253.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.253.1","DNSServer2":"0.0.0.0","Mac":"84:CC:A8:B0:C4:4E","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:01:57.776 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B0C44E","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:01:57.798 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-01-03T22:01:57","Local":"2022-01-03T23:01:57","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:05"}}
23:01:57.815 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-01-03T23:01:57"}}
23:01:57.822 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-01-03T23:01:57","Uptime":"0T00:06:11","UptimeSec":371,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Panorama Command","BSSId":"C4:41:1E:F7:D4:90","Channel":2,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-03:
Can you please provide the output of the command `i2cdriver`
The result should include the number `7`
If there is a `!` in front of `7`, type `i2cdriver7 1 ` to enable the ADE7953  driver

ebarretopr, 2022-01-03:
RESULT = {"I2CDriver":"7"}

[image]

barbudor, 2022-01-03:
Does your template configuration looks like the one below ?

Please set `Weblog 3` and then restart  (with `Restart 1` for exemple) and provide a copy of the logs from start up to first "tele STATE" message

[image]

barbudor, 2022-01-03:
Please also report the result of `i2cscan` command
thanks

ebarretopr, 2022-01-03:
Yes similar to yours

First Query:

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F7, Count 18
00:00:00.056 QPC: Count 1
00:00:00.058 CFG: CR 343/699, Busy 0
00:00:00.062 ROT: Mode 1
00:00:00.164 SRC: Restart
00:00:00.166 Project tasmota - ShellyEM Version 10.1.0(tasmota)-2_7_4_9(2021-12-08T14:47:33)
00:00:00.218 WIF: Attempting connection...
00:00:00.555 WIF: Connecting to AP1 Panorama Command Channel 2 BSSId C4:41:1E:F7:D4:90 in mode 11n as tasmota-B0C44E-1102...
00:00:01.753 WIF: Attempting connection...
00:00:02.752 WIF: Attempting connection...
00:00:03.752 WIF: Connected
00:00:04.004 HTP: Web server active on tasmota-B0C44E-1102 with IP address 192.168.253.59
00:00:04.672 RTC: UTC 2022-01-03T22:34:45, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
23:34:46.279 RSL: INFO1 = {"Info1":{"Module":"Shelly EM","Version":"10.1.0(tasmota)","FallbackTopic":"cmnd/DVES_B0C44E_fb/","GroupTopic":"cmnd/tasmotas/"}}
23:34:46.284 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-B0C44E-1102","IPAddress":"192.168.253.59"}}
23:34:46.294 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
23:34:46.301 RSL: RESULT = {"POWER":"OFF"}
23:34:46.304 RSL: POWER = OFF
23:34:47.400 QPC: Reset
23:34:49.403 APP: Boot Count 8
23:34:49.792 CFG: Saved to flash at F6, Count 19, Bytes 4096
23:34:50.381 RSL: STATE = {"Time":"2022-01-03T23:34:50","Uptime":"0T00:00:11","UptimeSec":11,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Panorama Command","BSSId":"C4:41:1E:F7:D4:90","Channel":2,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:05"}}

Second Query i2scan:

23:35:59.750 CMD: i2cscan
23:35:59.752 SRC: WebConsole from 192.168.253.141
23:35:59.753 CMD: Grp 0, Cmnd 'I2CSCAN', Idx 1, Len 0, Data ''
23:35:59.771 RSL: RESULT = {"I2CScan":"No devices found"}

barbudor, 2022-01-03:
What does means "similar" ?
It should be identical. Especially GPIO12 to GPIO14 MUST be the same as my screenshot.

Currently Tasmota cannot find the ADE7953  energy monitoring chip on the I2C
Either because your template is not correct, or because your hardware is deffective

ebarretopr, 2022-01-03:
[image]

Is there a way I could do a reflash? my hardware was working fine prior the flash.

barbudor, 2022-01-03:
If you want to reflash Tasmota, use OTA from the menu -> firmware upgrade and use http://ota.tasmota.com/tasmota/release/tasmota.bin.gz

If you want to revert to Shelly's original firmware, follow instructions from https://github.com/yaourdt/tasmota-to-mgos

ebarretopr, 2022-01-03:
After reflash same result after I2Cscan

23:50:14.637 CMD: Grp 0, Cmnd 'I2CSCAN', Idx 1, Len 0, Data ''
23:50:14.655 RSL: RESULT = {"I2CScan":"No devices found"}

barbudor, 2022-01-03:
I don't own a Shelly EM myself, so I have asked to other members to double check if they see any problem with v10.1.0.

When you say it was working ok prior to the flash, you mean with Shelly's original firmware ?
Is that a recent device ?

If other members confirm that v10.1.0 is working on their unit, it may be possible that Shelly's has changed the design and is now using a different, non I2C chip.
In that case, would you be able to open the device and check that ? Even providing good photos ?
I have to leave now. I'll be back tomorrow.

barbudor, 2022-01-03:
Hi. I've been told that after flashing you must disconnect the Shelly from power and reconnect.
A soft reboot is not enough

ebarretopr, 2022-01-03:
Ok, Will reflash tasmota and let you know the results

ebarretopr, 2022-01-04:
The reboot did it. Working great Thanks!

closed 2022-01-04 by ebarretopr