arendst/Tasmota #14262: Tasmota MQTT stops working

dog-rgb, 2022-01-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
An ESP32 device is setup and running with Tasmota32 Bluetooth. It should connect to some EQ3 and LYWSD03MMC / ATC devices and provide the data via MQTT. In general it works but after some hours are latest after several days it stops working and does not send out any data any longer. In such case it is also not possible to connect to the device via http. The only workaround I am using is to remove the power supply and connecting again after some seconds. After this cold start it is working again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____ ESP32 REV1 CH340
- [X] Tasmota binary firmware version number used: _____ 10.0.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: _____ Web Installer/Self update
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

If needed I can provide later (currently no access)
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

If needed I can provide later (currently no access)
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

If needed I can provide later (currently no access)
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

The console of the device is not accessible.
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
No steps for reproduction known

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The device should be available via TCP all time

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-01-03:
Upgrade to latest 10.1.0 release (or even to latest 2022.01.1 development version)
There was some problem in the Espressif Core SDK used in v10.0.1.0 that was fixed in a more rerecent Espressif Core SDK

Jason2866, 2022-01-05:
We have made the experience that doing only a OTA upgrade does NOT solve.
Erase flash and flash clean from scratch

ascillato2, 2022-01-06:
Please, when you have time, tell us if erasing and flashing latest version solves your issue. Thanks.

closed 2022-01-06 by ascillato2

dog-rgb, 2022-01-06:
I wanted to wait at least some time to see if it helps. I did the update via OTA only, but even with this the device is available since more than two days, There was at least one SW reset in between, but the device is still available. So this topic can be closed from my side,

gdavidp, 2022-01-13:
Looks like the the last Rule3 is solving my issue as a workaround. So far I have had one ESP32 restart on it's own and maintained WIFI connection. Thanks to all your input, I will keep watching them as it has been only 12 hours.